neu-rah/ArduinoMenu #7: Serial Menu + Encoder. It is possible??

techboycr, 2015-04-21:
This library is just amazing!

Is there a possibility to use Serial Menu with Encoder?? I am using this Serial LCD but I need the encoder for the menu.

How may I achieve this?

neu-rah, 2015-04-29:
Yes its possible, i've been using one (3 wires) but you can virtually adapt any other as the library input is generalized to streams, in this case my encoder driver (included quadEncoder.h) just outputs characters '+','-','Enter' for the movements and click.

techboycr, 2015-04-29:
Thar si a great thing to know. I Am so sorry for being this dumb but would you please send me an example in how to do it? 

Please!! I am desperate, I being trying and failing miserably.

neu-rah, 2015-05-28:
as in example/lcd_menu:

//define the pins to be used
#define encA 2
#define encB 4
//this encoder has a button here
#define encBtn A0

...

//the quadEncoder
quadEncoder quadEncoder(encA,encB);//simple quad encoder driver
quadEncoderStream enc(quadEncoder,5);// simple quad encoder fake Stream

//a keyboard with only one key :D, this is the encoder button
keyMap encBtn_map[]={{-encBtn,menu::enterCode}};//negative pin numbers means we have a pull-up, this is on when low
keyLook<1> encButton(encBtn_map);

//multiple inputs allow conjugation of the quadEncoder with a single key keyboard that is the quadEncoder button
Stream* in[]={&enc,&encButton};
chainStream<2> quadEncoder_button(in);

//alternative to previous but now we can input from Serial too...
Stream* in3[]={&enc,&encButton,&Serial};
chainStream<3> allIn(in3);

check out the example file, it runs on the lcd also used on ide demo

SrR0, 2015-05-28:
Hello Rui,

many thanks for your help! It works great now with 1.6.4. I asked about 
1.0.6 cause i have the libs of the 1.0.6 environment installed in my 
AtmelStudio which i am using instead of the original arduino ide. I 
think I have to play around a bit on installing the 1.5.x or the 1.6.x 
libraries instead of the 1.0.6. But it should be a not too big problem 
(planned to go this step sooner or later anyway).

regards and greetings from austria

Robert

Am 28.05.2015 um 02:36 schrieb Rui Azevedo:
> //define the pins to be used
> #define encA 2
> #define encB 4
> //this encoder has a button here
> #define encBtn A0
>
> ...
>
> //the quadEncoder
> quadEncoder quadEncoder(encA,encB);//simple quad encoder driver [...]

closed 2015-09-24 by neu-rah