neu-rah/ArduinoMenu #9: Compile problem with FIELD

SrR0, 2015-05-16:
Hello Rui, wrote an email yesterday but found the forum now, sorry for that:

I found your Arduino-Menue Lib in youtube. Its really a cool thing which saves a lot of time in coding embedded application. I would like to use it in my Balancing Robot project. It took me no affords to get the example code with the i2c lcd running in my arduino 1.0.6 ide on windows. But if i add a FIELD menue entry the compiler throws out an error:

I2C_menu:34: error: variable 'menuField<const char [5]> _menuField0' has initializer but incomplete type

the modified Code is here

/********************
Arduino generic menu system
Arduino menu on I2C LCD example
http://www.r-site.net/?at=//op%5B%40id=%273090%27%5D

Sep.2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
creative commons license 3.0: Attribution-ShareAlike CC BY-SA
This software is furnished "as is", without technical support, and with no
warranty, express or implied, as to its usefulness for any purpose.

Thread Safe: No
Extensible: Yes

LCD library:
https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home
http://playground.arduino.cc/Code/LCD3wires
*/
#include <Wire.h>
#include <LiquidCrystal_I2C.h>
#include <menu.h>//menu macros and objects
#include <menuLCDs.h>//F. Malpartida LCD's

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
menuLCD menu_lcd(lcd,16,2);//menu output device

int para=0;
void nothing() {}

MENU(mainMenu,"Main",
 FIELD(para,"Para","Unit",0,1000,10,1,nothing),
 OP("Frequency",nothing),
 OP("Dutty",nothing),
 OP("Setup",nothing)
);

void setup()
{
 Serial.begin(9600);
 Wire.begin();
 lcd.begin(16,2);
 lcd.print("Ok");
}

void loop()
{
 //serial can use numbers or +/- and Enter=* Esc=/ to navigate (nav keys can be redefined)
 mainMenu.poll(menu_lcd,Serial);
}

Maybe you can be so kind and point me to the right way cause my knowledge in #-directive is very weak (yes i am a spoiled java-guy in real life;-))

greetings from austria and thanks

robert

neu-rah, 2015-05-26:
#include "menuFields.h"
//is required to work with firlds

SrR0, 2015-05-26:
Hello,

thank you very much for your answer! But unfortunatly i am still stuck with this:

included now this files:
==============================================
#include <Wire.h> 
#include <LiquidCrystal_I2C.h>
#include <menuFields.h>
#include <menu.h>//menu macros and objects
#include <menuLCDs.h>//F. Malpartida LCD's

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
menuLCD menu_lcd(lcd,16,2);//menu output device
int para;
void nothing() {}

MENU(mainMenu,"Main",
  FIELD(para,"Para","Unit",0,100,10,1,nothing),
  OP("Frequency",nothing),
  OP("Dutty",nothing),
  OP("Setup",nothing)
);

==============================================
I get now some different Errors while compiling:
==============================================
This report would have more information with  "Show verbose output during compilation" enabled in File > Preferences. Arduino: 1.0.6 (Windows XP), Board: "Arduino Uno"  In file included from I2C_menu.ino:20:
C:\Dokumente und Einstellungen\Robert\Eigene Dateien\Arduino\libraries\ArduinoMenu-master/menu.h: In member function 'void menuToggle<T>::activate(menuOut&, Stream&, bool)':
C:\Dokumente und Einstellungen\Robert\Eigene Dateien\Arduino\libraries\ArduinoMenu-master/menu.h:238: error: object missing in reference to 'menu::sel'
C:\Dokumente und Einstellungen\Robert\Eigene Dateien\Arduino\libraries\ArduinoMenu-master/menuFields.h:166: error: from this location
==============================================

Are there still some includes or libs missing?

best regards and many thanks!

Robert

neu-rah, 2015-05-26:
> #include //menu macros and objects
> #include //F. Malpartida LCD's
> function 'void menuToggle::activate(menuOut&, Stream&, bool)':

Hi!

And you are using an I2C LCD?
Also you must have the LCD drivers refered, in this case the 
LiquidCrystal_I2C.
Its an independent display driver not included on the menu library...

i'm taking some guesses here...

you can send me your code if you wish

regards
Rui

SrR0, 2015-05-27:
Hello,

thanks for your help. I managed it to compile it with the 1.6.4 version 
of the ide with the original liquidcrystal lib wich is contained there. 
On 1.0.6 it did not compile. What is your compiler version an which libs 
are you using?

regards

Robert

neu-rah, 2015-05-28:
Last tested on 1.5.8
and yes, your specific lcd driver is crucial, you should have a working 
lcd just to ensure your driver is ok

I've tested linking with several drivers, including the original LCD 
example that comes with the IDE and some others like:
I2C Lcd's
A Graphics display with adafruit's GFX graphics, its important because 
it covers quite some screens (nice work adafruit!)
And recently with a touch device on arduino mega
But of course some other devices/drivers might have to had some adjustments.
Virtually any display can be adapted because we assume litle from them.
Its always a balance between power and abrangence. The lower line is of 
course the text console.
A text console can accept sequences of text to be drawn and eventually a 
newline and we cheat a lot with some other command like cursor position
we consider also a clearscreen (console just prints an extra blank line, 
its the best it can do)... but other screens should have an easy 
implementation for clear

So if one builds a routine that translates that needed functions to a 
particular device then that device can display the menu.
The core functionality is described in menu.h

class menuOut:public Print {...

existing derived classes are:

*menuPrint.h* //menu output to Print device (ex: Serial)
*menuLCD.h* //Use standard arduino LCD (LiquidCrystal library) as menu 
output
*menuLCDs.h* // implement menu output for Francisco Malpartida arduino 
LCD's (its a collection of drivers) from 
http://playground.arduino.cc/Code/LCD3wires
*menuGFX.h* // Use graphics screens (adafruit library based) as menu output

well this list should be taken with a grain of salt because display 
libraries can change versions and sometimes change behavior (avoided).
you should pick the appropriate include file for the driver in use. (or 
extend the core functionality, for other screens)

I didnt bothered with 1.0.6 (should we?) because 1.5.8 was out already,
some 1.0.6 errors on mac have been reported with solution by IDE upgrade

closed 2015-06-12 by neu-rah