neu-rah/ArduinoMenu #10: Text fields

neu-rah, 2015-06-12:
received suggestion Alex

Thank you for your Arduino menu, I run it on esp8266 device and it works good!
Can you add ability to edit string via menu?

neu-rah, 2015-06-12:
lets say, having a a string we can:
0) click to start editing the string
1) use encoder to position cursor (string navigation)
2) click to edit letter
3) use encoder to choose letter
4) click to end edit and return to string navigation
... well now we need a method to get out of string navigation

i'm accepting sugestions here

neu-rah, 2015-06-12:
from Alex

I think your string edit algorithm(0-4) is good.
Field for text I see like this:

TEXTFIELD(
*char string, - pointer to string array to edit
char length, - length of this string
"Name", - display name
*char values, - available letters list ["a","b","1","2","-"...]
,updateName - function after text was edited
)

I thin, program in step 3 must show letters from available letters array.
After edit call updateName function.

neu-rah, 2015-06-12:
I was considering a list a of valid symbols for each position, maybe using a string as available letters list (same functionality)

the format seems ok

however the implementation detail for encoder, the problem with encoder is that we only have 2 directions and 1 button (and i don't want to use double-clicks because this code is already large)

for a linear edit we can use the old algorithm as in game machines score names, introduced with a joystick and 1 or 2 buttons.

=> appending extra symbols as available "letters" to be used as commands
the most common use is the command END
eventually a delete or backspace

as alternative to that algorithm we can use a simple grid edit
=> string has  fixed length and we can change individual letters or set them as space (if allowed)
1-click to edit,
2- move to the letter we want to change,
3- click to change letter
4- move in allowed symbols
5- click to accept symbol
-> ok now we must get out of string edit and back to menu navigation
-> to change next letter you must start from step number 1 again

this is tedious but simple, ok for casual text insert

other algo's require commands as symbols, or commands as cell positions
- we could have an [end] symbol at the end of the string and user would then srcroll to that symbol and click to get out of the string editing mode, we could even had a cancel to ignore changes, this way we could remain at level 2

this methods are less tedious but... the have a larger footprint /and arduinos are small)
screens are sometimes very limited to have this sort of "buttons" appended to string

on the other hand some hardware might have 2 or more buttons and would benefit from a smarter implementation much less tedious

things like the string navigation level can be omitted and we could edit the string by remaining in the "symbol choose stage" and have the cursor moved to the next character on click, this way we can skip characters that we do not want to edit, still deleting a character would benefit of that extra commands as symbols.
extra commands as symbols make a bigger implementation but do not require extra screen space <-- i like this combination
we could remain at level 4

and well i should be looking at a way of pushing the menu data into PROGMEM to allow more memory space for user program instead of thinking of bigger things that will restrict user memory.

do you have any experience with PROGMEM?

what about string with length bigger than the screen? this must be considered because we have no idea what the screen would be and the scroll edit mechanism can consume some ram (i think)... well i have to look better at this part.

well at least user can toss things like allowed character tables into progmem.

char* digits="0123456789"
char **mask1=["012",digit,digits] <-- this must agree with string length

char* letters="abc...XYZ"
char **name =[letters,letters,letters,...,letters]

well no combinatories here, if we want alphanumerics its a complete new set

this should really be an issue on github (we can mark them as [enhancements])
and beneffit from ideas of other people, i'm open it now

SamuelGold, 2015-06-13:
I like idea to omit 5 level, but using PROGMEM can loss compatibility with another, non arduino platforms(I use esp8266, for example), also menu text is not so big.
Now I see whole algorithm to edit small string, I can show what it my mind on display:

Name: noname
Press button, for example < symbol is cursor:
Name: n<name
Edit n to A and press button:
Name: Ao<ame
Edit o to l and press button:
Name: Aln<me
Edit n to e and press button:
Name: Alea<e
Edit a to x and press button:
Name: Alexm<
Edit m to space and press button, new cursor > goes before the last symbol:
Name: Alex>e
Edit e to space and press button on last symbol to exit edit with confirmation:
Name: save?
Scroll change save to cancel:
Name: cancel?
Scroll change cancel to edit:
Name: edit?

If press button on "save?" then call updateName, or discard changes on "cancel?", or continue edit from first symbol on "edit?".

But if there is a long string string, we can use another cursor and scrolling string, for example small dispaly:

Name:  __>n<o
Name:  _A>o<n
Name:  Al>n<a
Name:  le>a<m
Name:  ex>m<e
Name:  x_>e<
Name:  save?

_ - is empty space.

neu-rah, 2015-09-24:
there is a prototype for that included in current versio, see TEXTFIELD's. However you can loop inside your function and do whatever needs to be edit before returning to the menu. It wont be as a field because the value will not be shown along with the option.
The prototype text input is very rude yet,

neu-rah, 2016-06-29:
better to suspend the menu (now w can) and use a function to edit the string

closed 2016-06-29 by neu-rah