neu-rah/ArduinoMenu #11: Serial (TX only) with Encoder

christophepersoz, 2015-08-03:
Hello,

Is there a way to use serLCD.h instead of the library you had implemented ? I'm using that library to communicate with a 1 wire serial LCD which are not supported yet by the LCD Libraries included.

Thank you

neu-rah, 2015-08-04:
Can you post a link to the library and
      lcd model?
      
      thanks
      
      
      On 03-08-2015 23:53, christophepersoz wrote:
    
    
      Hello,
      Is there a way to use serLCD.h instead of the library you had
        implemented ? I'm using that library to communicate with a 1
        wire serial LCD which are not supported yet by the LCD Libraries
        included.
      Thank you
      —
          it on GitHub.

christophepersoz, 2015-08-04:
Hello,

Here the display I'm using : https://www.sparkfun.com/products/9395
The library to drive it : https://github.com/nemith/serLCD which is an update of the library you can find on Arduino.cc playground website : http://playground.arduino.cc/Code/SerLCD
I had a look on your files, and all the LCD lib seems working on SDA/SCL or parallel mode - but maybe I'm wrong.

Thanks

neu-rah, 2015-08-04:
Looking at the library it looks ok.
      
      the display menu driver is similar to:
      https://github.com/neu-rah/ArduinoMenu/blob/master/menuLCD.h
      
      duplicate it and just use the menu lcd driver virtual functions
      (write,clear,setCursor) to call the apropriate LCD functions on
      your driver
      
      please, let me know about it
      
      
      
      On 04-08-2015 07:50, christophepersoz wrote:
    
    
      Hello,
      Here the display I'm using : https://www.sparkfun.com/products/9395
        The library to drive it : https://github.com/nemith/serLCD
        which is an update of the library you can find on Arduino.cc
        playground website : http://playground.arduino.cc/Code/SerLCD
        I had a look on your files, and all the LCD lib seems working on
        SDA/SCL or parallel mode - but maybe I'm wrong.
      Thanks
      —
          it on GitHub.

christophepersoz, 2015-08-06:
Hello,

After several tried it seems that everything works fine now, thanks you for your help.
I have a question, on a 20x4 LCD, I limited the menu to 2 lines with the command

menuLCD lcd(lcd1,20,2); // Let only 2 lines available for the menu

Is there a way to place the menu on line 2 and 3 instead of 0 and 1 ?
Also, is it possible to get a kind of menu index to know where I am. For example, am I inside the main menu or inside a SubMenu, and which one?

Thanks in advance

neu-rah, 2015-08-06:
Hi, I'm glad its working
    
    After several tried it seems that
        everything works fine now, thanks you for your help.
      
    No problem, If you fill like sharing it just fork the project and
    then do a pull-request with your changes.
    
    I have a question, on a 20x4 LCD, I
        limited the menu to 2 lines with the command
        menuLCD lcd(lcd1,20,2); // Let only 2 lines available for the
        menu
        Is there a way to place the menu on line 2 and 3 instead of 0
        and 1 ?
    
    Positioning the menu
    
    class menu:public menuNode lets you position the menu by changing
    members ox,oy
    
    example:
    MENU(mainMenu,"Main menu",
        /*OP("LED On",ledOn),
        OP("LED Off",ledOff),*/
        OP("Disabled option",none),
        SUBMENU(onoff_tog),
        FIELD(param,"Name","%",0,100,10,1)
      );
    
    
    
    just call 
    mainMenu.oy=2
    
    before using the menu (possibly arduino setup function)
    
    Also, is it possible to get a kind of
        menu index to know where I am. For example, am I inside the main
        menu or inside a SubMenu, and which one?
    
    Yes
    
    the menu device object has a menu* to the last drawn
    menu
    your driver derives from this class (menuOut) defined in menu.h
    
    them pointed object (menu) has int sel member that
    is the select menu option
    menuPrint menu_out(Serial);//describe output device
      menu_out.drawn //<-- the active menu
      menu_out.drawn->sel // <-- selected line
      
    
    I've not tested this code samples but let me know if there is some
    problem
    
    Thanks for using the software, hope it can be as helpful as it was
    for me

christophepersoz, 2015-08-06:
Hello,

Thanks a lot for all the tips and informations. Unfortunately, neither the position and index information is working.

Here the menu setup,
MENU(m_Main,"Main menu",
  OP("Test 0", nothing),
  OP("Test 1", nothing),
  OP("Test 2", nothing)
);

//describing a menu output, alternatives so far are Serial or LiquidCrystal LCD
menuLCD lcd(lcd1,20,2); // Let only 2 lines available for the menu

menuPrint m_output(Serial);//describe output device

Here my code for the setup (only the part)
void setup() {
Serial.begin(9600);
    Serial.println("menu system test");
    
    quadEncoder.begin();
    // Switch on the backlight and set display
    pinMode ( BACKLIGHT_PIN, OUTPUT );
    digitalWrite ( BACKLIGHT_PIN, HIGH ); // 70
    lcd1.begin(20,4);             // initialize the lcd
    lcd1.clear();
    lcd1.home();                   // go home
    lcd1.print("Menu test");

    pinMode(encBtn, INPUT);
    digitalWrite(encBtn,1);
    
    pinMode(LEDPIN,OUTPUT);

    m_Main.oy=2; // set the position on LCD on line 2 instead of 0
    // m_Main.setPosition(0,2); // I also tried this method without any success on positioning
}

And the code for the loop, for debug

void loop() {
    // testing the menu system
    m_Main.poll(lcd,quadEncoder_button);

    m_output.drawn; //<-- the active menu
    Serial.println("-> Active menu : ");
    m_output.drawn->sel; // <-- selected line
    Serial.println("-> Selected line : ");
    delay(300);
}

I had a look inside menu.h, but I don't have strong c++ skills, so the tips you gave me should work, but...
I saw a public method in class::menu called setPosition, but it did not work too.
Any idea of what I did wrong ?

As I'm continuing to test and try this nice menu class, I would like to know if there is a possibility to hide the menu (can be done by lcd.clear();), and show it up by moving encoder or press en encBtn ?
I know that lcd.clear(); only erase the screen, but all the interruptions behind are still running.

Thanks a lot !

christophepersoz, 2015-08-13:
Hello,

After few trials and trying to understand your code I notice that effectively m_output.drawn->sel works perfectly.
I just didn't find the way to know that I'm on top of the menu tree, so at this time I test canExit property which is not the best way, but it works.

Did you see my question about the possibility to show and hide the menu ? Is there an another way to do than clear/redraw ?

Thanks a lot for your help.

neu-rah, 2015-09-24:
class menuNode (base of menu) has a static member **menuNode** pointing to the current menu.
therefor

    menuNode::activeMenu;// is the current menu
    menuNode::activeMenu->sel;// is the index of selected option (hilited)

and I've just added functions redraw and focus

neu-rah, 2015-09-26:
would you be kind enought to share the one-wire lcd driver?
if so just fork and pull with driver

christophepersoz, 2015-09-26:
I finally use the LCD library extended by Bill Perry that can be found there : https://github.com/Xerxes3rd/LiquidCrystal. It's a smart library because a lot of driver chipset are available.

Add the following lines for the includes on I2C :

`#include <Wire.h>
#include <LiquidCrystal_IIC.h>  //LCD driver for IIC by Bill Perry (https://github.com/Xerxes3rd/LiquidCrystal)`

and then,

`LiquidCrystal_IIC _Display(0x3f, IIC_PCF8574, 2,1,0,4,5,6,7,3,POSITIVE);    // Define 20x4 LCD Display on I2C bus at address 0x3F / 63`

There is a smart method inside this library that can help you to find the address of your LCD device on I2C bus. In my case the manufacturer mentioned the wrong address on the documentation.

After that, I just use the common declaration

`//describe output device
menuLCD _menuLCD(_Display,20,4); `

closed 2016-06-28 by neu-rah