neu-rah/ArduinoMenu #14: Add a function on SUBMENU call

christophepersoz, 2015-09-02:
Hello,

You menu is great to create complex structure and easy debugging/navigation/use, thanks a lot to share it.
Now that I'm much more comfortable with your library, I would like to expand it and add a function on SUBMENU call. Is that possible ? I tried but it seems that is only possible for OP, FIELD and VALUE.
Any track to follow ?

Thanks in advance

neu-rah, 2015-09-24:
you can use a regular function and then use a loop inside of it to poll another menu

christophepersoz, 2015-09-26:
I tried to replace SUBMENU call by an OP calling a function that poll the SUBMENU which was called before but it result in a mess on display.

neu-rah, 2016-06-29:
please clear the screen before cycling the second menu

neu-rah, 2016-07-21:
just added it, now every element can have an associated action
examples of assigning actions:
```c++
submenu.action=trig;//test on submenu
sample_clock.action=trig;//test on choose
onoff_tog.action=trig;//test on toggle
mainMenu.data[1]->action=trig;//test on field
```
regular options already have an associated action, but it can be changed just like a field

closed 2016-07-21 by neu-rah