neu-rah/ArduinoMenu #15: Dynamic menu labels

nicodonte, 2015-09-26:
Hi @neu-rah, its me again!!

There is a way to make dynamic menu names? So if I change somenthing in a function then will change the name in the menu.

For example:

If I have a menu item with name: Start Time XX:XX

When I setup this time to YY:YY, when I return to the menu its shows Start Time YY:YY.

Thanks,
Nico.

nicodonte, 2015-09-26:
I managed to solve partially by converting integer to String, and then convert to char. But I don't know if there is an easier way to do it.

int nivelBlight=200;
String nivelBlightText = "Nivel: ";
char nivelText[10 + 1];

MENU(configMenu, "Configuracion",
  SUBMENU (relojMenu),
  OP(nivelText, bklMenu)
  );

  String nivelBlightText = "Nivel: ";
  nivelBlightText.concat(map(nivelBlight, 0 , 255 , 0 , 100));
  nivelBlightText.toCharArray(nivelText,11);

Regards,
Nico.

nicodonte, 2015-09-27:
Hi, could you make an example of:
menuNode::activeMenu;
menuNode::activeMenu->sel;

I can't figure out how to use it.

neu-rah, 2015-10-16:
you can change the menu text of a prompt, ex:

mainMenu.data[2]->text="Some new text";//change third option text
mainMenu.redraw(lcd,allIn);

this works, but flicks the menu on redraw, so frequent changes will leave the menu flicking

nicodonte, 2015-10-16:
Thanks for your reply.
I'm still struggling with the POINTER variables to know where I am located in the menu. I do not understand how to make a comparison of these variables.

neu-rah, 2015-10-17:
each menu prompt has an associated function that is called when the prompt is selected, if you have a different function for each prompt then you will just know.
the idea is to leave the navigation details to the menu.
however the associated function can be as simple as a "void f()" or can have extra information passed to it "void sayIt(prompt& p,menuOut& o,Stream &c)", on this version you will have a reference to the activated prompt, menuOut device , and input stream.
you can have any combination of the above, from () to the full length (it allows parameters to be omitted).
so you still need the navigation info?

nicodonte, 2015-10-17:
Yes, I still need it, it will simplify my code if I could use switch ... case structure. It will be useful for me also if I can put a variable to the OP menu command or maybe calling a procedure with a variable inside OP("text", routine(variable)).

Thanks in advance for your time and helpful hints!!

neu-rah, 2015-10-17:
well it might not be safe as internal navigation details can change... anyway
menuNode::activeNode is the pointer to an active menu node (or zero if none)
this is an object of type menuNode, but you can cast it to menu if you are sure it is a menu
((menu*)menuNode::activeNode)->...
with the casting you can access any of menu data members (you can cast to a menuField if its the case).
((menu*)menuNode::activeNode)->sel; // its the last selected option on current active menu
is that what you want?

nicodonte, 2015-10-17:
I don't need to access to one specific menu right now, I need to know which menu was the last selected, cos I can call the same routine and change variables according to the menu I was. In my project, I have four timers, each of them has same options, start time, period, fade time, and level max. If I use OP in the menu, I have to write four times the same code, but if I knew where I came, it could be easy as using an array variable to set the timers variables.

nicodonte, 2015-10-17:
When I tried to use menuNode::activeNode->sel; 

I get error: 'class menuNode' has no member named 'sel'
'class menuNode' has no member named 'sel'

neu-rah, 2015-10-17:
not it has not, you need to cast it to menu*

nicodonte, 2015-10-17:
How I do that?

neu-rah, 2015-10-17:
int selOp=menuNode::activeMenu->isMenu()?((menu*)menuNode::activeNode)->sel:-1;

this will set the variable selOp to the selected menu option or -1 if none

nicodonte, 2015-10-17:
if I use int selOp=menuNode::activeMenu->isMenu()?((menu*)menuNode::activeNode)->sel:-1; in the main loop, it makes the screen to blink  constantly.

If I use int selOp=((menu*)menuNode::activeNode)->sel; I can see the menu number, but as you said before, I can't know in which submenu that  I came.

neu-rah, 2015-10-17:
maybe you should look at the NoMacros example, as a way of build a menu-system without using the macros. That way you will have variables for each menu and prompt and can compare them.
The problem with macros is that you don't know the addresses of prompts menus other than the main menu.
guess that a

    if (menuNode::activeMenu==mySubMenu1) ...

will solve your problem

nicodonte, 2015-10-17:
Ok I will try that method.  I'm a bit newbie programming in C.

Thank you so much for your time and patience.

neu-rah, 2015-10-17:
You're welcome. :D

closed 2016-06-28 by neu-rah

neu-rah, 2020-05-04:
just for reference

https://github.com/neu-rah/ArduinoMenu/issues/293

https://github.com/neu-rah/ArduinoMenu/commit/5c9710a2975b3c45737f2367f9d082e708225e08