neu-rah/ArduinoMenu #17: 5-button Keyboard support

jamiekowalczik, 2015-10-31:
Great job on this Menu!  Would it be easy to use a 5-button keyboard for inputs?  I'm struggling with figuring out how to use this with your menu.
http://yourduino.com/sunshop2/index.php?l=product_detail&p=334

Any assistance or pointers would be greatly appreciated!  

Thanks again for your efforts on building this great menu!

neu-rah, 2015-10-31:
yes, its supported.

from the lcd example, the encoder uses a 1 key keyboard

here is a multi-key keyboard input example

    //buttons pin maps (choose yours)
    #define enterBtn 4
    #define upBtn 5
    #define downBtn 6
    #define leftBtn 7
    #define rightBtn 8

    //negative pin numbers means we have a pull-up, this is ON when low
    keyMap mykb_map[]={
        {-enterBtn,menu::enterCode},
        {-upBtn,menu::upCode},
        {-downBtn,menu::downCode},
        {-leftBtn,menu::leftCode},
        {-rightBtn,menu::rightCode},
    };

    keyLook<5> mykb(mykb_map);

then use it

    mainMenu.poll(lcd,mykb);

please note that i've not even compiled this particular piece of code, take it with a grain of salt

neu-rah, 2015-10-31:
sory, i then looked to your hardware link and i can see it uses an analog input, not a digital one

for this we have build a stream just like **keyLook** class (defined in keyStram.h)

neu-rah, 2015-10-31:
ok, just uploaded a genericKeyboard.h that can be wired to your keyboard read function, provided that you define the retue codes (keycodes) as any of the defined in menu.h + menu.cpp

menu::escCode, menu::enterCode, menu::upCode, menu::downCode, or -1 if no key available

jamiekowalczik, 2015-10-31:
I really appreciate your response and assistance but I'm not having any luck.  I'm still a newb to Arduino so please excuse the lack of troubleshooting skills.  

While compiling I get this error:
no matching function for call to 'genericKeyboard::genericKeyboard(int (&)())'

Does the read_keyboard need a different return type?

neu-rah, 2015-10-31:
my fault, corrected that

jamiekowalczik, 2015-10-31:
I still can't figure it out.  What needed correction?

neu-rah, 2015-11-01:
the genericKeyboard.h file is now corrected

jamiekowalczik, 2015-11-01:
It's not compiling for me:

In file included from I2C_menu.ino:16:0:
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:24:16: error: expected ')' before '*' token
   int(keyReader*)();
                ^
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:27:29: error: expected ')' before '*' token
   genericKeyboard(int(reader*)()):keyReader(reader),lastkey(-1) {}
                             ^
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:27:29: error: expected ')' before '*' token
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:27:23: error: expected ';' at end of member declaration
   genericKeyboard(int(reader*)()):keyReader(reader),lastkey(-1) {}
                       ^
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:27:30: error: expected unqualified-id before ')' token
   genericKeyboard(int(reader*)()):keyReader(reader),lastkey(-1) {}
                              ^
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:27:29: error: expected ';' at end of member declaration
   genericKeyboard(int(reader*)()):keyReader(reader),lastkey(-1) {}
                             ^
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:27:30: error: expected unqualified-id before ')' token
   genericKeyboard(int(reader*)()):keyReader(reader),lastkey(-1) {}
                              ^
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h: In member function 'virtual int genericKeyboard::peek()':
/Users/jk/Documents/Arduino/libraries/ArduinoMenu/genericKeyboard.h:37:26: error: 'keyReader' was not declared in this scope
   int peek(void) {return keyReader;}

neu-rah, 2015-11-01:
yes, I had some typos in the code, now it compiles (code wasn't tested)

jamiekowalczik, 2015-11-01:
Works like a charm, call this one closed!  Thank you so much!!  I owe you a few pints, send me a paypal link!

jamiekowalczik, 2015-11-03:
For the most part everything is working great except up is down and down is up for menu navigation but up is up and down is down for when changing the value of a field.  

Any idea where I could be going wrong?
I have my sketch at:
https://github.com/jamiekowalczik/sketches/blob/master/i2c_menu

I'm still open to send along a donation if you are interested.

neu-rah, 2015-11-05:
pretty weird, this might well be related with the new generic keyboard, still you can correct that by putting negative numbers for step on FIELD initialization...
on genericKeyboard, I was looking at the line 34

   if (ch==lastkey) return 0;

wonder if the "available" function is returning 1 to latter the read return -1 (no key available)
can you test the "available()" return value when no key is pressed?

also a debug line can be introduced to immediate return 0 if peek is -1 and see if that fixes the issue

jamiekowalczik, 2015-11-06:
Reversing menu::downCode/menu::upCode and using negative numbers for FIELD works..

...
#define btnRIGHT  menu::enterCode
#define btnUP     menu::downCode
#define btnDOWN   menu::upCode
#define btnLEFT   menu::escCode
#define btnENTER  menu::enterCode
#define btnNONE   -1
...
FIELD(ConfigID,"Unit ID","",0,99,-10,-1,updateConfigID),
...
------

Not sure if I'm debugging correctly but the following in genericKeyboard.h continually sends -1 to the Serial while idea/no keys being pushed.

int available(void) {
    int ch=peek();
    Serial.println(ch);
    if (ch==-1) return 0;
    if (lastkey==-1) {
      lastkey=ch;
      pressMills=millis();
    } else if (ESCAPE_TIME&&millis()-pressMills>ESCAPE_TIME) return 1;
    if (ch==lastkey) return 0;
    return 1;
  }

neu-rah, 2015-11-06:
please try adding

    if (ch==-1) return 0;

right after the int ch=peek(); line

jamiekowalczik, 2015-11-06:
Up is down and down is up still when adding that to available().

int available(void) {
    int ch=peek();
    if (ch==-1) return 0;
    if (lastkey==-1) {
      lastkey=ch;
      pressMills=millis();
    } else if (ESCAPE_TIME&&millis()-pressMills>ESCAPE_TIME) return 1;
    if (ch==lastkey) return 0;
    return 1;
  }

Another random thing I"m seeing is that when I first start up the sketch, depending on how many menu items I have I will see Exit on the bottom.  I can't navigate to it and when I go to a menu and back out Exit is gone.  Is that a bug or me implementing incorrectly?

Lastly,
I attempted moving my 'update id' to a sub menu but when I have just one field I get the following error.  
 note: in definition of macro 'MENU'
     menu id (text,sizeof(id##_data)/sizeof(prompt*),id##_data);
                                                     ^
'setupMenu_data' was not declared in this scope

Adding in the OP makes the error go away (working example below).
MENU(setupMenu,"Setup",
   FIELD(ConfigID,"Unit ID","",0,99,-10,-1,updateConfigID),
   OP("Empty",emptyCmd)
);

neu-rah, 2015-11-06:
main menu should not have an Exit (unless specified) this must be a bug, checking...

neu-rah, 2015-11-06:
and a menu should not have a single option, makes no sense

jamiekowalczik, 2015-11-06:
Understandable about the single option.. Learning/designing as I go (backwards and bad practice I know) and only had one thing I can build into my garden project for the setup at the moment.  Added a stub for Set Date/Time and all is well with that.  Thanks again!

jamiekowalczik, 2015-11-06:
I apologize for the barrage of questions and issues.  Extremely new to Arduino and a coding is a hobby, I haven't had a lot of instruction as I'm sure you can see...  I stumbled on your project and really like it/it's extensibility.  Not sure if the following is easily done (or at least easily comprehended by me from the examples) but can you let me know if it is possible to have a menu displaying the following info but have the values (temp/humidity) updated regularly to the lcd?  I had the previous menu option call a function instead of this menu example which just redrew the info on the lcd repeatedly but then i lose menu navigation and some other annoyances.  Curious how you would go about accomplishing such a task with your library.

MENU(showSensorData,"Sensor Data",
   OP("Water Level: High",nothing),
   OP("Water Temp: 73 F",nothing),
   OP("Room Temp: 78 F",nothing),
   OP("Room Humidity: 32 %",nothing),
   OP("Status: Idle",nothing)
);

neu-rah, 2015-11-06:
some have used the menu along with other applications, you can use your LCD to display that info while the menu is suspended, however it seems that you want to use the navigation select the viewed info, right?
this is not implemented, you can use the FIELD thing to hook into your info but:
1 - you will have to call an option redraw for them to be updated (menu will flick)
2 - info should be readonly, so when clocking the menu system wil try to change the variable /that might be undesirable)
you are the second person to request this kind of stuff, but this time I can see advantages because of the multiline info..
got to check if I can update when pooling whithout storing old values and without redrawing on each poll, otherwise the lcd will flick or a lot of mem can be consumed (by storing old values).
in worst case i guess we can have an update interval to redraw all without redrawing every poll

jamiekowalczik, 2015-11-06:
Using the navigation would be key.  I have more than I can store on the screen now and building out that navigation per menu/submenu is what led me to you.  :)

I will look into using FIELD for now.  I don't mind a flick, its not ideal but will do for my purposes.  The data that I plan to have updated regularly would be read only.  Anything that would need to be written to would be done in another menu.  Ideally I would like to have the menu which displays data & updates regularly be the default menu and then with a button press bring me into the other navigation menu for defining set points, running actions, etc but if not easily done with this library then I can deal without. :)

neu-rah, 2015-11-06:
this might be of interest:
https://github.com/neu-rah/ArduinoMenu/issues/15
this discussion lead to implementation of redraw functions
also this on focus and redraw
https://github.com/neu-rah/ArduinoMenu/wiki/Customizing-menus

jamiekowalczik, 2015-11-06:
This seems to be doing what I want but when first jumping into the menu I see 'junk' at line[1].

As soon as the temperature changes it looks fine. 

MENU(showSensorData,"Sensor Data",
   OP("Water Level: High",nothing),
   OP("Water Temp: ",nothing),
   OP("Room Temp: 78 F",nothing),
   OP("Room Humidity: 32 %",nothing),
   OP("Status: Idle",nothing)
);

int intWaterTemp, intOldWaterTemp;
void readTemperature() {
  intWaterTemp = sensors.getTempFByIndex(0);
  if(intOldWaterTemp != intWaterTemp) {
    String strWaterTempPre = "Water Temp: ";
    String strWaterTemp = String(intWaterTemp);
    String strJoinedString = strWaterTempPre+strWaterTemp+" F";
    //Serial.println(strJoinedString);
    Serial.println("----");
    int str_len = strJoinedString.length() + 1;
    char charWaterTemp[str_len];
    strJoinedString.toCharArray(charWaterTemp, str_len);
    Serial.println(charWaterTemp);
    showSensorData.data[1]->text = charWaterTemp;
    intOldWaterTemp = intWaterTemp;
    showSensorData.redraw(menu_lcd,mykb);
  }
}

I'm trying to figure out how to update it only after I'm on the menu (i think that is why it is appearing as junk?) but I can't figure out how to determine if it is the selected menu.

void loop() {
  mainMenu.poll(menu_lcd,mykb);
  if ((menu*)menuNode::activeMenu==showSensorData) {
    readTemperature();
  }
}

Error:
I2C_menu:361: error: 'activeMenu' is not a member of 'menuNode'
'activeMenu' is not a member of 'menuNode'

jamiekowalczik, 2015-11-07:
I think I figured it out.  I'm using the text of the current node to determine which menu I'm on.  Works well now.  Any idea how I can prevent Exit from appearing on my main menu?

String curMenu = ((menu*)menuNode::activeNode)->text;
  if (curMenu == "Sensor Data") {
    if(myThread.shouldRun())
      myThread.run();
  }

neu-rah, 2015-11-10:
about the swap of up/down... it happens because it was coded with an encoder in mind where we think as left/right (anti-clockwise/clockwise), then buttons end up conceptually swapped, I need a flag or some way to toggle this case.
https://github.com/neu-rah/ArduinoMenu/issues/15 talks about determining current position

jamiekowalczik, 2015-11-10:
Ok, that makes sense.  Until I can wrap my head around the code and understand it better, I took your earlier recommendation of using negative numbers for the Field step and swapping up/down.  That seems to be working well for my purposes at the moment and to get around the Exit on the main menu when the sketch is first uploaded, that seems to go away when I have 4 items to display so I put the current time on the top and now that is behind me.  My struggle now is with button press events being missed when I'm on a menu that is polling the sensors.  I'm trying to dig in to using interrupts/timers/fake threading and that has been a bit of a learning curve.  Having fun with this Arduino stuff, I can't thank you enough for your menu and continued support as it has allowed me to build out this idea quicker than I assumed.

#define btnRIGHT  menu::enterCode
#define btnUP     menu::downCode
#define btnDOWN   menu::upCode
#define btnLEFT   menu::escCode
#define btnENTER  menu::enterCode

MENU(setDateTimeMenu,"Set Date/Time",
   FIELD(bHour,"Hour","",0,23,-10,-1,updateRTC),
   FIELD(bMinute,"Minute","",0,59,-10,-1,updateRTC),
   FIELD(bMonth,"Month","",1,12,-10,-1,updateRTC),
   FIELD(bDay,"Day","",1,31,-10,-1,updateRTC),
   FIELD(iYear,"Year","",2015,3015,-10,-1,updateRTC)
);

MENU(setupMenu,"Setup",
   FIELD(ConfigID,"Unit ID","",0,99,-10,-1,updateConfigID),
   SUBMENU(setDateTimeMenu)
);

current code as a work in progress..
https://github.com/jamiekowalczik/sketches/blob/master/i2c_menu

TerryKing, 2017-01-04:
The example here and Jamie's code got me through making the first menus for my Kiln Control system. This is VERY useful!  I will write an example on http://ArduinoInfo.Info
Regards, Terry King
...In The Woods in Vermont, USA
terry@yourduino.com

bidoowee, 2017-07-03:
I've been looking for the genericKeyboard.h discussed in this thread (in order to implement an analog keypad) but I don't seem to be able to find it in the GitHub repository. Has this been superseded by another header file? 

many thanks,

bidoowee

neu-rah, 2017-07-04:
this thread is from version 2.x, files are different on 3.x

what is the driver that you use for your keypad now?

bidoowee, 2017-07-04:
Many thanks for the speedy response. I don't use a driver. It is just an analog voltage divider - the same dfrobot shield that Terry is using. I had a go at integrating the same code I use (without issue) with the keypad elsewhere into the codeCtl example but there seems to be interference with the serial communication (possibly because of nav.poll() / nav.doOutput()?). lcd.print() does not behave as expected. In any case, I will likely eventually want to concatenate the stream from the analog keyboard with another one from a rotary encoder. Help would be greatly appreciated.

neu-rah, 2017-07-11:
on version 3 there is the possibility of code control, see example https://github.com/neu-rah/ArduinoMenu/blob/master/examples/codeCtrl/codeCtrl/codeCtrl.ino

this example has also some extra code to make software debounce of keys. it can be omitted if using hardware debounce.

the idea is to cal lthe menu with Up/Down/Enter/Esc commands

navCmds are described on the wiki here https://github.com/neu-rah/ArduinoMenu/wiki/Navigation#navcmd-structure

TerryKing, 2017-07-11:
Hi,
I have working 5-button[ inside the code here:](https://arduino-info.wikispaces.com/Project-GlassKilnController-Code)

bidoowee, 2017-07-11:
Many thanks for the responses. 
Terry, I presume that you are using version 2.x of the menu library as per neu-rah's message from 7 days ago. It was your code example that brought me here in the first place :).  
In the interim however, I have implemented another menu library.
Thanks again.

closed 2017-07-12 by neu-rah

ironlungx, 2024-04-07:
Are 5-buttons really required, or can I make do with 3. I couldn't find any use-case(s) for the left & right button