neu-rah/ArduinoMenu #19: Arduino Menu for ATmega644P

erocircuits, 2016-03-12:
HI

i have tried with atmega2560. It is working fine.

But when i am trying with ATmega644A encoder is not working. It complains that  "uC PCINT REVERSE MAP IS NOT DEFINED".

i have checked the pcint.h file and there are only two micros 328p and mega2560.
How do i add ATmega644P micro?

Please help.

erocircuits, 2016-03-14:
hi there... any help

pepadew, 2016-03-14:
good luck getting any help from the maker, if you could post your code/sketch, maybe others find this issue and help that way.
also i could walk through your sketch and maybe help/spot errors

neu-rah, 2016-04-14:
yes that map is not defined...
those definitions are on pcint.h file
because interrupts are very specific to uC, the have to be added

in example, the existing ones
	#if ( defined(__AVR_ATmega328__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega16u4__) )
		const uint8_t PROGMEM pcintPinMap[3][8]={{8,9,10,11,12,13,-1,-1},{A0,A1,A2,A3,A4,A5,-1,-1},{0,1,2,3,4,5,6,7}};
	#elif ( defined(__AVR_ATmega2560__) )
		const uint8_t PROGMEM pcintPinMap[3][8]={{53,52,51,50,10,11,12,13},{0,15,14,-1,-1,-1,-1,-1},{A8,A9,A10,A11,A12,A13,A14,A15}};
	#else
		#error "uC PCINT REVERSE MAP IS NOT DEFINED"
	#endif

i've not checked but they migh be the same as inb the 2560, if its the case then just add the uC id with a || (or clause) as above

if its the case you can contribute to the library and make it available for other users

closed 2016-06-29 by neu-rah