neu-rah/ArduinoMenu #23: encoder issue

jamiekowalczik, 2016-05-14:
I have an encoder / sketch working on an Arduino Nano but can't get it to work on an Arduino Mega.  Are there any known issues with that hardware or maybe I'm just overlooking something obvious.  I am using pins 2[clk],3[dt],4[sw] for both pieces of hardware.

neu-rah, 2016-05-15:
the encoder needs interrupts as defined in pcint.h, the current pcint.h only maps pins in 
{53,52,51,50,10,11,12,13},{0,15,14,-1,-1,-1,-1,-1},{A8,A9,A10,A11,A12,A13,A14,A15}

jamiekowalczik, 2016-05-15:
O boy.  That simple.  Thank you!  I had a feeling changes needed to be in that file but I didn't(still really don't) quite understand how to change it.  I was under the impression that I should be using pins that allow for external interrupts which for the 2560 were 2,3[same as the Nano],18,19,20,21.  I'm using 20 and 21 for my 4x20 lcd display so I hoped to use any of the remaining 4 for my encoder.  I simply changed the pins in use to 10,11,12 and all is well.  As always, thank you so much!

closed 2016-05-15 by jamiekowalczik