neu-rah/ArduinoMenu #24: Thanks and two questions

emanuel001, 2016-05-28:
Hi. really cool menu :)
took me a while but managed to figure it out. I think it would be easier for people if you could make your examples a bit more complex. 
By this I mean: 
1) add some submenu (1 level or even 2 levels down) just so people can figure out easily how to construct complex menu. 
2) the toggle part was nice, but it would be great if you would add also some function calling, not only variable changing. for a beginner like me it would be easier to figure out everything. I am not a programmer, but an el. engineer. I like to tinker. i know how to program at a decent level (nothing fancy) but my lack of experience, makes it very hard for me to implement stuff, and it's easier just to construct by copy-pasting. 

Now my questions: 

1)  Could you maybe make the menu to have also a background/screensaver (i have no idea what to call it). Similar to what you will find on a 3d printer. 
a screen where you can see stats (whatever people want) and you first have to press select/enter to go into the actual menu. and when the menu times out, or you press exit, to go back to that stats screen. 
would this be possible?

2) would it be possible to add a time out function. if you don't touch anything for 30 sec/1 min the menu goes back to the main page?

Once again thank you very much.

emanuel001, 2016-05-28:
I've played some more with the code, so now I figured out that I can stop pooling the menu, and then use the LCD for something else. 
I'm now conditioning the pooling in the main loop though a bool. I created an exit button in the main menu to clear it. but i would like to use the encoder button to detect when it is pressed and activate back the pooling, but at the same time to not register that encoder click as an enter in the menu.. I tried for the past 2 hours to figure out how the streaming works, but no luck:(
thank you for your help.

neu-rah, 2016-06-06:
you can build you own fake stream for the encoder, it acts like an encoder driver, see the quadEncoder.h how it is implemented, by making a "new driver" you can catch keys when menu is not active

for now its the resource available, in the future i might consider a deactivation flag to ignore pooling as well as input... did not check if that can work yet, its just a thought

neu-rah, 2016-06-28:
encoder button stream is still readable outside the menu loop, ex:

    allIn.read()

where allIn is the keyboard stream efined for the encoder button

closed 2016-06-28 by neu-rah