neu-rah/ArduinoMenu #25: U8Glib and interactivity question

christophepersoz, 2016-06-28:
Hello,

I plan to use you fantastic menu library for an another project and I wonder if it is adaptable to use U8Glib graphic library instead of Adafruit library ?
Also, do you think that the menu behavior can act as an all interactive options available — I mean changing directly the values without changing the content of the display. 

Thanks a lot !
Christophe

neu-rah, 2016-06-28:
Hi!

good questions! Yes the menu can be adapted to  virtually any input or output.
There are (included) some drivers for TFT and even touch screens, they are implemented by derivind the base stream class. Take a look on those drivers to get inspired (in ex: https://github.com/neu-rah/ArduinoMenu/blob/master/menuUTFT.h).
On that file class menuUTFT derives/extends the base menuOut and implements some functions to work with the specific screen (UTFT), its a 130 lines files, that should be easy.
the derived class must fill out some values for character size and screen size, how to set colors and how to clear a line and the screen and some other stuff that the menu system needs to call for the screen.
Eventually you might even derive from menuUTFT if your display is similar enough.
After making a derived class for U8Glib you can use it on the menu system.
Please send me a link of the U8Glib if you have any difficulties.

christophepersoz, 2016-06-28:
Thanks for you quick reply !

I'll have a look on the example you mentioned. Do you think I can derives the display to have the following behaviors:

• Submenu acting as pages
• All dynamic values (byte mostly) on a page can be change without changing the structure of the page.
• Passing from one value to the next one is done using the encoder or usual entry device (buttons or anything else)
• Enum values switching between them until the good one.
Validation of a value is done by clicking on the encoder button (usual behavior on encoder)
Switching between pages is done by a long press on encoder button followed by rotation to select the right page.
I plan to use ClickEncoder library that include long press, double click and so on. I think it will be ok to include it.
For U8glib, that's an another problem, but I think that will be ok so. If I'm good with it I'll provide you an example using it. 
Note that in top of that I plan to run your lib on Teensy 3.2 instead of Arduino. Maybe I will avoid the use of Macros. 
I hope it will work !

christophepersoz, 2016-07-15:
Hello !

I'm trying to make MENU compatible with U8Glib, but I'm still stuck at this time. I'm attaching to this post my current dev, and portability tests. I use menuGFX.h as an example and create my own menuU8G.h (placed in Menu folder). At this time the build and upload works properly, the Serial output also but nothing belonging to Menu appears on the display. Only the title placed in the draw() function do appears, (that's allow me to be sure that the display is properly running).

So I place in this post the files I modified and created for the tests. Please note the I'm running it on Teensy board 3.2 instead of Arduino, but it should works on Arduino too.

Just in case, here the link for the U8Glib library : [https://github.com/olikraus/u8glib](https://github.com/olikraus/u8glib).

[menuU8G.h.zip](https://github.com/neu-rah/ArduinoMenu/files/366119/menuU8G.h.zip)
[MOCO_menuWithU8Glib.ino.zip](https://github.com/neu-rah/ArduinoMenu/files/366134/MOCO_menuWithU8Glib.ino.zip)

If you can have a look and help me to solve what's going wrong that would be nice !
Thanks !

neu-rah, 2016-07-15:
you can test your menuU8G menuOut driver by calling the driver functions directly (without even calling the menu poll)
stating by setCursor and write... then clearLine and clear.

// No need to clear, the U8Glib display loop clear screen on each refresh
??? it does? usually this results in flicking... be sure that the printing will be on display enough time between clears in order to be seen, i would avoid this cycle cleaning because the menusystem is designed to minimize redraws (avoid flicking and delays) and this successive cleaning invalidates that.

also assuming that U8Glib setcursor and print work on pixel coordinates (should be so on a gfx display), on text displays usually we set the font size to 1x1

please let me know the result so that we can look for something else...

christophepersoz, 2016-07-15:
Thanks for the tips. I tried and indeed I can print text on display using the drivers fonctions:

`
draw(){
   gfx.setCursor(2,2);
  gfx.write(123);
}

void loop() {
  // picture loop
  _U8G.firstPage();
  do {
    draw();
    //mainMenu.poll(menu_out,Serial);
    delay(550);
  } while( _U8G.nextPage() );
}
`
In the meantime, as soon as I'm calling `mainMenu.poll(gfx,Serial);` I'm just getting few lines of pixels (3 to 5 lines I think) for few seconds and then no more. I tried to change resX and resY from the size of the font size (6x9) to (1x1) but that does not change anything and the pixels displayed remains the same.
If I remove the delay(550) in the loop the refresh is so fast that I merely can't see the pixels.

Maybe all the lines of the menu are condensed into few lines of pixels.

_I don't know how U8Glib is done, but using HW SPI for display on Teensy you can easily have a frame rate of 50 to 90fps, so no, there is no flickering ;)_

christophepersoz, 2016-07-16:
I found a couple a issues and correct some stuff to have a line properly displayed on screen now.
The main issue at this time is that I need to rewrite the write() function to have the Progmem strings correctly displayed - if you have a good example or so ;)
After that I could maybe understand why the display clear itself after few seconds.

neu-rah, 2016-07-16:
welp progmem is one of my wanted/need help features, guess you can try with normal mem just for testing (maybe on a smaller menu)
about the screen being blank... the menu system only draws things if something changed. ie: selection or value, otherwise nothing is printed (printing all the menu every time reveals to be damn slow on some screens), however U8Glib uses that paging thing and clears it all... therefor nothing is shown after a while (if so then changing selection will flick something briefly i guess).
you can force the menu to draw every poll by calling <your_menuOut>.redraw() just before the <menu>.poll

christophepersoz, 2016-07-17:
Hi !
I'm finally get the display works and displaying items, selection overlay, etc. So cool ! (the redraw tips works well at this time, I have to see if I can override this).
I need to fine-tune some stuff to get the job done perfectly, but it works.
At this time I still have no stream so I'm using Serial Monitor to select items and change values.
Somehow, I still have some strange behaviors, for example :
• When I'm selecting a dynamic value, the display is cleared until I have enter a value, as soon as it is done I can see the new value displayed
• It seems that the menu property : setPosition(ox,oy); is not passed to menuOut, so I can't move the menu on the screen.
• Finally, how can I add a Exit items by default on Menu declaration and have it shown and working ?
I find how to display it (`mainMenu.poll(gfx,Serial, true`); instead of `mainMenu.poll(gfx,Serial);`) but I can't have it working. Can I declare the function related to the Exit ?
Thanks !

neu-rah, 2016-07-17:
> • Finally, how can I add a Exit items by default on Menu declaration and have it shown and working ?
> I find how to display it (mainMenu.poll(gfx,Serial, true); instead of mainMenu.poll(gfx,Serial);) but I can't have it working. Can I declare the function related to the Exit ?

yes, that "canExit" parameter is a fossil from before non-bocking, need some clean.
you can suspend the menu by not calling **.poll**, so you can either use a bool var for that and a toggle menu option as "Exit" to control it... or a regular option to toggle the poll control and clean the screen if needed.

of coz a way of reentering the menu will be needed. on reentry a redraw must be signaled.

christophepersoz, 2016-07-17:
You mean test the var canExit ? like this :

` if (!mainMenu.canExit)
     mainMenu.pool();`

 Otherwise sorry but I don't understand where the boolean resulting of Exit action is stored. Can you provide me an example ?

neu-rah, 2016-07-20:
the relevant code part for suspend/resume a menu.

```c++
bool runMenu=false;

void pauseMenu(prompt &p, menuOut &o, Stream &i) {
  runMenu=false;
  o.clear();
}

MENU(mainMenu,"Main menu",
  SUBMENU(onoff_tog),
  FIELD(param,"Name","%",0,100,10,1),
  OP("Nothing",nothing),
  OP("Exit",pauseMenu)
);

void loop() {
  if (runMenu) mainMenu.poll(lcd,enc);
  else if (enc.read()==menu::enterCode) {//menu reenter method
    lcd.redraw();
    runMenu=true;
  } else {
    lcd1.setCursor(0,1);
    lcd1.print(millis()%1000);
  }
}
```

we can use similar feature to run multiple menus, provided that we have a method of entering for each.

this is the kind of thing that I might include into the poll, because several people are asking for this, provided that i manage to leave the possibility of multiple menus.

would it be usefull?

christophepersoz, 2016-07-20:
Really useful ! Thanks, I will try that after debugging the issue I have with MenuU8G.h when I'm setting a dynamic parameter (i.e. integer). At this time, when I'm editing a value the screen become blank (almost) and I can't see the value changing. Once validated, everything goes back to normal state.
In which method and file the print of an edited value is done ? printPrompt() in the graphic lib or directly inside menu.h ?

neu-rah, 2016-07-20:
https://github.com/neu-rah/ArduinoMenu/blob/master/menuFields.h#L58

christophepersoz, 2016-07-21:
After analyze it seems that the problem does not come from menuFields.h. In fact printTo() do only a print the the display device (Serial, display, etc). What I'm seeing is too many refresh or erase/refresh when I'm editing a value. I tried to remove some clearing options or redraw but it does not work. On the best case the problem remain the same, or the menu disappear until the end of the editing. But in the meantime, the toggle value works perfectly.

Here a small video showing the problem.
[IMG_1125_converted.mp4.zip](https://github.com/neu-rah/ArduinoMenu/files/376319/IMG_1125_converted.mp4.zip)

I'm continuing to try to understand what happens, but if you have an idea or a track let me know.
Thanks !

neu-rah, 2016-07-21:
i was wiring a PDC8544 to track this down, its supported by g8glib...
guess it will envolve generalized modifications and eventually sync with draws... just guessing

christophepersoz, 2016-07-21:
Oh, great ! I can send you my current files and lib which includes some other enhancements like correct setPosition() functionality which works correctly now. Let me know.

neu-rah, 2016-07-21:
yes please

christophepersoz, 2016-07-23:
Here they are ! Sorry for the delay I was away for work.
Here the current state of my dev and all the mods I made in libraries. I hope it will help.

[MenuWithU8Glib.zip](https://github.com/neu-rah/ArduinoMenu/files/379466/MenuWithU8Glib.zip)
[Libraries.zip](https://github.com/neu-rah/ArduinoMenu/files/379467/Libraries.zip)

neu-rah, 2016-07-28:
thanks all for the help, I've included U8Glib and ClickEncoder + examples, please report bugs

closed 2016-07-28 by neu-rah