neu-rah/ArduinoMenu #26: bug on graphics on long menus

schroejahr2, 2016-07-02:
Hi, i just used your great lib to do some menu actions.

However i noticed a -maybe- bug in your lib. 

i have some long menus, with different options. And as you see on the screenshot there is a displaying bug at "input 6". The appeareance of this bug differs if i change the text in this line, but it is laways present.

Any ideas ? 

screenshot: [https://pl.vc/pt8go](url)

code (to reproduce the bug, not the whole project, only the menu code): 

`/***************************************
Arduino generic menu system
GFX menu - unsing adafruit gfx screens
tested on ATmega2560
http://www.r-site.net/?at=//op%5B%40id=%273090%27%5D

Sept. 2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

creative commons license 3.0: Attribution-ShareAlike CC BY-SA
This software is furnished "as is", without technical support, and with no 
warranty, express or implied, as to its usefulness for any purpose.

Thread Safe: No
Extensible: Yes

Adafruit GFX library from:
  https://learn.adafruit.com/adafruit-gfx-graphics-library/overview
Display ST7735 driver
  http://www.adafruit.com/product/358
***/
/******************
Rui Azevedo 2014
ruihfazevedo@rrob@gmail.com
********/
#include <SPI.h>
#include <menu.h>//menu macros and objects
#include "menuFields.h"
//#include <pcint.h>//this is incompatible with software serial (arduino needs an handler!)
#include <quadEncoder.h>//quadrature encoder driver and fake stream
#include <keyStream.h>//keyboard driver and fake stream (for the encoder button)
#include <chainStream.h>// concatenate multiple input streams (this allows adding a button to the encoder)
//#include <menuPrint.h>//Print (Serial) menu
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library
#include <menuGFX.h>

#if defined(__AVR_ATmega2560__)
  ///////////////////////////////////////////////////////////////////////////
  //TFT + SD
  //#define sdCS  9//not using SD card
  #define tftCS 10
  #define dc    8
  #define rst   -1
  ////////////////////////////////////////////
  // ENCODER (aka rotary switch) PINS
  #define encA 10
  #define encB 12
  #define encBtn 13
#elif defined(__AVR_ATmega328__) || defined(__AVR_ATmega328P__)
  ///////////////////////////////////////////////////////////////////////////
  //TFT + SD
  //#define sdCS  9//not using SD card
  #define tftCS 10
  #define dc    8
  #define rst   -1
  ////////////////////////////////////////////
  // ENCODER (aka rotary switch) PINS
  #define encA 2
[... 407 lines cut ...]
  //size is within screen limits even after rotation
  //this limits are not constrained, please ensure your text fits
  gfx.maxX=24;
  gfx.maxY=16;
  gfx.bgColor=BLACK;

  pinMode(encBtn, INPUT);
  digitalWrite(encBtn,1);
  
  encoder.begin();

  delay(300);
  tft.fillScreen(BLACK);
}

///////////////////////////////////////////////////////////////////////////////
// testing the menu system
void loop() {
  mainMenu.poll(gfx,allIn);
}
`

I am using an rotary encoder, arduino uno and teh adafruit 1.8 breakout board.

Have a nice day !

neu-rah, 2016-07-21:
just to check  if its a lib bug or memory problem, can you reduce the menu a bit?
lets say by commenting out  some parts like LOG, LCD, TIME

thanks

neu-rah, 2016-08-25:
version 2.4 uses flash memory, that might help on long menus

closed 2016-09-21 by neu-rah