neu-rah/ArduinoMenu #27: Use of a different Encoder library

christophepersoz, 2016-07-08:
Hello,

Do you think ids that possible to use an another Encoder library (https://github.com/0xPIT/encoder) with ArduinoMenu ? The library mentioned have some behaviors that are interesting for my project, but I'm not sure I can map it to your Menu Library and take advantage of Acceleration, Button Click, Held, Double Click, etc.
If you can have a look, that could be great !
Thanks a lot.

neu-rah, 2016-07-16:
the "encoder" must be a Stream (see quadEncoder.h)
and the stream must give characters + and - depending on the rotation direction
please note that the actual characters used are defined class menu (you can customize it to be something else)
the menu system acceps Streams as inputs so if you can put your encoder working as a strem it will do

christophepersoz, 2016-07-18:
Hello !

I made a stream library based on the quadEncoder.h and I can have have it running. It streams - and + when rotating CW or CCW. But, when I'm connecting it to the menu, I'm getting something weird. As soon as a stream value is sent, just after I'm getting a kind of reinit on the display to the initial position, like if the cursor was reseted. Have a look on the video.
I tried for nearly 2 hours to trace some values in menu.h, chainStream.h, keyStream.h but I never found in which part of the code this reset is done and if the -1 returned by the stream when not active can make the display position reseted ?
Do you have any idea where I can have a look and make some test ?

Here the stream I have created.

[ClickEncoderStream.h.zip](https://github.com/neu-rah/ArduinoMenu/files/368542/ClickEncoderStream.h.zip)

And the Ino file that I'm running.

[menuWithU8Glib.ino.zip](https://github.com/neu-rah/ArduinoMenu/files/368545/menuWithU8Glib.ino.zip)

Thanks !

TomaTLAB, 2016-07-18:
Christopher, look at https://github.com/TomaTLAB/ArduinoMenu/blob/master/quadEnc.h
This is a simple poll driver for encoder, maybe helpfull.

christophepersoz, 2016-07-18:
Thanks TomaLAB,

But the problem is not the reading of the quad encoder rotation, which works perfectly, but much more the Stream sent to Menu. It seems that the Stream contains a return to original position as soon as I'm rotating the encoder. So the movement is correct, but as soon as the Menu reach the good position it's going back the position 0. Weird because the encoder does not sent any value when it is not moving.

neu-rah, 2016-07-20:
got it working with this code (rotation and button click)

```c++
//
//  ClickEncoderStream.h
//  
//
//  Created by Christophe Persoz on 17/07/2016.
//  Copyright © 2016 Christophe Persoz. All rights reserved.
//

#ifndef ClickEncoderStream_h
#define ClickEncoderStream_h

#include <stdint.h>
#include <avr/io.h>
#include <avr/interrupt.h>
#include <avr/pgmspace.h>
#include "Arduino.h"
#include "ClickEncoder.h"

/*  emulate a stream based on clickEncoder movement returning +/- for every 'sensivity' steps
 buffer not needer because we have an accumulator
 */
class ClickEncoderStream:public Stream {
public:
    ClickEncoder &enc; //associated hardware clickEncoder
    int8_t sensivity;
    int oldPos;
    int pos;
    ClickEncoder::Button btn;

    inline void update() {
      pos+=enc.getValue();
      if (btn==ClickEncoder::Open) btn=enc.getButton();
    }
    ClickEncoderStream(ClickEncoder &enc,int sensivity)
      :enc(enc),
      pos(0),
      oldPos(0),
      btn(ClickEncoder::Open),
      sensivity(sensivity) {}
    
    inline void setSensivity(int s) {sensivity = s;}
    int available(void) {return peek()!=-1;}
    
    int peek(void) {
        update();
        if (btn==ClickEncoder::Clicked) return menu::enterCode;
        int d = pos - oldPos;
        if (d <= -sensivity)
            return menu::downCode;
        if (d >= sensivity)
            return menu::upCode;
        return -1;
    }
    
    int read()
    {
        int ch=peek();
        btn=ClickEncoder::Open;
        if(ch==menu::upCode) oldPos += sensivity;
        else if(ch==menu::downCode) oldPos -= sensivity;
        return ch;
    }
    
    void flush() {
        update();
        oldPos = pos;//encValue;
    }
    size_t write(uint8_t v)
    {
        oldPos = v;
        return 1;
    }
};

#endif /* ClickEncoderStream_h */
```

enhancements: can still add doubleclick or something as escape

christophepersoz, 2016-07-20:
Thanks a lot for the adds and the corrections ;)
I did made some test and the line `pos += enc.getValue();` is not working. The pos value remain to 0 when I print it into Serial Monitor. To have the position updates, I had to initialise the the enc.getValue() in the constructor of the stream inside ClickEncoderStream.h.

```c++
/**************
 
 ClickEncoderStream.h, created to work with the library ClickEncoder.h
 
 Modified by Christophe Persoz and Rui Azevedo.
 Based on keyStream.h in Arduino Menu library developed by Rui Azevado,
 and ClickEncoder library by Peter Dannegger.
 
 July 2016 Christophe Persoz - christophepersoz(@rrob@)gmail.com
 Sept. 2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
 creative commons license 3.0: Attribution-ShareAlike CC BY-SA
 This software is furnished "as is", without technical support, and with no
 warranty, express or implied, as to its usefulness for any purpose.
 
 Thread Safe: No
 Extendable: Yes
 
 quick and dirty keyboard driver
 metaprog keyboard driver where N is the number of keys
 all keys are expected to be a pin (buttons)
 we can have reverse logic (pull-ups) by entering negative pin numbers
 ex: -A0 means: pin A0 normally high, low when button pushed (reverse logic)
 ***/

#ifndef __ClickEncoderStream_h__
#define __ClickEncoderStream_h__

#ifdef defined(ARDUINO) // Arduino 1.0 and 1.5 specific
#include <stdint.h>
#include <avr/io.h>
#include <avr/interrupt.h>
#include <avr/pgmspace.h>
#endif

#include <Arduino.h>
#include "ClickEncoder.h"

/*  emulate a stream based on clickEncoder movement returning +/- for every 'sensivity' steps
 buffer not needer because we have an accumulator
 */
class ClickEncoderStream:public Stream {
public:
    ClickEncoder &enc; //associated hardware clickEncoder
    int8_t sensivity;
    int oldPos;
    int pos;
    ClickEncoder::Button btn;
    
    inline void update() {
        pos += enc.getValue();
        
        if (btn == ClickEncoder::Open)
            btn = enc.getButton();
    }
    
    ClickEncoderStream(ClickEncoder &enc,int sensivity)
    :enc(enc),
    pos(0),
    oldPos(0),
    btn(ClickEncoder::Open),
    sensivity(sensivity) {
        pos = enc.getValue();
    }

    
    inline void setSensivity(int s) {
        sensivity = s;
    }
    
    int available(void) {
        return peek() != -1;
    }
    
    int peek(void) {
        update();
        
        if (btn == ClickEncoder::Clicked)
            return menu::enterCode;
        
        if (btn == ClickEncoder::DoubleClicked)
            return menu::escCode;

        int d = pos - oldPos;
        if (d <= -sensivity)
            return menu::downCode;
        if (d >= sensivity)
            return menu::upCode;
        return -1;
    }
    
    int read()
    {
        int ch = peek();
        btn = ClickEncoder::Open;
        if (ch == menu::upCode)
            oldPos += sensivity;
        else if (ch == menu::downCode)
            oldPos -= sensivity;
        return ch;
    }
    
    void flush() {
        update();
        oldPos = pos;
    }
    
    size_t write(uint8_t v) {
        oldPos = v;
        return 1;
    }
};

#endif /* ClickEncoderStream_h */
```

Otherwise I simplified the keyMap and the Stream init in the .ino, they are useless now the button is directly managed in the Stream.

LuckyLuzz, 2016-07-20:
Hey, how would the implemtation to the main code be like? It's pretty interessting cause it look's like the default code is missing rotation steps.

[https://github.com/neu-rah/ArduinoMenu/wiki/Encoder](url)

neu-rah, 2016-07-20:
```c++
#include <Arduino.h>

/********************
Arduino generic menu system
with U8Glib for Teensy and Arduino boards
http://www.r-site.net/?at=//op%5B%40id=%273090%27%5D

Dec.2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
creative commons license 3.0: Attribution-ShareAlike CC BY-SA
This software is furnished "as is", without technical support, and with no
warranty, express or implied, as to its usefulness for any purpose.

Thread Safe: No
Extensible: Yes

UTFT library from:
  http://henningkarlsen.com/electronics/library.php?id=50
UTouch library from:
  http://henningkarlsen.com/electronics/library.php?id=56
*/

//#include <SPI.h>          // SPI transaction library
//#include <U8glib.h>       // OLED display U8Glib
//include <U8g_teensy.h>  // OLED display driver u8g_teensy
//#include <ClickEncoder.h> // Quad encoder
#include <TimerOne.h>     // ISR on Quad encoder

//////////////////////////////////////////////////////////////////////////
// LCD
#define RS 8
#define RW A5
#define EN 3

#include <LiquidCrystal.h>
#include <menuLCD.h>
LiquidCrystal lcd1(RS, RW, EN, 4, 5, 6, 7);
//describing a menu output, alternatives so far are Serial or LiquidCrystal LCD
menuLCD lcd(lcd1,16,2);

#include <menu.h>
//#include <menuU8G.h> // Core graphics library adapted to work with U8Glib
//#include <quadEncoder.h>//quadrature encoder driver and fake stream
//#include <keyStream.h>//keyboard driver and fake stream (for the encoder button)
//#include <chainStream.h>// concatenate multiple input streams (this allows adding a button to the encoder)
#include <menuFields.h>
#include <menuPrint.h>//Print (Serial) menu

/*#define OLED_CS     10
#define OLED_A0     9
#define OLED_RST    8*/

#define CK_ENC  2 // Quand encoder on an ISR capable input
#define DT_ENC  A2
#define SW_ENC  A3

//U8GLIB_NHD31OLED_GR _U8G(OLED_CS, OLED_A0, OLED_RST);  // HW SPI Com: CS = 10, A0/SDIN = 9, [RESET = 8]
//U8GLIB _U8G(&u8g_dev_ssd1322_nhd31oled_2x_gr_hw_spi, u8g_com_hw_spi_fn); // HW SPI Com: CS = 10, A0 = 9 (Hardware Pins are  SCK = 13 and MOSI = 11)

#include "ClickEncoderStream.h"
ClickEncoder qEnc(DT_ENC, CK_ENC, SW_ENC, 4, LOW);

///////////////////////////////////////////////////////////////////////////
//functions to wire as menu actions

//aux function
void nothing() {}

/////////////////////////////////////////////////////////////////////////
// MENU DEFINITION WITH MACROS
boolean test=false;
int param;

TOGGLE(test,onoff_tog,"led: ",
  VALUE("On",HIGH),
  VALUE("Off",LOW)
);

MENU(mainMenu,"Main menu",
  SUBMENU(onoff_tog),
  FIELD(param,"Name","%",0,100,10,1),
  OP("Nothing",nothing)
);

ClickEncoderStream enc(qEnc,1);
void timerIsr() {qEnc.service();}

menuPrint menu_out(Serial);//describe output device

void setup() {
  Serial.begin(115200);
  while(!Serial) {}

  //Serial Print
  Serial.print("Menu system test");

  lcd1.begin(16,2);
  lcd1.print("Menu test");

  // Encoder init
  qEnc.setAccelerationEnabled(true);
  qEnc.setDoubleClickEnabled(true);
  Timer1.initialize(1000);
  Timer1.attachInterrupt(timerIsr);
}

void loop() {
  mainMenu.poll(lcd,enc);
}
```

neu-rah, 2016-07-20:
can you fork and submit your changes? @christophepersoz

christophepersoz, 2016-07-20:
I can't test that configuration because I do not have a LCD here for testing, but I will fork and put an example that should work with this extension ;)

neu-rah, 2016-07-20:
i can work on the example as I have the code here, provided that you add your "ClickEncoderStream.h" file to the library folder

LuckyLuzz, 2016-07-20:
Hi, changed the code to an I2C LCD, but unforgently it is not working :(

Before with default Rotarty code it was working.

`/********************
Arduino generic menu system
LCD menu - unsing arduino classic LCD library
http://www.r-site.net/?at=//op%5B%40id=%273090%27%5D

Sept.2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
creative commons license 3.0: Attribution-ShareAlike CC BY-SA
This software is furnished "as is", without technical support, and with no
warranty, express or implied, as to its usefulness for any purpose.

Thread Safe: No
Extensible: Yes

*/
#include <Wire.h> 
#include <LiquidCrystal_I2C.h>
#include <menu.h>//menu macros and objects
#include <menuLCDs.h>//F. Malpartida LCD's
#include <menuFields.h>
#include "ClickEncoderStream.h"

#define LEDPIN 13

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);
menuLCD menu_lcd(lcd,16,2);
////////////////////////////////////////////
// ENCODER (aka rotary switch) PINS
// rotary
#define CK_ENC  2 // Quand encoder on an ISR capable input
#define DT_ENC  3
#define SW_ENC  4
ClickEncoder qEnc(DT_ENC, CK_ENC, SW_ENC);
ClickEncoderStream enc(qEnc,1);

void timerIsr() {qEnc.service();}
///////////////////////////////////////////////////////////////////////////
//functions to wire as menu actions

//aux function
void nothing() {}

void ledOn() {digitalWrite(LEDPIN,1);}
void ledOff() {digitalWrite(LEDPIN,0);}

/////////////////////////////////////////////////////////////////////////
int frequency = 50;
// MENU DEFINITION
// here we define the menu structure and wire actions functions to it
// empty options are just for scroll testing
MENU(mainMenu,"LED ON/OFF",
  OP("LED On",ledOn),
  OP("LED Off",ledOff),
  OP("Empty 1"),
  OP("Empty 2"),
  OP("Empty 3"),
  OP("Empty 4"),
  OP("Empty 5..."),
  FIELD(frequency,"Freq","Hz",0,100,1,0)
 );

/////////////////////////////////////////////////////////////////////////
void setup() {
  Serial.begin(9600);
  Serial.println("menu system test");
  Wire.begin();
  lcd.begin(16,2);
  lcd.print("Menu test");
  delay(2000);
}

///////////////////////////////////////////////////////////////////////////////
// testing the menu system
void loop() {
      mainMenu.poll(menu_lcd,enc);
      }`

christophepersoz, 2016-07-20:
I did one ;)
I put a fork of ArduinoMenu (example) and ClickEncoder (ClickEncoderStream.h) on my page, is that enough ?

neu-rah, 2016-07-20:
do a pull request

christophepersoz, 2016-07-20:
Done !

LuckyLuzz, some includes are missing in order to work. <ClickEncoder.h>, then <TimerOne.h>, at least.

LuckyLuzz, 2016-07-20:
I am working on it ;) Sometimes I should take a bit more time.

LuckyLuzz, 2016-07-20:
Hope you guys get this to work! Would be awesome! :)

neu-rah, 2016-07-20:
@LuckyLuzz 
what screen are you using?

LuckyLuzz, 2016-07-20:
Standart 16x2 Display with a I2C Backpack. Its like rotate one click+ "Emty 1" rotate one click+ "Emty 2" rotate one click+ "Emty 1"

[image]

LuckyLuzz, 2016-07-20:
ok I had to change the pin confi to 

#define CK_ENC  2 // Quand encoder on an ISR capable input
#define DT_ENC  3
#define SW_ENC  4
 
ClickEncoder qEnc(DT_ENC,SW_ENC,CK_ENC , 4, LOW);

But the button is still not working

LuckyLuzz, 2016-07-21:
qEnc.setAccelerationEnabled(true);
  qEnc.setDoubleClickEnabled(true);

Ok also I have changed those to to false in beginning, thought double klick is not nesseary. But it's actully a normal click funktion. So now the whole thing is working and the handling/feeling is way better then before with default rotary libery. Thank's to you guys for that great work! :)

neu-rah, 2016-07-21:
will someone make a pull request with the changes?

christophepersoz, 2016-07-21:
Don't you received mine ? I did two yesterday

closed 2016-07-28 by neu-rah