neu-rah/ArduinoMenu #28: Menu with ATmega 1284p

benovitch, 2016-07-09:
I'm trying to load a sketch with your menu system. However I got the following error: #error "uC PCINT REVERSE MAP IS NOT DEFINED"
I found in a previous issue that someone hat the same problem. I entered in the PCINT.h file the following line: 
#elif ( defined(__AVR_ATmega1284__) )
	    const uint8_t PROGMEM pcintPinMap[4][8]={{A0,A1,A2,A3,A4,A5,A6,A7},{0,1,2,3,4,5,6,7},{16,17,18,19,20,21,22,23},{8,9,10,11,12,13,14,15}};
as you sugested in the previous issue. 
I found the pin numbers on the follow site: https://github.com/NicoHood/PinChangeInterrupt.
Ben

neu-rah, 2016-07-10:
that error means that the __AVR_ATmega1284__ is not defined at compile time... maybe it is lacking a terminal p, try with __AVR_ATmega1284P__ instead
and let me know if the code works so that we can include it into the lib

benovitch, 2016-07-10:
Well I think I found a sollution. 
I have changed the following in the PCINT.h file:

	#if ( defined(__AVR_ATmega328__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega16u4__) )
		const uint8_t PROGMEM pcintPinMap[4][8]={{8,9,10,11,12,13,-1,-1},{A0,A1,A2,A3,A4,A5,-1,-1},{0,1,2,3,4,5,6,7},{-1,-1,-1,-1,-1,-1,-1,-1}};
	#elif ( defined(__AVR_ATmega2560__) )
		const uint8_t PROGMEM pcintPinMap[4][8]={{53,52,51,50,10,11,12,13},{0,15,14,-1,-1,-1,-1,-1},{A8,A9,A10,A11,A12,A13,A14,A15},{-1,-1,-1,-1,-1,-1,-1,-1}};
	#elif ( defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644__))
	    const uint8_t PROGMEM pcintPinMap[4][8]={{A0,A1,A2,A3,A4,A5,A6,A7},{0,1,2,3,4,5,6,7},{16,17,18,19,20,21,22,23},{8,9,10,11,12,13,14,15}};
	#else
		#error "uC PCINT REVERSE MAP IS NOT DEFINED"
	#endif
	
	#define digitalPinFromPCINTSlot(slot,bit) pgm_read_byte(pcintPinMap+(((slot)<<4)+(bit)))
	#define pcintPinMapBank(slot) ((uint8_t*)((uint8_t*)pcintPinMap+((slot)<<4)))
	#define digitalPinFromPCINTBank(bank,bit) pgm_read_byte((uint8_t*)bank+bit)

It compilles but I still need to test it on the chip itself.

Ben

benovitch, 2016-07-13:
Well I tried to run my code and unfortunatley it doesn't work. When I attach the encoder to D2 and D3 or D8 and D9 the chip seems to reset itself. On other digital pins I get no response. Maybe I did something wrong on the PCINT.h file.
 `
#include <Wire.h>
#include <HardwareSerial.h>
#include <LiquidCrystal_I2C.h>
#include <menu.h>//menu macros and objects
#include <menuFields.h>
#include <pcint.h>//this is incompatible with software serial (arduino needs an handler!)
#include <quadEncoder.h>//quadrature encoder driver and fake stream
#include <keyStream.h>//keyboard driver and fake stream (for the encoder button)
#include <chainStream.h>// concatenate multiple input streams (this allows adding a button to the encoder)
#include "menuLCDs.h"
#include "HX711.h"

  #define DOUT1  28
  #define CLK1  29

  //LiquidCrystal lcd1(7, 6, 5, 4, 3, 2);
  
LiquidCrystal_I2C lcd1(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
menuLCD menu_lcd1(lcd1,20,4);//menu output device

  HX711 scale_EDF(DOUT1, CLK1);

 float calibration_factor_EDF = 219.25;
 float EDFmaxThrust = 0, EDFminThrust = 10000;

////////////////////////////////////////////
// ENCODER (aka rotary switch) PINS
// rotary
  #define encA 8
  #define encB 9
  //this encoder has a button here
  #define encBtn 4

  
 int opdracht = 0;
 const int ledPin =  13;

//functions to wire as menu actions

//aux function
void nothing() {}

void Thrust(){opdracht = 1; lcd1.clear();}
void Current(){opdracht = 2; lcd1.clear();}
void Power(){opdracht = 3; lcd1.clear();}
void MaxValues(){opdracht = 4; lcd1.clear();}
void Reset(){opdracht = 5; EDFmaxThrust = 0;}
void Save (){ digitalWrite(ledPin, HIGH); delay(200); digitalWrite(ledPin, LOW); }

/////////////////////////////////////////////////////////////////////////
// MENU DEFINITION
// here we define the menu structure and wire actions functions to it
// empty options are just for scroll testing

MENU(Settings,"Settings", 
  FIELD(calibration_factor_EDF,"calfac Scale","",0,300,1,0.01,nothing),
  OP("Save",Save)
  );

MENU(mainMenu,"Main menu",
OP("EDF Thrust",Thrust),
OP("Current",Current),
OP("Power",Power),
OP("Max Values",MaxValues),
OP("Reset Max Values",Reset),
SUBMENU(Settings)
);

[... 144 lines cut ...]
     lcd1.clear();
     lcd.drawn=NULL;
     delay(200);
     
      break;
      
    default:
     mainMenu.poll(lcd,allIn);
     break;
    }
  
  if (opdracht != 0) {  
    int stateButton = digitalRead(encBtn);
    if(stateButton == 0) {
     opdracht = 0;
     lcd1.clear();
     lcd.drawn=NULL;
     delay(200);
    }
  }
  
 if (millis() > NextTrigger){
  NextTrigger = (long)(millis() + plotDelay);
 } 
}`

neu-rah, 2016-07-16:
try making port B the first port (you can leave the rest as -1 for now, just for testing), and of coz wire the encoder to portB pins

benovitch, 2016-07-16:
Tested with only the B ports in the PCINT file. No succes on this side. I only got response on the pins D2 and D3. But then the board resets itself.

neu-rah, 2016-07-17:
on the above pin map "PROGMEM pcintPinMap[4][8]" there should be ONLY 3 bytes defined, as this code derives from atmega328p code and event tables support only 3 port.
then, what board definition are you using? and what IDE?
looking on the net i saw several pin numbering for the atmega1284p, the biard definition maps arbitrary numbers over port pins... so port definition is crucial.

benovitch, 2016-07-17:
I tried "PROGMEM pcintPinMap[4][8]"  as well as PROGMEM pcintPinMap[3][8]". With the B pins first and all the rest -1. 
I'm woorking with the 1.6.9 IDE. I'm using the board definitions from this link:   [https://github.com/JChristensen/mighty-1284p/tree/v1.6.3](url) 
I'm using the Bobuino variant.

neu-rah, 2016-07-18:
const uint8_t PROGMEM pcintPinMap[3][8]={{21,20,19,18,17,16,15,14},{4,5,6,7,10,11,12,13},{22,23,24,25,26,27,28,29}};

can you check this?

benovitch, 2016-07-18:
Well I've tested your input. With the Bobuino board I can't get it to work. I instaled the Uno Pro files in the \Arduino\hardware\arduino\avr\variants folder that i got from the vendors page:  [http://www.hobbytronics.co.uk/arduino-uno-pro-1284p](url) It seems for now that I only can use the Analog pins to work with the encoder.

neu-rah, 2016-07-29:
const uint8_t PROGMEM pcintPinMap[3][8]={{21,20,19,18,17,16,15,14},{4,5,2,3,10,11,12,13},{22,23,24,25,26,27,28,29}};

extracted from the pins_arduino.h file downloaded from that link, mapping for the Uno*Pro board...

neu-rah, 2016-07-30:
I have created this sketch that can be used on any arduino like board and print out to Serial the c++ definition of board pin map

https://github.com/neu-rah/PCINT/tree/master/examples/mkPCIntMap

benovitch, 2016-07-31:
Thanks for the help on the issue. Now the scroling true the menu works. But the selection of the menu item doesn't work. Its like the arduino doesn't rec the pushbutton. I ve tested the button in another sketch en the hardware works. When i touch the knob the screen begins to flicker.

neu-rah, 2016-07-31:
what output device are you using? I've just pushed some corrections on the LCD driver to deal with PROGMEM (was causing resets)

closed 2016-08-01 by neu-rah