neu-rah/ArduinoMenu #29: Unable to compile after adding FIELD

LuckyLuzz, 2016-07-20:
Hi and thank for this great lib!
I got a issue, hope you can help me. After adding FIELD to the Main Manu in the I2C Example the Arduino IDE 1.6.9 is not able to comple the code.

MENU(mainMenu,"Main",
  OP("Frequency",nothing),
  OP("Dutty",nothing),
  OP("Setup",nothing)
  FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing()),
);

ERROR:

Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Thomas\Documents\Arduino\I2C_menu\I2C_menu.ino:20:0:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:123:28: error: expected constructor, destructor, or type conversion before '(' token

   #define FIELD(...) FIELD_(__COUNTER__,__VA_ARGS__)

                            ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:95:26: note: in definition of macro 'DECL'

   #define DECL(x) DECL_##x

                          ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:46:5: note: in expansion of macro 'FOR_EACH_2'

     FOR_EACH_2(what, __VA_ARGS__)

     ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:49:5: note: in expansion of macro 'FOR_EACH_3'

     FOR_EACH_3(what,  __VA_ARGS__)

     ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:38:37: note: in expansion of macro 'FOR_EACH_4'

   #define CONCATENATE2(arg1, arg2)  arg1##arg2

                                     ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:93:34: note: in expansion of macro 'FOR_EACH_'

   #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)

                                  ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:99:5: note: in expansion of macro 'FOR_EACH'

     FOR_EACH(DECL,__VA_ARGS__)\

     ^

C:\Users\Thomas\Documents\Arduino\I2C_menu\I2C_menu.ino:28:1: note: in expansion of macro 'MENU'

 MENU(mainMenu,"Main",

 ^

C:\Users\Thomas\Documents\Arduino\I2C_menu\I2C_menu.ino:32:3: note: in expansion of macro 'FIELD'

   FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing()),

   ^

I2C_menu:28: error: 'mainMenu_data' was not declared in this scope

 MENU(mainMenu,"Main",

      ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:103:26: note: in definition of macro 'MENU'

     menu id (text,sizeof(id##_data)/sizeof(prompt*),id##_data);

                          ^

I2C_menu:28: error: 'mainMenu_data' was not declared in this scope

 MENU(mainMenu,"Main",

      ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:103:53: note: in definition of macro 'MENU'

     menu id (text,sizeof(id##_data)/sizeof(prompt*),id##_data);

exit status 1
'mainMenu_data' was not declared in this scope

[I2C_menu.zip](https://github.com/neu-rah/ArduinoMenu/files/374116/I2C_menu.zip)

neu-rah, 2016-07-20:
yes, pay attention to separators, commas are not well placed between options and macros dont behave well with that.

```C++
MENU(mainMenu,"Main",
OP("Frequency",nothing),
OP("Dutty",nothing),
OP("Setup",nothing),
FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing())
);
```

LuckyLuzz, 2016-07-20:
I see.. just tried it again with your code and still not working. 

In file included from C:\Users\Thomas\Documents\Arduino\I2C_menu\I2C_menu.ino:20:0:
I2C_menu:32: error: variable 'menuField<int> _menuField3' has initializer but incomplete type
 FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing())
       ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:128:81: note: in definition of macro 'DECL_FIELD_'
   #define DECL_FIELD_(cnt,target,...) menuField<typeof(target)> _menuField##cnt(target,__VA_ARGS__);
                                                                                 ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:40:36: note: in expansion of macro 'DECL'
   #define FOR_EACH_1(what, x, ...) what(x)
                                    ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:43:5: note: in expansion of macro 'FOR_EACH_1'
     FOR_EACH_1(what,  __VA_ARGS__)
     ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:46:5: note: in expansion of macro 'FOR_EACH_2'
     FOR_EACH_2(what, __VA_ARGS__)
     ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:49:5: note: in expansion of macro 'FOR_EACH_3'
     FOR_EACH_3(what,  __VA_ARGS__)
     ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:38:37: note: in expansion of macro 'FOR_EACH_4'
   #define CONCATENATE2(arg1, arg2)  arg1##arg2
                                     ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:93:34: note: in expansion of macro 'FOR_EACH_'
   #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
                                  ^
C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-master/menu.h:99:5: note: in expansion of macro 'FOR_EACH'
     FOR_EACH(DECL,__VA_ARGS__)\
     ^
C:\Users\Thomas\Documents\Arduino\I2C_menu\I2C_menu.ino:28:1: note: in expansion of macro 'MENU'
 MENU(mainMenu,"Main",
 ^
C:\Users\Thomas\Documents\Arduino\I2C_menu\I2C_menu.ino:32:1: note: in expansion of macro 'FIELD'
 FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing())
 ^
exit status 1
variable 'menuField<int> _menuField3' has initializer but incomplete type

neu-rah, 2016-07-20:
on your code

```c+
MENU(mainMenu,"Main",
OP("Frequency",nothing),
OP("Dutty",nothing),
OP("Setup",nothing)     //<-- missing a comma her!
FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing()),  //<-- extra comma here!
);
```

this kind of errors produces an avalanche of macro errors

also on the last line you are calling the function "nothing" instead of giving the address.

```c++
MENU(mainMenu,"Main",
OP("Frequency",nothing),
OP("Dutty",nothing),
OP("Setup",nothing),
FIELD(frequency,"Freq","Hz",0,16000000,100,1,nothing)
);
```

this last version should work

LuckyLuzz, 2016-07-20:
Hey found it, I did't know I had to #include <menuFields.h> .... have seen that in the Sample "Seriel Menu". I have tried to compile "Seriel Menu" too and it was not working. Maybe you could have a look at that too. Thanks for your help!

LuckyLuzz, 2016-07-20:
#include <menuFields.h>

LuckyLuzz, 2016-07-20:
"#include <menuFields.h> "

closed 2016-07-20 by LuckyLuzz