neu-rah/ArduinoMenu #33: Changing CHOOSE or TOGGLE behavior on graphic display

christophepersoz, 2016-07-21:
Me again :D

Can you tell me where is the CHOOSE or TOGGLE behavior and methods in the lib ? I would like to change the way how the values are displayed - On graphic display you often have enough room to do that.So, I though to something like that :

Menu Choose/Toggle selected > Click (will stay on the menu screen) to edit value > Up/Down will change the value with datas from the VALUE fields > Click to validate changes.

I don't know which one is the easiest to change, maybe TOGGLE.
What do you think ?

TomaTLAB, 2016-07-21:
In the menuFields.h file. But it define behavior on any output device, i think.
IMO it'll be better to add some extra type of fields/submenus.

TomaTLAB, 2016-07-21:
BTW extra numeric multy-field for time, date, IP-address etc... is not bad to have.

christophepersoz, 2016-07-21:
Maybe it's a good idea to have #indef for Graphics vs LCD and change the behaviors.
Note that this change can also be compatible with small display (16x2), the only problem can be the length of the VAULES displayed that could be truncated.

TomaTLAB, 2016-07-21:
I see the only problem with IP on small display, but... It's impossible to fit unfitable ;)

TomaTLAB, 2016-07-21:
I think a little... imo we need a some sort of horizontal menu for multi-field. It's need to think a bit more :)
So, offtopic...

neu-rah, 2016-07-22:
butoons
button matrix
layouts
but agree.. lets think first
maybe we can do a better system from ground up,
the basic idea was to call function on click/activate and in some functions i use to call a field editor, this is still valid.
but the menu was blocking on early days.. to do it now field editors must be non blocking.
fields came andy because they show the value on the menu, so i implemented them as part of the system
btw, field editors can be blocking... some systems dont care

to cross menu object with screen outputs and navigation...

menu objects: should only contain descriptions and status
output objects: represent descriptions
input objects: do navigation

using Streams as inputs was a good idea?
the objective was to get down to serial comunication level

draw systems should operate in minimal and full mode
- minimal, draws only changed parts
- full, draw all block, to deal with dual buffered screens <-- not sure if its the solution, not tried yet

...

christophepersoz, 2016-07-22:
I understand what you are saying by blocking systems. Good point. I think that U8G2 can help, Oliver has developed it mainly for monochrome displays and it managed full buffered screen or pages in I2C, HW SPI and so. Maybe it can worth it to have a look on that but the way it works it completely different to your current system, so... 

To come back to my first question, it was just changing the way you are browsing all the available VALUES for a TOGGLE field. Instead of clicking to change from a value to an another, using up/down to browse them (after clicked once for editing mode) without changing the screen and just refreshing the value for the selected field.

neu-rah, 2016-07-22:
that click and browse is the CHOOSE

neu-rah, 2016-07-22:
i see, very good lets call it SELECT (?)
plus TOGGLE | CHOOSE | SELECT can have option to update on move or update on click
right?

christophepersoz, 2016-07-22:
Let go for SELECT then ;)
Nice behavior could be : 
1. Click to activate on browsing value mode (without changing the display appearance)
2. Using Up/Down to browse the values (like on the current TOOGLE, but using up/down instead of multiple click)
3. Click again to validate the current displayed value and returning to normal menu mode
Am I clear ? It seems that to closer model to this seems to be TOOGLE, but maybe CHOOSE can be easier to modify and adapt it to this new SELECT mode.

neu-rah, 2016-07-22:
browses like choose but displays like toggle... we will mix'em up

christophepersoz, 2016-07-22:
Exactly, which also means less refresh pages and quicker edit when you are using a rotary encoder... At least I think so :D

neu-rah, 2016-08-02:
added new SELECT field with:
-click to enter edit mode
-rotate to change value
-click to end edit

christophepersoz, 2016-08-02:
Fantastic ! Thanks a lot, can't wait to try v2.4 ;)

neu-rah, 2016-08-02:
hope it has not much bugs :bug:

christophepersoz, 2016-08-02:
I will soon keep you in touch about that ;)

christophepersoz, 2016-08-03:
Hello Rui,

I tried to build this morning the Sketch U8Glib_menu.ino first for Teensy 3.1/3.2 boards but I'm getting some errors (see next post). So, I tried to build il for MEGA2560, and I'm also getting an error on a var type inside u8glib library, sic... I don't know why because this library usually works well in HW SPI.

Here the error output for MEGA2560 (only the end, otherwise it is a bit too long :
```c++
Arduino: 1.6.7 (Mac OS X), TD: 1.29, Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose 
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI" "-I/Users/christophepersoz/Developer/_Arduino/libraries/U8glib" "-I/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src" "-I/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility" "/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp" -o "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp.o"
/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp: In function 'uint8_t u8g_com_arduino_hw_spi_fn(u8g_t*, uint8_t, uint8_t, void*)':
/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp:263:37: error: invalid conversion from 'void*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
             register uint8_t *ptr = arg_ptr;

                                     ^
/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp:273:37: error: invalid conversion from 'void*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
             register uint8_t *ptr = arg_ptr;

                                     ^
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
Using library U8glib in folder: /Users/christophepersoz/Developer/_Arduino/libraries/U8glib (legacy)
Using library Menu at version 2.4 in folder: /Users/christophepersoz/Developer/_Arduino/libraries/Menu 
exit status 1
Error compiling for "Arduino/Genuino Mega or Mega 2560"
```

Let's see now the errors on a Teensy build ;)

christophepersoz, 2016-08-03:
Here the build error I'm getting on Teensy board:

```c++
Arduino: 1.6.7 (Mac OS X), TD: 1.29, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), US English"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serial,speed=96opt,keys=en-us -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -verbose "/Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serial,speed=96opt,keys=en-us -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -verbose "/Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino"
Build options changed, rebuilding all

...

"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino5/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=129 -DARDUINO=10607 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3" "-I/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI" "-I/Users/christophepersoz/Developer/_Arduino/libraries/U8g_teensy" "-I/Users/christophepersoz/Developer/_Arduino/libraries/U8glib" "-I/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src" "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp/sketch/U8Glib_ClickEncoder_Teensy_Demo.ino.ino.cpp" -o "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp/sketch/U8Glib_ClickEncoder_Teensy_Demo.ino.ino.cpp.o"
In file included from /Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino:29:0:
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuU8G.h:102:8: warning: extra tokens at end of #endif directive [enabled by default]
 #endif RSITE_ARDUINOP_MENU_LCD
        ^
In file included from /Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino:27:0:
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h: In constructor 'menuOut::menuOut(int, int, int, int)':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:48:9: warning: 'menuOut::maxY' will be initialized after [-Wreorder]
     int maxY;
         ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:45:9: warning:   'int menuOut::top' [-Wreorder]
     int top;//top for this device
         ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:56:12: warning:   when initialized here [-Wreorder]
     inline menuOut(int x=0x7F,int y=0x7F,int resX=1,int resY=1)
            ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:51:9: warning: 'menuOut::resY' will be initialized after [-Wreorder]
     int resY;
         ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:44:11: warning:   'menu* menuOut::drawn' [-Wreorder]
     menu* drawn;//last drawn menu, avoiding clear/redraw
           ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:56:12: warning:   when initialized here [-Wreorder]
     inline menuOut(int x=0x7F,int y=0x7F,int resX=1,int resY=1)
            ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h: In constructor 'prompt::prompt(const char*)':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:123:10: warning: 'prompt::enabled' will be initialized after [-Wreorder]
     bool enabled;
          ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menu.h:121:18: warning:   'promptAction prompt::action' [-Wreorder]
     promptAction action;
                  ^
[... cut ...]
  //U8GLIB_PCD8544 u8g(U8_CS, U8_DC, U8_RST) ;
#elif defined (__MK20DX256__) || defined(__MK20DX128__) // Teensy 3.2, 3.1 and 3.0
  U8GLIB u8g(&u8g_dev_ssd1322_nhd31oled_2x_gr_hw_spi, u8g_com_hw_spi_fn); // HW SPI Com: CS = 10, A0 = 9 (Hardware Pins are  SCK = 13 and MOSI = 11)
#else
  #error "U8GLIB DISPLAY NEEDED"
#endif

menuU8G gfx(u8g,1,0,1,1,7,9);
```

As you see, there much is much more issues on Teensy. Maybe they are all linked. The main concern is `pgm_read_ptr_near` which seems not recognized.

neu-rah, 2016-08-03:
<avr/pgmspace.h> is the file for pgm_read_ptr_near
strange...
I'm trying to instalkl teensy board defs on 1.6.10 to track that

neu-rah, 2016-08-03:
can you open this as a new issue.. errors building for teensy or something like that?

neu-rah, 2016-08-03:
Ok, just pushed some changes to allow conditional usage of flash memory, pgm is very specific to AVR's I guess

now you must use
```c++
#define USEPGM
```
in your sketch before any menu includes to activate the PROGMEM usage
otherwise ram will be used (teensy seems to have a lot of ram)

TomaTLAB, 2016-08-03:
Yep! It's quite important for non AVR platform, e.g. stm32duino I'm using.

closed 2016-08-22 by neu-rah