neu-rah/ArduinoMenu #34: Refreshrate of Menu

LuckyLuzz, 2016-07-21:
Hi,
I have an analoginput as value in a FIELD to show it on Homescreen. But now it's only refreshing the screen when I navigate thru the Menu. Is there a way to refresh it faster?

`MENU(mainMenu,"Aquarium",
  FIELD(TempInput,"Temp  "," C",0,0,0,0), 
 FIELD(TempOutputProzent,"Light "," %",0,0,0,0),
  SUBMENU(subMenuSettings)
 );`
[ArduinoMenu_I2C_LCD_RotaryEncoder-2016.07.22.zip](https://github.com/neu-rah/ArduinoMenu/files/377152/ArduinoMenu_I2C_LCD_RotaryEncoder-2016.07.22.zip)

neu-rah, 2016-07-22:
are you cleaning the menu somewhere else?
the menu system only draws when changed...
however you can force a menuOut.redraw()
but if you are clearing the screen somewhere else you will experience flicking

LuckyLuzz, 2016-07-22:
Hey,
no I don't do lcd.clear() it's just the menu system writing the LCD.

So now I have forced a redraw in VOID Loop, but it's flicking. 
   `mainMenu.poll(menu_lcd,enc);` 
   mainMenu.redraw(menu_lcd,enc,false);`

What would you say is the best way to have a HomeScreen with a clock or a analog Temperatur?

neu-rah, 2016-07-22:
i would suspend the menu and use the screen for nice updated display of the values
and the menu only for adjustments
see
https://github.com/neu-rah/ArduinoMenu/issues/25#issuecomment-234102400

LuckyLuzz, 2016-07-22:
Got it! I also implementet a 10s Timeout for the Menu.

```const long interval = 10000;          // Menu 10 sTimeout
   unsigned long currentMillis;
   unsigned long previousMillis; `
   currentMillis = millis();

  if (runMenu) {
        if (currentMillis - previousMillis >= interval){
         pauseMenu(mainMenu,menu_lcd,enc);}
        {mainMenu.poll(menu_lcd,enc); }}
  else if (enc.read()==menu::enterCode){        //menu reenter method
          menu_lcd.redraw();
          runMenu=true;
          previousMillis = currentMillis; 
      } else
      {
```

Is it possible to listen not just for the enterCode but for Up & Down also? I tried it with 
```if (enc.read()==menu::enterCode||enc.read()==menu::enterupCode||enc.read()==menu::downCode){  ```
but it was not reacting every time... :(

neu-rah, 2016-07-22:
the read removes the character from the stream, use peek or read into a variable before comparing
peek still needs the read after, so i would prefer reading into a variable

LuckyLuzz, 2016-07-22:
hmm the Menu itself is not missing keys, it works perfectly fine. And yes I guess I am not using PCINT. 

I actually just wone build a Timeout function for the Menu, like" go back to Home after 10s if no movment from Encoder" The 10s working but it's not recognizing the keys. My try with (see code) was not succesfull.

char movment = enc.read()
If ((movment == '+')|| (movment == '-'))
       previousMillis = currentMillis;

Since everything else is working fine I will continue with implementing TOGGLE. You will see in a week I got every feature of your lib in my project :)

neu-rah, 2016-07-22:
again use enc.peek() to avoid removing keys from the stream

the included encoder driver uses PCINT, but the keyboard does not (it depends on repeated checks) so if delay between checks gets too long then keys can be lost.
one of this days i might build a new keyboard driver to use PCINT and use a buffer to avoid key loose

LuckyLuzz, 2016-07-24:
ok, I am using the ClickEncoderStream.h and not the default one. Is it still then possible to do it with  enc.peek() ?

neu-rah, 2016-07-24:
guess so, but that file is not part of the repo... it was not included on none of the pull requests, i can add it but i have not the last changes and i would prefer it assigned to the author.

anyway, peek should to all the work but not remove the key from stream

closed 2016-07-24 by LuckyLuzz