neu-rah/ArduinoMenu #36: Examples won't compile out of the box

etkArduino, 2016-07-25:
Just trying the samples included in the download.

The error I got after compile in Arduino IDE version 1.6.3:       "streamFlow.h: No such file or directory"
Missing file?

christophepersoz, 2016-07-25:
Have you tried to comment the line `#include <streamFlow.h>` and try to replace it by `#include <chainStream.h>`, if this line doesn't exist, in the example you are talking about ?

neu-rah, 2016-07-26:
streamFlow.h is something i use for debugging and is not included on the repo
please remove that `#include <streamFlow.h>` 
sorry about that

etkArduino, 2016-07-26:
Glad you are willing to help me out of this. Thanks.
I am not very good in programming so it is not very obvious to me, still
learning ...

I replaced the includes in "serial_menu.ino" as wel as in "menu.h" but
still got a compile error:

>>>>>>
In file included from serial_menu.ino:16:0:
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h: In member
function 'virtual void menuPrint::printMenu(menu&, bool)':
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:38:55: error: no
matching function for call to 'menuPrint::printPrompt(prompt&, bool, int,
int, int&)'
      printPrompt(*m.data[i],i==m.sel,i+1,i-top,m.width);
                                                       ^
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:38:55: note:
candidate is:
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:26:16: note:
virtual void menuPrint::printPrompt(prompt&, bool, int, int, int, int)
   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int
posY,int width) {
                ^
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:26:16: note:
candidate expects 6 arguments, 5 provided
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:39:74: error: no
matching function for call to 'menuPrint::printPrompt(prompt&, bool, int,
int, int&)'
    if (drawExit) printPrompt(menu::exitOption,m.sel==m.sz,0,i-top,m.width);

 and more ...

<<<<<<

Hope this helps.

Regards,
Eric

etkArduino, 2016-07-26:
I removed the includes in "serial_menu.ino" (as an example) and also in
"menu.h" but I still get :

>>>>>>
In file included from serial_menu.ino:16:0:
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h: In member
function 'virtual void menuPrint::printMenu(menu&, bool)':
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:38:55: error: no
matching function for call to 'menuPrint::printPrompt(prompt&, bool, int,
int, int&)'
      printPrompt(*m.data[i],i==m.sel,i+1,i-top,m.width);
                                                       ^
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:38:55: note:
candidate is:
D:\Documents\Arduino\sketchbook\libraries\menu/menuPrint.h:26:16: note:
virtual void menuPrint::printPrompt(prompt&, bool, int, int, int, int)
   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int
posY,int width) {
                ^

and more ...

<<<<<<

Thanks for your help.

Regards,
Eric

neu-rah, 2016-07-26:
are you using https://github.com/neu-rah/ArduinoMenu/releases/tag/2.1 ?

note that 2.3 is under development and examples are not updated yet

etkArduino, 2016-07-27:
Hi Rui,

Thanks for your help.

I have downloaded the library from    https://github.com/neu-rah/ArduinoMenu
and using the download button assuming it is the latest released version.
As far I can tell it is the 2.1 version.

In the mean time I upgraded to IDE version 1.6.10 (using c++11 ?).

I replaced in "serial_menu.ino" and "menu.h"  the include "streamFlow.h"
into "chainStream.h"

Errors I get now:
>>>>
In file included from
D:\Documents\Arduino\sketchbook\libraries\menu/menu.h:35:0,
                 from
D:\Documents\Arduino\sketchbook\LIBRARIES\menu\examples\serial_menu\serial_menu.ino:14:
D:\Documents\Arduino\sketchbook\libraries\menu/macros.h:65:17: error:
pasting "DECL_" and "/*OP("LED On",ledOn),
  OP("LED Off",ledOff),*/" does not give a valid preprocessing token
 #define DECL(x) DECL_##x
                 ^
D:\Documents\Arduino\sketchbook\libraries\menu/macros.h:15:3: note: in
expansion of macro 'DECL'
   what(x)\
   ^
D:\Documents\Arduino\sketchbook\libraries\menu/macros.h:8:35: note: in
expansion of macro 'FOR_EACH_3'
 #define CONCATENATE2(arg1, arg2)  arg1##arg2
                                   ^
D:\Documents\Arduino\sketchbook\libraries\menu/macros.h:63:32: note: in
expansion of macro 'FOR_EACH_'
 #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__),
what, x, __VA_ARGS__)
                                ^
D:\Documents\Arduino\sketchbook\libraries\menu/macros.h:69:3: note: in
expansion of macro 'FOR_EACH'
   FOR_EACH(DECL,__VA_ARGS__)\
   ^

and more ...
<<<<

Regards,

Eric

neu-rah, 2016-07-27:
looking at the errors its 2.3, I've tried the download and got 2.3 also... have to check that
in the mean while here is the link for 2.1 https://github.com/neu-rah/ArduinoMenu/archive/2.1.zip

etkArduino, 2016-07-27:
Thanks.

I downloaded 2.1 from the link (it is still hard to find out which version
one is dealing with ...) and compiled "serial_menu.ino" (with Arduino IDE
version 1.6.10).

The result:
>>>>>>>
In file included from
D:\Documents\Arduino\sketchbook\LIBRARIES\menu\examples\serial_menu\serial_menu.ino:17:0:
D:\Documents\Arduino\sketchbook\libraries\menu/menu.h:95:19: error: pasting
"DECL_" and "/*OP("LED On",ledOn),
  OP("LED Off",ledOff),*/" does not give a valid preprocessing token
   #define DECL(x) DECL_##x
                   ^
D:\Documents\Arduino\sketchbook\libraries\menu/menu.h:45:5: note: in
expansion of macro 'DECL'
     what(x)\
     ^
D:\Documents\Arduino\sketchbook\libraries\menu/menu.h:38:37: note: in
expansion of macro 'FOR_EACH_3'
   #define CONCATENATE2(arg1, arg2)  arg1##arg2
                                     ^
D:\Documents\Arduino\sketchbook\libraries\menu/menu.h:93:34: note: in
expansion of macro 'FOR_EACH_'
   #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__),
what, x, __VA_ARGS__)
                                  ^
D:\Documents\Arduino\sketchbook\libraries\menu/menu.h:99:5: note: in
expansion of macro 'FOR_EACH'
     FOR_EACH(DECL,__VA_ARGS__)\
     ^
...
and more ....
...
<<<<<<<

Is this an issue that has to do with compiler settings?  I am not an expert.

The  LCD_menu  and  the  I2C_LCD_menu did compile OK.
Did not bother to download the Adfruit drivers so I don't know if gfx_menu
compiles.

Regards,

Eric

neu-rah, 2016-07-27:
i'm using 1.6.9 (for 2.3 develop), downloading.. 1.6.10

neu-rah, 2016-07-27:
please remove lines 39 and 40 on serial_menu.ino

    39  /*OP("LED On",ledOn),
    40 OP("LED Off",ledOff),*/

etkArduino, 2016-07-27:
Yep !
That's it !

Glad you found it.
Apparently for the IDE it was commented out (greyed out) but the compiler
did not like it?

Thanks for your help.

Regards,

Eric

neu-rah, 2016-07-27:
somehow macros didn't like it

closed 2016-07-27 by neu-rah