neu-rah/ArduinoMenu #40: ArduinoMenu v2.4 does not build on Teensy 3.1/3.2 due to PROGMEM

christophepersoz, 2016-08-03:
It is just a move of my feedback concerning the build of v2.4 on Teensy board.
Before posting it I tried to `#include <avr/pgmspace.h>` In menuFields.h but as you noticed this does not solve it. I saw some lib using the same functions that you are using for PROGMEM which seems working. I'm just wondering if it's not a problem related to Template/Macro  

I tried to build this morning the Sketch U8Glib_menu.ino first for Teensy 3.1/3.2 boards but I'm getting some errors (see next post). So, I tried to build il for MEGA2560, and I'm also getting an error on a var type inside u8glib library, sic... I don't know why because this library usually works well in HW SPI.

Here the error output for MEGA2560 (only the end, otherwise it is a bit too long :
```c++
Arduino: 1.6.7 (Mac OS X), TD: 1.29, Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino"
Build options changed, rebuilding all

....

"/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp" -o "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp.o"
/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp: In function 'uint8_t u8g_com_arduino_hw_spi_fn(u8g_t*, uint8_t, uint8_t, void*)':
/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp:263:37: error: invalid conversion from 'void*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
             register uint8_t *ptr = arg_ptr;

                                     ^
/Users/christophepersoz/Developer/_Arduino/libraries/U8glib/utility/u8g_com_arduino_hw_spi.cpp:273:37: error: invalid conversion from 'void*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
             register uint8_t *ptr = arg_ptr;

                                     ^
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
Using library U8glib in folder: /Users/christophepersoz/Developer/_Arduino/libraries/U8glib (legacy)
Using library Menu at version 2.4 in folder: /Users/christophepersoz/Developer/_Arduino/libraries/Menu 
exit status 1
Error compiling for "Arduino/Genuino Mega or Mega 2560"
```

Let's see now the errors on a Teensy build ;)

neu-rah, 2016-08-03:
thanks **all** for all testing and feedback, I do really appreciate it.

christophepersoz, 2016-08-03:
Does MenuField.h include your changes with `#define MEMMOD` ?
I'm still getting some error related to `ppm_read_ptr_near()` with the latest v2.4.

neu-rah, 2016-08-03:
well there still was a **pgm_read_byte_near** on menu.cpp
I've removed that now

neu-rah, 2016-08-03:
not good... broke something

neu-rah, 2016-08-04:
its not possible to use a preprocessor define to turn progmem usage on/off because I have dependent code inside a .cpp file (and its gota be there), so i managed to make it automatic by checking #ifdef pgm_read_ptr_near, it will not be defined for teensy.
It compiles here but i have no teensy upload it.
I've noticed some general issues relating to navigation also (not teensy related), will check that...

christophepersoz, 2016-08-04:
If you push it here in a zip file I can upload it onto my teensy and check it.
I do not understand why pgm_read_ptr_near() does not compile, because the function seems working on Teensy.

neu-rah, 2016-08-04:
its on current version

yea, that pgm_read_ptr_near somehow seems to not be defined on teensy 3.1/3.2 and i'm using that fact to distinguish between the cases

christophepersoz, 2016-08-04:
Exact, it is not. Maybe be you could ask to Paul on [PJRC forum](https://forum.pjrc.com), he usually answer to this kind of question. It is a bit too technical to my knowledge.

neu-rah, 2016-08-04:
i dont mind to use ram on teensy, theres plenty, i'm trying to support both modes, if I succeed then i might support dynamic menus.
does it run on teensy? (changes are already pushed)

christophepersoz, 2016-08-05:
Indeed, that could be nice to support both. I tried yesterday to build the 2.4 on Teensy 3.2 and using MEMMOD tag, but it still some errors related to pgm_read_ptr_near(), here the log.

```c++
Arduino: 1.6.7 (Mac OS X), TD: 1.29, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), US English"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serial,speed=96opt,keys=en-us -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -verbose "/Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/christophepersoz/Developer/_Arduino/libraries" -fqbn=teensy:avr:teensy31:usb=serial,speed=96opt,keys=en-us -ide-version=10607 -build-path "/var/folders/93/95zp52wn3dq81d0dgrtmj_b80000gn/T/build0da4f63f4b42a48bacd9931d3ed57970.tmp" -warnings=none -verbose "/Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino"

...

            ^
In file included from /Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino:28:0:
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h: In member function 'virtual bool menuVariant<T>::needRedraw(menuOut&, bool)':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h:138:56: error: there are no arguments to 'pgm_read_ptr_near' that depend on a template parameter, so a declaration of 'pgm_read_ptr_near' must be available [-fpermissive]
    bool nr=((menuValue<T>*)pgm_read_ptr_near(&data[sel]))->value!=target;//||p.lastSel!=sel;
                                                        ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h:138:56: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h: In member function 'void menuVariant<T>::sync()':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h:146:52: error: there are no arguments to 'pgm_read_ptr_near' that depend on a template parameter, so a declaration of 'pgm_read_ptr_near' must be available [-fpermissive]
      if (((menuValue<T>*)pgm_read_ptr_near(&data[n]))->value==target)
                                                    ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h: In member function 'virtual void menuVariant<T>::printTo(menuOut&)':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h:152:42: error: there are no arguments to 'pgm_read_ptr_near' that depend on a template parameter, so a declaration of 'pgm_read_ptr_near' must be available [-fpermissive]
    ((prompt*)pgm_read_ptr_near(&data[sel]))->printTo(p);
                                          ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h: In member function 'virtual bool menuSelect<T>::needRedraw(menuOut&, bool)':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h:181:67: error: there are no arguments to 'pgm_read_ptr_near' that depend on a template parameter, so a declaration of 'pgm_read_ptr_near' must be available [-fpermissive]
    return ((menuValue<T>*)pgm_read_ptr_near(&menu::data[menu::sel]))->value!=menuVariant<T>::target;
                                                                   ^
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h: In member function 'virtual void menuSelect<T>::printTo(menuOut&)':
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuFields.h:187:54: error: there are no arguments to 'pgm_read_ptr_near' that depend on a template parameter, so a declaration of 'pgm_read_ptr_near' must be available [-fpermissive]
    ((prompt*)pgm_read_ptr_near(&menu::data[menu::sel]))->printTo(p);
                                                      ^
In file included from /Users/christophepersoz/Library/Mobile Documents/com~apple~CloudDocs/Developer/_Arduino/libraries/Menu/examples/U8Glib_ClickEncoder_Teensy_Demo.ino/U8Glib_ClickEncoder_Teensy_Demo.ino.ino:29:0:
/Users/christophepersoz/Developer/_Arduino/libraries/Menu/src/menuU8G.h: In constructor 'menuU8G::menuU8G(U8GLIB&, unsigned char, unsigned char, unsigned char, unsigned char, uint8_t, uint8_t)':
[... cut ...]
                                                                                  ^
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI 
Using library U8g_teensy in folder: /Users/christophepersoz/Developer/_Arduino/libraries/U8g_teensy (legacy)
Using library U8glib in folder: /Users/christophepersoz/Developer/_Arduino/libraries/U8glib (legacy)
Using library Menu at version 2.4 in folder: /Users/christophepersoz/Developer/_Arduino/libraries/Menu 
Error compiling for "Teensy 3.2 / 3.1"
```

neu-rah, 2016-08-05:
I've compiled that source here for teensy, with no errors

that menuFields.h:138 line is:
```c++
bool nr=((menuValue<T>*)pgmPtrNear(&data[sel]))->value!=target;//||p.lastSel!=sel;
```
on avrs it should translate to this (by detecting the presence of pgm_read_ptr_near macro)
```c++
bool nr=((menuValue<T>*)pgm_read_ptr_near(&data[sel]))->value!=target;
```
on teensy (macro pgm_read_ptr_near is not defined) it should translate to:
```c++
bool nr=((menuValue<T>*)(&data[sel]))->value!=target;
```
so it seems that our environment is a bit different and somehow...

btw, do not define MEMMODE, it should be automatic

on macros.h:
```c++
#ifdef pgm_read_ptr_near
  //storing some values into avr flash memory (saving ram space)
  #define MEMMODE PROGMEM
  #define pgmPtrNear(addr) pgm_read_ptr_near(addr)
  #define pgmByteNear(addr) (pgm_read_byte_near(addr))
#else
  //use ram for non-avr devices
  #define MEMMODE
  #define pgmPtrNear(addr) (addr)
  #define pgmByteNear(addr) (*addr)
#endif
```

neu-rah, 2016-08-05:
if errors persist when compiling simple examples like serial menu then please try replacing that conditional section #ifdef pgm_read_ptr_near .... with only the ram part by removing the avr defs like this:
```c++
/*#ifdef pgm_read_ptr_near
  //storing some values into avr flash memory (saving ram space)
  #define MEMMODE PROGMEM
  #define pgmPtrNear(addr) pgm_read_ptr_near(addr)
  #define pgmByteNear(addr) (pgm_read_byte_near(addr))
#else*/
  //use ram for non-avr devices
  #define MEMMODE
  #define pgmPtrNear(addr) (addr)
  #define pgmByteNear(addr) (*addr)
//#endif
```

christophepersoz, 2016-08-05:
Sorry, I think it was a problem related to the version, I was downloading the 2.4 instead of the master branch where indeed I'm seeing the #ifdef you mentioned. So I'm going to make a build with that one ! Let's try and see what happen.

christophepersoz, 2016-08-05:
So, I can build it now, sorry for the lost of time. But, the menu does not work, at least it does not display anything. So I'm going to investigate a bit to see what's going on.

neu-rah, 2016-08-05:
serial_menu can be a good point

if it does not run ok => menu nav system
if it runs ok => display system

no time lost my friend, and I'm grateful for your patience. I had some errors, the kind of ones that compile ok (logic errors), like storing on ram and then trying to read from flash... it compiles but will print trash or reset the chip (unpredictable result).

and i hope i don't have much more of those ;D

christophepersoz, 2016-08-05:
How do you activate those lines ?
```c++
Serial<<"printing menu "<<m<<" top:"<<top<<endl;
Serial<<*menu::activeNode<<" <-> "<<*drawn<<endl;
```

I tried with Streaming.h library but it seems it does not understand the objects or pointers, Serial.print neither.

I can have some serial printing on the Monitor during the menuPoll, but nothing on display except String that are not in the menu and which are printed directly with DrawStr() from U8GLib.

neu-rah, 2016-08-05:
that << (stream operator) is with active on my debug utility streamFlow.h, its lighter that the standard c++ implementation and is handy for printing.

guess i can publish that... brb

christophepersoz, 2016-08-05:
That's why I tried with Streaming.h but it seems no so powerful...

BTW, I think that there is several issues. I included <menuPrint.h> to poll the menu described in U8Glib_menu.ino sketch to print it out directly on serial, and it does not print the right content, I only got `1>` nothing more.

neu-rah, 2016-08-05:
https://github.com/neu-rah/streamFlow

neu-rah, 2016-08-05:
yes i confirm that as a bug, by forcing ram build on arduino nano I had the same problem as you have on teensy...
means the back compatibility to ram mode is not ok yet

christophepersoz, 2016-08-05:
Yes, it seems so, streamFlow does not output a lot too :

```c++
printing menu Main menu top:0
Main menu <-> 
```

neu-rah, 2016-08-05:
there was a macro mess (about who gets the reference) that messed up ram mode for teensy
changes pushed to mater

neu-rah, 2016-08-09:
did I broke everything?

christophepersoz, 2016-08-09:
Hi Rui,

Sorry, I'm abroad for work. I will have a look on the latest version when I will be back next week. I'll keep you in touch. Thanks !

christophepersoz, 2016-08-15:
Hello Rui,

I'm back and I took few minutes to test the latest version of Menu on my Teensy board. First good news it does build, and "works" on Teensy, which is a quite good news, isn't it ;)

But, not everything is working well. I'm still using the file `U8Glib_ClickEncoder_Teensy_demo.ino`, which print out the menu on Serial and on display.

Here the bugs :

• On display, all the strings are not written (blank lines). Only one is visible: Exit
• On Serial and Display, the encoder is restricted to lines 1 and 2 of the menu, while it content 4 lines or more
• Linked to that, the cycling navigation does not work

Any ideas of what's going wrong with the navigation ? It was working well before.
Here the Sketch file.

[U8Glib_ClickEncoder_Teensy_Demo.zip](https://github.com/neu-rah/ArduinoMenu/files/418783/U8Glib_ClickEncoder_Teensy_Demo.zip)

christophepersoz, 2016-08-22:
Hi Rui,

Here a complete test of the v2.4 on Teensy. I made a video of that will show you the behavior of the v2.4 on Teensy. The video is based on the Sketch I sent you in my previous post.
Do you have an idea of what's going wrong - aside of the PROGMEM/Char* issue ?

[Test ArduinoMenu 2.4 on Teensy.mp4.zip](https://github.com/neu-rah/ArduinoMenu/files/429554/Test.ArduinoMenu.2.4.on.Teensy.mp4.zip)

neu-rah, 2016-08-22:
not yet, sorry
been working on 3.0
but i will look at it...
it was not a good idea to make so changes at once, this is getting though :D
PROGMEM was a big change it should have been a change for 3.0
learning...

neu-rah, 2016-08-22:
noticed that you are calling poll in parallel for multiple devices... this will not work because menu structures hold variable like top that are device specific and also because printing an options will clear its needRedraw state and cause it not to be drawn on the other device

this is one of the issues i'm dealing with version 3.0, trying to achieve device independence

please comment out one of the .poll's

    mainMenu.poll(gfx,allIn);
    mainMenu.poll(menu_out,Serial);

also on input initialization (I'm using c++ metaprogramming)

Stream* inputs[]={&enc,&Serial};
chainStream<2> allIn(inputs);

the number 2 is the length of inputs list (be sure they match)

christophepersoz, 2016-08-22:
Thanks for the explanations and your quick answer. I tried to comments the gfx output and fix the chainStream value (2 in my case) and you were totally right, the serial output works well.
On the gfx side, the navigation is almost OK when I comments the Serial, but, the Exit seems unreachable and disappear on maxIndex-1. And of course, I can't see anything except blank lines due to the PROGMEM issue on Teensy.

Do you think is that possible to have previous PROGMEM working version ?

neu-rah, 2016-08-22:
one way to check if its a PROGMEM problem is to use pure Serial as IO.

I have compiled here for nano (disabling progmem) and it seems to work, however on my device I had to set that contrast line to 128, u8g.setContrast(128); 0 or 255 would leave the screen black or white.
I'm also using the   u8g.setFont(u8g_font_unifont); because the other was not installed available, so please check also if its not a font problem.  
Also the wrap and exit function seem to work ok on serial input, didn't test with clickencoder...

christophepersoz, 2016-08-22:
Thanks for your feedback. What do you mean by use pure Serial ? I think that the Serial on Teensy is quite different than the one on Arduino (much much faster).

About U8Glib, the font I'm using is working well, otherwise I think that "Exit" won't shows up. I also know it works because I'm using the same font and parameters for my main program and other classes, which are working nicely. It seems it's much more behaviors linked to gfx displays rather than issues on Serial which is working perfectly now (I didn't know that I can't poll the menu on 2 different devices ;) )

I gong to have a closer look. Also the architecture of Teensy is quite different than the Arduino ones. I let you have a check on that : [PROGMEM on Teensy 3.1](https://forum.pjrc.com/threads/28126-PROGMEM-can-it-be-used-on-Teensy-3-1)

Also Paul Stoffregen mentioned that:
_"On Teensy 3.0, all you have to do is use "const" in the variable declaration. Or at least that's all you're supposed to do. Everything that's "const" gets put into flash without consuming any RAM._

_PROGMEM, pgm_read_byte() and the other AVR names are defined only so code designed for AVR can compile. But they do absolutely nothing on Teensy 3.0._

_I'd recommend adding const to that huge array. If you're not planning to compile on Teensy 2.0 or other AVR boards, you might as well remove the PROGMEM stuff. Accessing the array normally is much nicer and more readable code that using pgm_read_byte()."_

neu-rah, 2016-08-22:
the PROGMEM is important for small AVR's, they are short on RAM.

the point of using Serial for IO is to have a simple case, if it print on serial them PROGMEM stuff is ok.

its important to narrow down the problem cause.

for my compiles here, using just Serial or gfx+Serial they seem to work ok on nano (either using RAM or PROGMEM)

so, I'm a bit lost...

yeah const on C/C++ puts the data  on the program segment, for AVR its flash, and for AVR's flash addressing is separate from RAM address, hence the need pf pgm macros

for not using ram as you can see here https://github.com/neu-rah/ArduinoMenu/blob/master/src/macros.h#L9

all macros are just doing nothing

> PROGMEM, pgm_read_byte() and the other AVR names are defined only so code designed for AVR can compile. But they do absolutely nothing on Teensy 3.0.

then even using progmem on teensy should work, because they already do nothing, but we got a compile error instead

christophepersoz, 2016-08-22:
What I would say is that Teensy is an ARM architecture. Where AVRs have their RAM splitted in two different blocks and you need to use PROGMEM to move constant datas to PROGMEM instead of RAM. That's for ARM, and for sure it is important and save RAM on small boards.

On the other side, as you said, if it prints on Serial, it should be printed on GFX... I'm gonna have a look on the print section in the coming days, I will keep you in touch.

christophepersoz, 2016-08-22:
Ok, I found the problem. The issue wasn't on your library side, it was due to the call of the function `u8g.setFontPosTop();` inside the setup(). I have to admit that didn't find out immediately.
So you can definitively say that your lib is fully compatible with U8Glib on Teensy ;)

I will have a look later why this create so many problems, because there is a `gfx.setFontPosBottom(); // U8Glib font positioning` on `menuOut()` of `menuU8G.h` that should overcome that, but it seems it does not...

closed 2016-08-22 by christophepersoz