neu-rah/ArduinoMenu #41: menuLCDs.h error

dontsovcmc, 2016-08-16:
If compile I2C menu example, I get error in printPrompt function:
`
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h: In member function 'virtual void menuLCD::printMenu(menu&, bool)':
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:47:77: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'
       printPrompt(*(prompt*)pgmPtrNear(m.data[i]),i==m.sel,i+1,i-top,m.width);
                                                                             ^
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:47:77: note: candidate is:
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)
   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {
                ^
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:50:61: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'
     printPrompt(menu::exitOption,m.sel==m.sz,0,i-top,m.width);
                                                             ^
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:50:61: note: candidate is:
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)
   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {
                ^
/Users/dontsov/Documents/Arduino/libraries/ArduinoMenu-master/src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided
`
I prepare, you need add 'index' argument when call 'printPrompt':

```
printPrompt(*(prompt*)pgmPtrNear(m.data[i]),i==m.sel, i,   i+1,i-top,m.width);
printPrompt(menu::exitOption,m.sel==m.sz,  0,  0,i-top,m.width);
```

neu-rah, 2016-08-17:
thanks for noticing it, just pushed changes to deal with this issue

neu-rah, 2016-08-22:
is it solved?

dontsovcmc, 2016-08-22:
yes, thanks!

LuckyLuzz, 2016-09-13:
Hey,
I just downloaded 2.4 and tried i2c Menu. Looks like the changes didn't help. 

```
In file included from C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\examples\I2C_menu\I2C_menu.ino:21:0:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h: In member function 'virtual void menuLCD::printMenu(menu&, bool)':

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:47:85: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'
printPrompt(*(prompt*)pgm_read_ptr_near(&m.data[i]),i==m.sel,i+1,i-top,m.width); 

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:47:85: note: candidate is:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)
   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:50:61: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)
     printPrompt(menu::exitOption,m.sel==m.sz,0,i-top,m.width);

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:50:61: note: candidate is:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(promt&, bool, int, int, int, int)
   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided
exit status 1
Error compiling for board Arduino/Genuino Uno.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.```

LuckyLuzz, 2016-09-19:
solved it by copy that. Looks like it was not in the 2.4 release. 

printPrompt((prompt)pgmPtrNear(m.data[i]),i==m.sel, # i, i+1,i-top,m.width);
printPrompt(menu::exitOption,m.sel==m.sz, # 0, 0,i-top,m.width);

LuckyLuzz, 2016-09-19:
Seem like there is still a bug. Crashes on Exit in the second menu level. 

printPrompt(menu::exitOption,m.sel==m.sz, # 0, 0,i-top,m.width);

closed 2016-09-21 by neu-rah