neu-rah/ArduinoMenu #42: Create MultiDesktop config

dontsovcmc, 2016-08-21:
Thanks for the library, it's really perfect.
I want to create device with main screen and menu for change settings. I think, it's rather common aim.
As I see "custom menu drawing" in you API, I think, my idea is possible be solved by your library. Isn't it?

[http://ic.pics.livejournal.com/vostnod/12893397/243154/243154_900.png](url)

neu-rah, 2016-08-22:
at the current state it will be useful only to do the menu part, all the desktop part would have to be coded apart, but the menu can cooperate as you can call/suspend several menus whenever needed.
doing the desktop part poses some problems for the current version
1 - the menu system is not doing associations with button and actions, actions are only associated with menu prompts
2 - menu prompts are limited to one line

dontsovcmc, 2016-08-22:
If I code my idea only using your classes, it it  possible to use buttons associated with UP, DOWN codes when menu is not polling?

void loop() {
      if (!runMenu)  {
          if (allIn.read() == menu::enterCode) {
            runMenu = true;
            mainMenu.sel = 0; // reset the menu index fornext call
          } else if (allIn.read() == menu::upCode) {
            //do smth;
          } else if (allIn.read() == menu::downCode) {
            //do smth;
          }
    } else
      mainMenu.poll(menu_out,allIn); // temporary control and display on Serial Monitor
}

neu-rah, 2016-08-22:
yes, you can use input and output classes without/outside menu polling.
also the menu can do some association between buttons and actions, if your keyboard driver sends numeric characters for button press then the menu will call associated option.
options are numbered from 1 to 9 on serial menu and characters 1...9 have the effect of clicking the  option.

beware that each .read() removes the character, it would be better to do a single read and then test the character.

neu-rah, 2016-08-22:
```c++
void loop() {
      if (runMenu)  
          mainMenu.poll(menu_out,allIn); // temporary control and display on Serial Monitor
      else if (allIn.available()) {
          char ch=allIn.read();
          if (ch == menu::enterCode) {
            runMenu = true;
            mainMenu.sel = 0; // reset the menu index fornext call
          } else if (ch == menu::upCode) {
            //do smth;
          } else if (ch == menu::downCode) {
            //do smth;
          }
    }
}
```

dontsovcmc, 2016-08-22:
super!
I have 2 buttons... 
upCode & enterCode

I prepair add Exit option to exit from submenu and stop polling in main menu.
There is a same menu control in GoPro :)

dontsovcmc, 2016-10-28:
I done library:
https://github.com/dontsovcmc/ArduinoMultiDesktop

neu-rah, 2016-10-28:
one possible way to reuse a menu structure, there are others...

https://github.com/neu-rah/ArduinoMenu/blob/master/examples/reuse/reuse.ino

can be combined with menu suspension from other examples

closed 2017-03-01 by dontsovcmc