neu-rah/ArduinoMenu #43: Is keyStream debounce safe?

dontsovcmc, 2016-08-24:
As I understand (by practice and from source code), keyStream should be used only with non-debounce buttons... Can you add this info to wiki? 

Maybe It should be modified the same way: https://programmingelectronics.com/tutorial-19-debouncing-a-button-with-arduino-old-version ?

neu-rah, 2016-08-25:
yes, please
make it a different header file because sometimes debounce can be done in hardware, for that we can still use the actual keyStream.h
do a pull request with it and i will be glad to integrate it

dontsovcmc, 2016-09-02:
So, this is key driver for debouncing buttons.
It was difficult to understand you logic. 
[keySafeStream.h.txt](https://github.com/neu-rah/ArduinoMenu/files/452481/keySafeStream.h.txt)

I think, the keyLook works as buffer. So, if I call read(), the buffer flushed. And available() while not flushed() or read(). 
I check it with this sketch: 
[test_keySafeStream.ino.txt](https://github.com/neu-rah/ArduinoMenu/files/452486/test_keySafeStream.ino.txt)

closed 2017-03-01 by neu-rah