neu-rah/ArduinoMenu #44: embedXCode and ArduinoMenu build errors since 2.4

christophepersoz, 2016-08-26:
Hi Rui,

I'm keep going on my dev and since I stop the test and move back to my dev environment (Xcode 7.3.1 + embedXcode+ ) I'm getting some errors on build. I think the compiler is more restrictive than the one on Arduino IDE (I'm still looking for a way to change that).

Here are the errors that pops up on build (I got them several times, but it's always the same in a way).

1.` Error on promptAction,
Menu/src/macros.h:186:68: error: conversion from 'void()' to 'promptAction' is ambiguous`

2. `Menu/src/macros.h:179:41: error: 'menuTextField' does not name a type`
(it seems that the Macro for TEXTFIED isn't working)

3. `Menu/src/macros.h:178:94: error: invalid conversion from 'void (*)()' to 'bool (*)()' [-fpermissive]`
It seems that activate the -fpermissive option on the compiler can turn this error into warning (looking for that)

4. `Menu/src/menuFields.h:59:3: error:   initializing argument 8 of 'menuField<T>::menuField(T&, const char*, const char*, T, T, T, T, bool (*)()) [with T = unsigned char]' [-fpermissive]`

5. `Menu/src/menuFields.h:41:12: error:   initializing argument 3 of 'menuValue<T>::menuValue(const char*, T, promptAction) [with T = unsigned char]'`

6. `Menu/src/macros.h:190:34: error: '_menuTextField24' was not declared in this scope`
(I'm trying to use the TEXTFIELD() to have a project name edited, not sure that it works)

7. `Menu/src/macros.h:190:34: error: '_menuTextField26' was not declared in this scope`

8. `Menu/src/menu.h:124:12: error:   initializing argument 2 of 'prompt::prompt(const char*, promptAction)'`

Any idea or clue to solve that ? On the other hand I ask some help to embedXcode developper to find out a solution for -fpermissive on build which can solve a lot.

Thanks and have a great day !

neu-rah, 2016-08-26:
Hi! :D

no, textfield will not work for sure

https://github.com/neu-rah/ArduinoMenu/blob/master/controls/common/menuTextFields.h#L3

its some experimental code started for an old menu version and abandoned since... because I still have no idea what to do with it.

didn't even had a clear idea of how to edit text on that.

any clue?

christophepersoz, 2016-08-26:
Hi Ruy,

I saw the experimental line, but I told myself, let's give a try ;) So now I know it isn't working anymore !
That's true it seems quite complicated to edit text with only a rotary encoder, maybe using the way that car GPS are working when you are entering an address. Just a loud thinking :
ENTER EDIT MODE > 1st CHAR EDIT > BROWSE CHARS WITH ROTARY ENCODER > VALID BY BUTTON CLICK > JUMP TO NEXT CHAR, and so on.
Maybe the Char definition can be `"ABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789.-_"` plus 2 characters for BACKSPACE (is it really needed, you can also overlap the existing on edit), and one for exit. Or for exit editing take into account a long press. Does it help ?

Otherwise, even by removing the TEXTFIELD macro in my MENU declaration I still have the other errors on build. Any idea ?

neu-rah, 2016-08-26:
ending the edit:
1 - using special characters to bkspace, end is not available for some devices
2 - long press or double click is also not available for serial...
3 - double enter (without character change) can work on any device
4- device specific. serial can use plain text entry however but to be so, the method depends on the input device and not on the menu object, therefor all input devices should also provide text entry methods

i prefer the nr 4, still its harder to code and produces bigger code, and all input devices must provide one

?

neu-rah, 2016-08-26:
current code on numeric fields can accept numbers (over serial) by doing a float parse, this code is on the field class (not on the input), crossing input devices with menu objects will scale too much... hmm

christophepersoz, 2016-08-26:
Maybe the behavior can be different on Serial and others. If you have at least 3 button (up-left/down-right/enter) or a rotary encoder (faster), the methods 1,2,and 3 works well. The only issue is related to Serial.

Also, instead of thinking to backspace, maybe just a direct editing is enough. Just place the cursor at the right place (from beginning to the last know character), and overwrite the current value, no need to back space. Can be longer if you want to change a lot, but it's an idea.

neu-rah, 2017-08-30:
one year latter this feature request goes on at https://github.com/neu-rah/ArduinoMenu/issues/105

christophepersoz, 2017-08-30:
👍🏻

christophepersoz, 2017-09-05:
👍🏻

neu-rah, 2017-10-20:
its done :D

closed 2017-10-20 by neu-rah