neu-rah/ArduinoMenu #46: Strange behavior with ClickEncoderStream.h

christophepersoz, 2016-09-11:
Hi Rui,

I'm still going on my project and Teensy debug, and I'm still notice a bug I mentioned you when you were working on PROGMEM issue. I thought it was OK since then, but it is not ;)
Here the problem : 

• On Serial and Display, the encoder is restricted to lines 1 and 2 of the menu, even if the menu have more lines.
• The strange behavior is, if the line 2 is a FIELD, I can edit the value (limited to the first 2 values on a SELECT or a TOGGLE), validate it, and then the limitation shift of one line. So after that I can only select lines 2 and 3, and so on. The range is still limited to 2 lines.
• Linked to that, the cycling navigation does not work.

Any ideas of what's going wrong with the navigation ? It was working well on v2.1 if I remind me well.
I will check the values by putting some Serial outputs on QuadEncoder changes to see what's going on. If on your side you have an idea...

Thanks !
Christophe

christophepersoz, 2016-09-12:
I found the bug. It is due to a different way to sign Type under ARM and AVR. On AVR a int is normally signed if unspecified where it is unsigned on ARM, and that cause an issue on `sel` var which is int and should be `signed int` for ARM ;)
I'm going to pull a request on menu.h

neu-rah, 2016-09-12:
thank you :D

christophepersoz, 2016-09-12:
Maybe there is other issue due to this same problem on other var. Do you have other signed var that can have the wrong type due to this difference ?

LuckyLuzz, 2016-09-15:
Hey, 
due the name fits I will post my issue hear. I got a thing I am curious about. The Rotary Encoder Button is ignoring when it's pushed. The menu is not reacting, only when I turn the knob.
So if I am on "Exit" I have to push the btn two times and rotate "down" to actually exit, when I turn "up" it's jumping in the upper MenuEntry. Any ideas?

neu-rah, 2016-09-16:
can we see some code? is this about the included examples?

LuckyLuzz, 2016-09-16:
Hi, its`s not one of the examples. But stripped the code to it's Menu bones so now it could be one. Bug is still there. My guess is that there could be someting wrong with enc.read()==menu::enterCode. 

Thanks for helping!

IDE 1.6.11 Arduino Uno            ArduinomMenuLib Version 2.0, still can't compile 2.4.
```
//*-------------------------------------------------------------------------------------------------------------------------------------
// Define
//* ------------------------------------------------------------------------------------------------------------------------------------
#include <LiquidCrystal_I2C.h>
#include <menu.h>//menu macros and objects
#include <menuLCDs.h>//F. Malpartida LCD's
#include <menuFields.h>
#include "ClickEncoderStream.h"
#include <TimerOne.h>

#define CK_ENC  4                             // Encoder Input Button
#define DT_ENC  2                             // Encoder Input 
#define SW_ENC  3                             // Encoder Input 

const char        degree = 223;               // Degree symbol
bool              runMenu = false;
double            TempSetpoint = 23.00;
double            TempInput;
double            TempOutput  = 125.00;
double            TempAlarm   = 28;
byte              PID_AutoManOff = 1;
byte              PID_MAN   = 80;
byte              Xd        = 3;
byte              Xi        = 5;
byte              Xs        = 1;
byte              LightMaxY = 90;
byte              LightMinY = 10;
byte              LightOn = 9;
byte              LightOff = 21;
//*-------------------------------------------------------------------------------------------------------------------------------------
// Create object references
//* ------------------------------------------------------------------------------------------------------------------------------------
LiquidCrystal_I2C   lcd (0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);       // Create LCD object
menuLCD             menu_lcd(lcd, 16, 2);                               // Create MENU object
ClickEncoder        qEnc(DT_ENC, SW_ENC, CK_ENC , 4, LOW);              // Pass reference to ClickEncoder library
ClickEncoderStream  enc(qEnc, 1);                                       // Pass reference to ClickEncoderStream library
//*------------------------------------------------------------------------------------------------------------------------------------- 
// VOID's
//*-------------------------------------------------------------------------------------------------------------------------------------
void timerIsr()                                                         // for RotaryEncoder
{
  qEnc.service();
} 

void pauseMenu(prompt &p, menuOut &o, Stream &i)                         //Stop MENU     
{
  runMenu=false;
  o.clear();
}
//*------------------------------------------------------------------------------------------------------------------------------------
// MENU DEFINITION
//*-------------------------------------------------------------------------------------------------------------------------------------
MENU(subMenuTemperature,  "Temp/Light/PID",
     FIELD(TempSetpoint,   "SetPoint", "C", 15, 30, 1, 0),
     FIELD(TempAlarm,      "Alarm   ", "C", 15, 30, 1, 0),
     FIELD(PID_AutoManOff, "PID Mode   ", "", 0, 3, 1, 0),
     FIELD(Xd,             "PID Xd     ", "", 0, 10, 1, 0),
     FIELD(Xi,             "PID Xi     ", "", 0, 10, 1, 0),
     FIELD(Xs,             "PID Xs     ", "", 0, 10, 1, 0),
     FIELD(PID_MAN,        "PID MAN-Y  ", "%", 0, 100, 1, 0),
     FIELD(LightMaxY,      "PID MaX-Y  ", "%", 50, 100, 1, 0),
     FIELD(LightMinY,      "PID MiN-Y  ", "%", 0, 50, 0, 0),
     FIELD(LightOn ,       "Light On   ", "h", 0, 24, 1, 0),
     FIELD(LightOff ,      "Light Off  ", "h", 0, 24, 1, 0)
[... 39 lines cut ...]
          {
              if (enc.read()==menu::enterCode) //menu reenter method
                        {
                          menu_lcd.redraw();
                          runMenu=true;          
                         }
             else
                        {
                          lcd.setCursor(0, 0);
                          lcd.print(F("T="));
                          lcd.print(degree);
                          lcd.print(F("C"));
                          lcd.setCursor(10, 0);
                          lcd.print(F("Y="));
                          lcd.print(F("%"));
                         }
                    }
//*-------------------------------------------------------------------------------------------------------------------------------------
// END
//*-------------------------------------------------------------------------------------------------------------------------------------
}

[ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.15.zip](https://github.com/neu-rah/ArduinoMenu/files/476421/ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.15.zip)

```

neu-rah, 2016-09-16:
OK got your code working, not sure what was wrong, however with my experience coding the new version I've changed the pgm macros by removing the _near.
maybe that was the cause... anyway i'm sending the code where I've just changed the screen to match mine.
[ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.15.17.45.zip](https://github.com/neu-rah/ArduinoMenu/files/477404/ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.15.17.45.zip)

LuckyLuzz, 2016-09-16:
The pin 4 input was a thing I added after the bug appears, deleted it now. You changed my code to 2.4 right? With Arduino IDE I still cannot compile it, same error as before. Now I going to install Platform IO so we working on the same IDE. Also when I put the enc.read() on serial monitor it show numbers 43 for left, 45 for right turn but nothing for klick or double klick, always just -1 for nothing (return -1 form void Peek() )
I will try more in two days.

LuckyLuzz, 2016-09-19:
got it. It was so strange it couldn't be software. Somehow the encoder had a bug. replaced it and now everything is working... thanks for help! :)

closed 2016-09-19 by neu-rah

DangerousDaveUK, 2017-12-24:
Just looking at your code above LuckyLuzz, you have the Clickencoder setup the wrong way round.
From the Clickencoder library - ClickEncoder(int8_t A, int8_t B, int8_t BTN)
You have B and the BTN switched round in your code

doriancameron, 2020-05-23:
If anyone finds this because of the encoder issue like i did. Double check the wiring on the rotary encoder. Mine was not standard and took trial and error. Once wired correctly everything else was perfect. Thanks.