neu-rah/ArduinoMenu #49: setPosition() is no more working inside a SUBMENU

christophepersoz, 2016-09-12:
Hi ;)

I have notice in the latest version that setPosition() is no longer working when you are entering in a SUBMENU. So the position I'm defining just before the pool is only taken in account in the ROOT.

I've tried to change that in menuField.h, with no success. Do you have an idea ?

christophepersoz, 2016-09-13:
Do you have a ox and oy value attached to every MENU ? It seems that only the first main MENU have the good values and all the other ones have a value of 0.
Any way to change that  ?

neu-rah, 2016-09-13:
just assign to them...
```c++
//a submenu
MENU(ledMenu,"LED on pin 13",
    OP("LED On",ledOn),
    OP("LED Off",ledOff)
);

ledMenu.ox=10;
ledMenu.oy=20;
```

christophepersoz, 2016-09-13:
Oh, didn't thought to that. Thanks.
Is there a way that SUBMENU inherit their position from ROOT MENU ? - I tried but it does not work.

neu-rah, 2016-09-13:
maybe we could make that relative coordinates... but it would require traversing the all parents

christophepersoz, 2016-09-13:
Does the sibling can acces it easily ?

neu-rah, 2016-09-13:
nope, and none is easy because prompt does not refer to parent object, it must be supplied by calling function, siblings can only be accessed if parent object is supplied and still has to iterate the parent data list

christophepersoz, 2016-09-13:
Hmm, understood. I will set the sibling MENU individuals though ;). Thanks

closed 2016-09-13 by christophepersoz