neu-rah/ArduinoMenu #51: chainStream.h peek and read returns -1 on Teensy

christophepersoz, 2016-09-13:
Hi Rui,

I discover that chainstream.h and particularly peek() and read() returns -1 every twice move or click when used with ClickEncoder. The clickencoderstream seems working well, but when I'm tracing on Serial the values returned by chainstream there is this -1 that creates missed clicks or move of the encoder. At this time the workaround I found is to replace the -1 by 1 ( not really academic I admit). I don't know if this is linked to ARM platform. I don't have a Arduino to test it. 

Thanks

neu-rah, 2016-09-13:
-1 is returned when no value available, but we can check if values are available before read or peek

christophepersoz, 2016-09-13:
I thought that too, and that I understood by examining the code. But in the fact, on the first click I got -1, on the second 42 for example if I'm tracing the values.

neu-rah, 2016-09-13:
even when calling .available() on the stream?

christophepersoz, 2016-09-13:
Yes! So curious

christophepersoz, 2016-09-13:
I can add this to help :

• If a restore the `return -1;` inside peek() of ClickEncoderStream.h with only one input as chainStream<1> I have the limiting bug to two values that comes back, giving me a read() changing to 43 to 45.
• When I'm putting `return 1`; inside peek() of ClickEncoderStream.h with one or multiples input in `chainStream<1>` or `chainStream<2>` with encoder + serial input, everything is working fine, except that the stream is continuous instead of parse the stream when new datas are available.

What I'm parsing is the `streams[n]->peek` value of peek() and `key` value of read() inside chainStream.h

neu-rah, 2016-09-21:
guess this is solved, right? if not pls reopen

christophepersoz, 2016-09-21:
It is not, I just kept the return to 1 instead of -1 to have the right behavior.

closed 2016-09-25 by neu-rah

christophepersoz, 2016-09-29:
I'm gonna test that tomorrow ;) Thanks !

neu-rah, 2016-09-29:
hope it work, if not you know.. just reopen this.
thanks :D

christophepersoz, 2016-09-29:
It works ! ;) Thanks a lot