neu-rah/ArduinoMenu #52: What's to change when updating 2.0 to 2.4?

LuckyLuzz, 2016-09-13:
Hey, 
got a bit time and wanted update to 2.4 cause of PROGMEM feature. Got a lot compile errors. Some known things to change in my project?

Thanks!

```
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino:14:0:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h: In member function 'virtual void menuLCD::printMenu(menu&, bool)':

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:47:85: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'

       printPrompt(*(prompt*)pgm_read_ptr_near(&m.data[i]),i==m.sel,i+1,i-top,m.width);

                                                                                     ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:47:85: note: candidate is:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)

   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {

                ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:50:61: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'

     printPrompt(menu::exitOption,m.sel==m.sz,0,i-top,m.width);

                                                             ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:50:61: note: candidate is:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)

   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {

                ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided

In file included from C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:36:0,

                 from C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino:13:

C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino: At global scope:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:162:41: error: conversion from 'void()' to 'promptAction' is ambiguous

   prompt op##cnt(title_##cnt,__VA_ARGS__);

                                         ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:117:17: note: in expansion of macro 'DECL_OP_'

 #define DECL(x) DECL_##x

                 ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:12:34: note: in expansion of macro 'DECL'

 #define FOR_EACH_1(what, x, ...) what(x)

                                  ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:15:3: note: in expansion of macro 'FOR_EACH_1'

   FOR_EACH_1(what,  __VA_ARGS__)

   ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:18:3: note: in expansion of macro 'FOR_EACH_2'

   FOR_EACH_2(what, __VA_ARGS__)

   ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:10:35: note: in expansion of macro 'FOR_EACH_3'

 #define CONCATENATE2(arg1, arg2)  arg1##arg2

                                   ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:112:32: note: in expansion of macro 'FOR_EACH_'

 #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)

                                ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/macros.h:121:3: note: in expansion of macro 'FOR_EACH'

   FOR_EACH(DECL,__VA_ARGS__)\

   ^

C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino:95:1: note: in expansion of macro 'MENU'

 MENU(subDateTime, "Date/Time",

 ^

C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino:76:6: note: candidates are:

 void setTimeRTC()

      ^

In file included from C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino:13:0:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:109:10: note: promptAction::promptAction(promptAction::callback) <near match>

   inline promptAction(callback f):hFn(f) {}

          ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:109:10: note:   no known conversion for argument 1 from 'void()' to 'promptAction::callback {aka bool (*)(prompt&, menuOut&, Stream&)}'

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:108:10: note: promptAction::promptAction(bool (*)(prompt&, menuOut&)) <near match>

   inline promptAction(promptFeedback (*f)(prompt&,menuOut&)):hFn((callback)f) {}

          ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:108:10: note:   no known conversion for argument 1 from 'void()' to 'bool (*)(prompt&, menuOut&)'

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:107:10: note: promptAction::promptAction(bool (*)(prompt&)) <near match>

   inline promptAction(promptFeedback (*f)(prompt&)):hFn((callback)f) {}

          ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:107:10: note:   no known conversion for argument 1 from 'void()' to 'bool (*)(prompt&)'

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:106:12: note: promptAction::promptAction(bool (*)()) <near match>

     inline promptAction(promptFeedback (*f)()):hFn((callback)f) {}

            ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:106:12: note:   no known conversion for argument 1 from 'void()' to 'bool (*)()'

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menu.h:125:12: error:   initializing argument 2 of 'prompt::prompt(const char*, promptAction)'

     inline prompt(const char * text,promptAction action)

            ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

neu-rah, 2016-09-13:
the prompt actions should return a bool since 2.3

https://github.com/neu-rah/ArduinoMenu#23

LuckyLuzz, 2016-09-13:
Ok got that, we are getting closer. I guess the rest of it is related to the other bug.

[https://github.com/neu-rah/ArduinoMenu/issues/41]

```
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Thomas\Documents\Arduino\Projects\Aquarium\ArduinoMenu_I2C_LCD_16x2_RotaryEncoder\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13\ArduinoMenu_I2C_LCD_RotaryEncoder-2016.09.13.ino:14:0:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h: In member function 'virtual void menuLCD::printMenu(menu&, bool)':

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:47:85: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'

       printPrompt(*(prompt*)pgm_read_ptr_near(&m.data[i]),i==m.sel,i+1,i-top,m.width);

                                                                                     ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:47:85: note: candidate is:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)

   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {

                ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:50:61: error: no matching function for call to 'menuLCD::printPrompt(prompt&, bool, int, int, int&)'

     printPrompt(menu::exitOption,m.sel==m.sz,0,i-top,m.width);

                                                             ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:50:61: note: candidate is:

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note: virtual void menuLCD::printPrompt(prompt&, bool, int, int, int, int)

   virtual void printPrompt(prompt &o,bool selected,int idx,int posX,int posY,int width) {

                ^

C:\Users\Thomas\Documents\Arduino\libraries\ArduinoMenu-2.4\src/menuLCDs.h:35:16: note:   candidate expects 6 arguments, 5 provided

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

LuckyLuzz, 2016-09-13:
The other question I got is how to enable PROGMEM or better on which devices is it enabled?

#define USEPGM ?? Is that all?

neu-rah, 2016-09-13:
on 2.4 its automatic if your uC has flash memory and pgm macros then flash will be used

closed 2016-09-21 by neu-rah