neu-rah/ArduinoMenu #55: FIELD and Double

LuckyLuzz, 2016-09-22:
Hi,
is it possible to change a double value behind the floating point with "Tune"? Like 28.30? Also I d like to ask if there is a way to hide the numbers behind the point like with lcd.print(TempSetpoint,**0**); ?

double            TempSetpoint  = 23.00;
FIELD(TempSetpoint,   "SetPoint", "C", 15, 30, 1, 1);

Thanks!
FIELD(variable,name,units,min,max,step,tune,function)

neu-rah, 2016-09-23:
it should be, by deriving the menuField class and re-implementing its print method... but to use it one should use no macros, because macros always pick up the original menuField macro.
This reminds me to include an alternative macro where one can specify alternative classes for every element, allowing user extension.

LuckyLuzz, 2016-09-23:
looking forward to version 3 :)

LuckyLuzz, 2016-12-15:
Hallo Rui,
any progress with this enhancement for Version 3?

Regard :)

neu-rah, 2016-12-15:
Hi! :)
V3 is usable and examples include a special field with custom print

LuckyLuzz, 2016-12-17:
Sounds great!! :)
When do you go to reales v3?

neu-rah, 2016-12-20:
welp... I'm a bit stuck on the esp8266 menu over wifi webserver + websocket... and then there are the docs. Have not started to write them yet.
Hope i can have it on Feb 2017, but that will depend on some yet unknown variables as how busy will i be with other stuff that i MUST do.
apart from esp8266 stuff the v3 works pretty well... but ofcoz, no docs yet.

LuckyLuzz, 2016-12-20:
esp+wifi+Menu, thats sound awesome! I recently have read a lot about that kinda stuff. You gone add  support for the esp so that it will be possible to control the menu over wifi as well as over a lcd+rotary encoder? 
If you do that this will be the most powerfull MenuLib known! Offline and Online control as well as better visualisation of the menu in a webserver, right?    I am totally looking forward to that chances! :)
What will be the hardware structur then  lcd - arduino - esp OR lcd - esp ?

neu-rah, 2016-12-20:
right know its possible to use the menu (v3) on esp8266-arduino using serial IO and guess that also using encoder/screen, i'm implemented output to web browser and websocket input from web browser to menu control. current examples on v3 are more like tests and i have to put all that together.

neu-rah, 2017-03-01:
v3.x allows custom option printing, included a "special prompt" in some examples

closed 2017-03-01 by neu-rah