neu-rah/ArduinoMenu #56: Question on menuU8G.h and coloring

christophepersoz, 2016-09-30:
Hi Rui,

Yesterday a question pop-up to me and after a quick look I'm not sure that I can solve it easily, that's why Im' asking. For multicolor display or gray shade display, do you think it is easily possible to have a different color between the "Name" and the "Variable" for FIELD, TOGGLE (Name of the Toggle, and the VALUE), SELECT, CHOOSE ?
I'm not sure because it's all generated in menuFields.h and by printTo() or printPrompt() (not sure about that), which means that I should derivate or create a new menuFields.h in which I can pass a parameter about the color I want to display - Am I right ?

Thanks

neu-rah, 2016-09-30:
the only possible way right now is to derive the menuField (or other variant) and implement specific printTo.. but for that to work on must not use the macrtos (macros are wired to the original classes).

same as in here https://github.com/neu-rah/ArduinoMenu/issues/55

on v3 i have macros to use alternate classes, and i can include extra colors on the base classes.

another possibility is to include extra color variables, use them on the printTo, and push the changes to the repository :D if you kind enough.

christophepersoz, 2016-09-30:
Hmm I read that issue #55 when it was posted. So it's the same. No macro, I'm a bit afraid regarding the size of my menu to change everything in no macro mode :D
I'm more interested by changing the printTo method. About that, just a question. gfx only exist in menuU8G.h, and when you are creating the menu instance, does menuField.h inherit of gfx and the attributes of u8g or not ?

neu-rah, 2016-09-30:
it does not inherit, for this to work well the printTo would have to be split into 3 parts name|value|unit
as it is a virtual function for non fields the value|unit would just do nothing

there are other possibilities... like adding a virtual function setColor to menuOut that would do nothing on most devices (as setCursor) and move the color setting to be non gfx specific.

i prefer the first

christophepersoz, 2016-09-30:
So you mean that printTo should be included in menuU8G.h in order to work and understand the gfx properties and methods ?
Now I have to learn what is a virtual function... :)

neu-rah, 2016-09-30:
hehehe... classes derivations and virtual functions

i can do this its pretty straight forward.

just curious, is C/C++ your first language?

christophepersoz, 2016-09-30:
I'm totally newbie, programming is not my work neither my studies, but I like challenges and learn, so... that way, I have no first language, but know a bit of php, js, c++ now since arduino.

If I understood what I read, I can create a virtual class inside menuU8G, named differently, and based on printTo() which is virtual and which allow me to add dedicated properties and function of u8glib, right ?

neu-rah, 2016-09-30:
classes can derive from other classes, this is useful to establish basic behavior (on base classes) and specific behavior on derived ones, code can call base functionality and run specific behavior depending on what specific derived class is selected.

adding new methods (behaviors) is not very useful because other code would not call it (it will call base class methods).

but we can always redefine existing methods (virtual).

so virtual functions/methods are functions that can be redefined and each object will run its own class version. This way code can be made abstract. its even allowed for the base class to provide JUST the definition of a virtual functions making it =0 (no function body), this are called pure virtual functions.

christophepersoz, 2016-09-30:
Interesting... I'm far from that, but I imagine that it's extremely powerful and convenient.

So if I'm creating a virtual methods called printTo() inside menuU8G.h it will be understood by this derivate class and it could refers to the base classes. Am I still right ?

Thanks for the explanations, really.

christophepersoz, 2016-09-30:
Or maybe I should create 3 new printTo() virtual methods in `menuFields.h`, like `printToTxt()`, `printToValue()` and `printToUnit()` and then a virtual methods in `menuU8G.h` that call those methods instead of just using `printTo()` inside `printPrompt()`.

I'm gonna try it ;)

neu-rah, 2016-09-30:
I've created a new branch (valuecolor) it compiles ok here but i cannot test it because i need to repair my raspi sdcard (it controls my breadboard power supply)
on this branch menuField should print value with color, for that i've changed menuU8G to have 2 extra colors (valueColor and valueColorHi) to be used when printing the fields. Also addded printName, printValue and printUnit to base class prompt with the original printTo just calling the new 3 functions.

On menuU8G the printPrompt now calls the 3 new functions changing color between instead of just calling printTo

if this works we can implement the 3 new functions on select, toggle and choose also.

To be the most correct (clean inplementation) colors should be enumerated at menu base definition and base menu definition should provide color change functions instead of this implementation...
but i will leave correctness for v3.

the pain of this is c++ is this inter-dependency of stuff and cross calling virtual functions + template syntax hell and need of macros to patch C++ poor allocation syntax.

I've used C++ for 20+ years, its not my studies field also but i just luv codding and do web development and system maintenance for a living.

C++ was my favorite language for those 20+ yrs.. until i met haskell :D
I'm also fluent on JS and can so python or assembly among other languages.

christophepersoz, 2016-09-30:
Thanks ! I'm going to check that a bit later, I've to go. Thanks a lot for your help. That way I could compare and check what I did wrong ;)

christophepersoz, 2016-09-30:
Hi Rui!

You rock, it does work, and it does improve the reading on screen, that's great. Is it hard to extend it to select, google and choose ? Because I don't see any printTo() for those ones...

I just wandering why do you need to declare 2 new colors instead of using the ones already declared ?
It seems that the 20+ years of c++ are useful ;) Templates and Virtual and your Macro still a bit mysterious to me ^^

Thanks a lot

neu-rah, 2016-10-01:
the others derive from menuVariant

https://github.com/neu-rah/ArduinoMenu/blob/valuecolor/src/menuFields.h#L155

welp i used 2 new color (normal and high-lite) independent from other just because i didn't knew what colors to use if recycling old colors.. they all have some specific meaning and recycling sometimes i could end up with a bad combination of low or no contrast.

christophepersoz, 2016-10-01:
Hmm, you are right for the color, it's all about a contrast ;)
I going to try to adapt the menuVariant and see if it's working here, step by step, following what you did for FIELD.

christophepersoz, 2016-10-01:
OK, that is working 💃 
Is there a way to have a different treatment on OP ? I would like to have the same color on OP text that I have on value for FIELD?

On the other and, there is a bug that appears on ClickEncoderStream now, the first click does not work, I have first to change the encoder position down/up to activate the click... It wasn't on the previous version, so I have to check the differences between the two.

christophepersoz, 2016-10-01:
For the first click, it seems that there is something missing on init. When I put a Serial out in ClickEncoderStream.h, it never reach the condition until I move the encoder. Weird. 

```c++
if (btn == ClickEncoder::Clicked)
        {
            Serial.print("Clicked");
            return menu::enterCode;
        }
```

christophepersoz, 2016-10-01:
Hi Rui,

I made some search about the first click issue and here the results (all on Teensy 3.2 of course, I don't know the behaviors on Arduino):

• The ClickEncoder.cpp works well, all event are returned well, first click also
• The ClickEncoderStream.cpp have an issue on first click, it needs to have a up/down first (ch == 43 or ch==45) before seeing any changes on btn == ClickEncoder::Clicked returned true (ch==42)

Do you think that there any kind of bad init or bad buffering that hang the digitalRead() and does not communicate it the the Stream ? I think so, but I did not found why.

I checked if it could be a signed/unsigned var issue, but it is not. Even by changing all the int var/functions that return -1 and should be signed, into signed var/function give the same results. (ARM compile by default int into unsigned int and that may cause troubles ^^).

Otherwise, I still not found how to know if the menu item is an OP to display it differently.

See you in a bit, I'm leaving for few days ;)
Thanks Rui

christophepersoz, 2016-10-20:
Hi Rui,

Back and testing the last enhancement of this version. I notice a bug since we introduced the color change on name and value. For CHOOSE, SELECT and TOGGLE the value displayed is not the correct one but the value[0], it seems the the index is not considered. I haven't noticed that when I was focused on design, but now that I'm back to functionality, I saw that all those values are not displayed correctly. Any idea ?

Thanks !

neu-rah, 2016-10-20:
yes, I've only implemented it on field as an example... that need to be transposed to choose, select and toggle.
but its too much trouble, I've implemented color support at the base of version 3 as an enumeration (completely different process). Version 3 is in the dev branch... at it is in a very rude form right now.

christophepersoz, 2016-10-20:
Hum, in fact it's more like the first display of the menu that is not OK, as soon as I change a value, leave the menu and go back in it, the changed value is correctly displayed. Not the other that remain wrong on screen, but good in code.

christophepersoz, 2016-10-20:
Do you think that you can have a look to make the initialization works on the current version (2.4 I guess) or is it too messy to have it works ?

neu-rah, 2016-10-20:
have no idea how messy it can be.. but will check the "messiness" and tell you :D

neu-rah, 2016-10-21:
quite a mess
- my ST7735 color display is not supported by U8Glib (only adafruit gfx)
- the 2.8 inch TFT LCD 240x320 is only for arduino due ILI9325 -> UTFT (https://pt.aliexpress.com/store/product/2-8-inch-2-8-TFT-LCD-Module-Display-ILI9325-240x320-with-Touch-panel-SD-card/1194705_32251525148.html?detailNewVersion=&categoryId=400401)

the only display i have here for U8Glib is the the PCD8544 -> and that is a monochromatic display

plus the color example is only implemented in U8Glib (other libraries are just ignoring the color)
 
to implement the color i've just added some color values to U8Glib https://github.com/neu-rah/ArduinoMenu/blob/valuecolor/src/menuU8G.h#L30 and use the color on U8Glib printPrompt https://github.com/neu-rah/ArduinoMenu/blob/valuecolor/src/menuU8G.h#L88

to be able to do that i've splited the prompt::printTo into several functions https://github.com/neu-rah/ArduinoMenu/blob/valuecolor/src/menu.h#L128

printName, printValue and printUnit

so for other menu objects to use the color schema they will have to implement this functions when apropriate (thats the case of toggle, select, choose).

menufield implements them here https://github.com/neu-rah/ArduinoMenu/blob/valuecolor/src/menuFields.h#L71

so all menuVariant's would have to implement the functions printTo, printName, printValue and printUnit in a similar way.

I've done that in a color blind way because of my monochrome display.

to do otherwise i would have to implement this color schema in adafruit's or UTFT library.

and to be honest its a very complicated schema involving many virtual functions because color formats are device specific.

using color enumeration is the way of doing it, but would break all API compatibility, while the current schema preserves the API.

christophepersoz, 2016-10-21:
Hi Ruy. 

Understood. I have to show you the actual state of the menu, you will see that is really well working. The only issue I'm seeing is the unit value shown as I told you ;)
By the end of the day I will post it tonight.

neu-rah, 2016-10-21:
yes, please do so because i'm not getting the initial status part

christophepersoz, 2016-10-22:
Hi Rui,

Sorry but Github was down yesterday night. Here the video. Please note that I'm on version valuecolor, not on the official version ;)

The first video show you changes I'm making on vars. They are stored in EEPROM after exit menu.
When I'm returning inside the menu, they are well displayed, as they have been changed.
[OK on reenter Menu.mp4.zip](https://github.com/neu-rah/ArduinoMenu/files/545673/OK.on.reenter.Menu.mp4.zip)

The second video shows you after a restart, the values are restored from EEPROM (I check that on Serial output), but when I'm calling back the menu, the menu display the [0] values in SELECT, TOGGLE, CHOOSE, instead of the stored one.
[BAD on restart.mp4.zip](https://github.com/neu-rah/ArduinoMenu/files/545680/BAD.on.restart.mp4.zip)

Does it helps you ?
Thanks a lot.

neu-rah, 2016-10-22:
Hi :D

ahh, i see, so its not a color issue, its a value issue

value changes are not correctly reflected after eeprom restore, correct?

christophepersoz, 2016-10-22:
Exact ;) They are not good on the first menu call.

neu-rah, 2016-10-22:
reflections seem to work well on uno/nano, maybe it is a teensy issue, can you paste the menu definition of LCD_Demo example and load it on teensy using your display?

neu-rah, 2016-10-22:
or at least this part 

```c++
#define LEDPIN 13
int ledCtrl=0;

bool ledOn() {
  Serial.println("set led on!");
  digitalWrite(LEDPIN,ledCtrl=1);
  return false;
}

bool ledOff() {
  Serial.println("set led off!");
  digitalWrite(LEDPIN,ledCtrl=0);
  return false;
}

TOGGLE(ledCtrl,setLed,"Led: ",
    VALUE("On",HIGH,ledOn),
    VALUE("Off",LOW,ledOff)
);

MENU(mainMenu,"Main menu",
  SUBMENU(setLed),
  OP("LED On",ledOn),
  OP("LED Off",ledOff)
);

```

christophepersoz, 2016-10-22:
It is quite long but...

It is quite long, but here it is
```c++
/* _____ MENUS DEFINITION _____ */
const uint8_t _fontW = 7;
const uint8_t _fontH = 9;
const uint8_t _cBack = 0;
const uint8_t _cDisable = 1;
const uint8_t _cEnable = 2;
const uint8_t _cHighlight = 3;
//const u8g_fntpgm_uint8_t _font * = "u8g_font_03b_04e"; // font used for the menu

// Page A - PROJECTS
TOGGLE( _indSCN, idxScn, "INDEX:",
       VALUE( "-", 0, updTAKES ),
       VALUE( "A", 1, updTAKES ),
       VALUE( "B", 2, updTAKES ),
       VALUE( "C", 3, updTAKES ),
       VALUE( "D", 4, updTAKES ),
       VALUE( "E", 5, updTAKES ),
       VALUE( "F", 6, updTAKES ),
       VALUE( "G", 7, updTAKES ),
       VALUE( "H", 8, updTAKES )
       );
/*
MENU( setClap, "SET CLAP",
     FIELD( _noSCN, "SCENE NUMBER :","", 1, 255, 1, 0, doNOTHING ),
     FIELD( _takeSCN, "TAKE NUMBER :", "", 1, 255, 1, 0, doNOTHING ),
     SUBMENU( idxScn )
     );
*/

CHOOSE( _idLOADSCN, loadPrj, "LOAD FROM <",
       VALUE( "<< BACK", -1, exitPRJMEM),
       VALUE( "BANK A", 1, loadPRJ ),
       VALUE( "BANK B", 2, loadPRJ ),
       VALUE( "BANK C", 3, loadPRJ ),
       VALUE( "BANK D", 4, loadPRJ ),
       VALUE( "BANK E", 5, loadPRJ ),
       VALUE( "BANK F", 6, loadPRJ ),
       VALUE( "BANK G", 7, loadPRJ ),
       VALUE( "BANK H", 8, loadPRJ )
       );

CHOOSE( _idSAVESCN, savePrj, "SAVE IN >",
       VALUE( "<< BACK", -1, exitPRJMEM),
       VALUE( "BANK A", 1, savePRJ ),
       VALUE( "BANK B", 2, savePRJ ),
       VALUE( "BANK C", 3, savePRJ ),
       VALUE( "BANK D", 4, savePRJ ),
       VALUE( "BANK E", 5, savePRJ ),
       VALUE( "BANK F", 6, savePRJ ),
       VALUE( "BANK G", 7, savePRJ ),
       VALUE( "BANK H", 8, savePRJ )
       );

MENU( mPageA,"PROJECTS",
     FIELD( _idSCN, "SCENE IDENTIFIER:","", 1, 127, 1, 0, updTAKES ),
     SUBMENU( idxScn ),
     FIELD( _noSCN, "SHOOT NUMBER:","", 1, 255, 1, 0, updTAKES ),
     FIELD( _tkSCN, "TAKE NUMBER:", "", 1, 255, 1, 0, doNOTHING ),
     SUBMENU( loadPrj ),
     SUBMENU( savePrj )
     );

// Page B - MOTION
MENU( linearMove, "LINEAR MOVE",
     FIELD( _mMAXLENGTHMM, "MAX LENGTH RANGE:", "mm", 50, 6000, 100, 10, updLINEARMOVE),
     FIELD( _mWHEELDIAM, "WHEEL DIAMETER:", "mm", 5, 250, 10, 1, updLINEARMOVE)
[... 122 lines cut ...]
       VALUE( "NO", ACT::OFF )
       );

MENU( mPageF, "SYSTEM",
     SUBMENU ( autoSaveEE ),
     SUBMENU( reloadSettings )
     );

// MAIN Menu
MENU( mainMenu, "MAINMENU",
     SUBMENU( mPageA ),
     SUBMENU( mPageB ),
     SUBMENU( mPageC ),
     SUBMENU( mPageD ),
     SUBMENU( mPageE ),
     SUBMENU( mPageF ),
     OP("<< EXIT", menuEXIT)
     );

ClickEncoderStream _QENCMENU(_QENC, 1);// simple quad encoder fake Stream
Stream* _menuInputs[]={&_QENCMENU};
chainStream<1> allIn(_menuInputs);

menuU8G _menuU8G(_U8G, _cHighlight, _cBack, _cEnable, _cDisable, _fontW, _fontH);
```

neu-rah, 2016-10-22:
ok, here is a sketch (using serial IO) that updates the menu with loaded settings
the trick is to set the output device ".drawn=NULL" on the function that loads the settings
this causes the menu to display the new values.

the functions obtains access to the output device by parameters, ex:

    bool loadA(prompt &p, menuOut &o, Stream &i);

the complete sketch:

```c++
#include <Arduino.h>

/********************
Arduino generic menu system
U8GLib menu example
U8Glib: https://github.com/olikraus/U8glib_Arduino
http://www.r-site.net/?at=//op%5B%40id=%273090%27%5D

Jul.2016 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
creative commons license 3.0: Attribution-ShareAlike CC BY-SA
This software is furnished "as is", without technical support, and with no
warranty, express or implied, as to its usefulness for any purpose.

Thread Safe: No
Extensible: Yes
*/

#include <menu.h>//menu macros and objects
#include <menuPrint.h>
#include <menuFields.h>

/* _____ MENUS DEFINITION _____ */
enum nSTEPS {_200,_400,_800};

int _mMAXSPD=1;
int _mGEARBOX=1;
nSTEPS _mSTEPS=_200;
int _hCURRENTMODE=0;

bool updMCONTANTS() {return false;}
bool updMKVALS() {return false;}

// Page E - HARDWARE
TOGGLE( _mSTEPS, nbMSteps, "STEPPER RESOLUTION:",
       VALUE( "200 steps", nSTEPS::_200, updMCONTANTS ),
       VALUE( "400 steps", nSTEPS::_400, updMCONTANTS ),
       VALUE( "800 steps", nSTEPS::_800, updMCONTANTS )
       );

TOGGLE( _hCURRENTMODE, holdCurrents, "HOLD CURRENTS STRENGTH:",
      VALUE( "LIGHT", 0, updMKVALS ),
      VALUE( "STANDARD", 1, updMKVALS ),
      VALUE( "STRONG", 2, updMKVALS )
      );

bool loadA(prompt &p, menuOut &o, Stream &i);
bool loadB(prompt &p, menuOut &o, Stream &i);

// MAIN Menu
MENU( mainMenu, "MAINMENU"
     ,SUBMENU( nbMSteps )
     ,SUBMENU( holdCurrents )
     ,FIELD( _mMAXSPD, "MAX SPEED:", " stp/s", 1, 8000, 100, 5, updMCONTANTS )
     ,FIELD( _mGEARBOX, "GEARBOX RATIO 1 :", "", 1, 255, 5, 1, updMCONTANTS )
     ,OP("Load settings A",loadA)
     ,OP("Load settings B",loadB)
     );

bool loadA(prompt &p, menuOut &o, Stream &i) {
 _mMAXSPD=4000;
 _mGEARBOX=100;
 _mSTEPS=_400;
 _hCURRENTMODE=1;
 o.drawn=NULL;
 return false;
}

bool loadB(prompt &p, menuOut &o, Stream &i) {
 _mMAXSPD=8000;
 _mGEARBOX=255;
 _mSTEPS=_800;
 _hCURRENTMODE=2;
 o.drawn=NULL;
 return false;
}

void setup()   {
  Serial.begin(115200);
  menu::wrapMenus=true;
}

menuPrint menu_out(Serial);//describe output device

void loop()
{
  mainMenu.poll(menu_out,Serial);
}
```

christophepersoz, 2016-10-22:
Hi Rui,

Thanks for the example, that can be really useful ! But... why using a load function for parameters ? When I'm loading var values in the application it's done inside setup(). Once I'm in loop() I'm just changing some of them, save it in EEPROM for the next restart but no more. That's why I don't understand why the loaded values are not shown as it should in Menu.

Like if your previous example you had _ hCURRENTMODE = 2; in setup() but when you are displaying the menu in loop() you got on display :
HOLD CURRENTS STRENGTH: LIGHT (corresponding to 0 instead of 2).

neu-rah, 2016-10-22:
**not in setup, in loadB**

the _ hCURRENTMODE is initialized to 0 on declaration

    int _hCURRENTMODE=0;

whenever you change values you can set the output device .drawn member to NULL to force a redraw on next poll. In the example I used a load function just to wire the actions, but they can be changed anywhere as long as you invalidade the output .drawn

the values should be automatically updated... but somehow they are not, therefor the need of the output invalidation (force menu redraw). Its probably a bug and the example is a workaround. so please do:

    <your_menu_output_device>.drawn=NULL

just after you load the parameters from eprom

if you initialize the parameters on setup then be sure that the eeprom load is done after.

christophepersoz, 2016-10-22:
Ok, I'm gone try that tonight and I keep you in touch...  ;)

christophepersoz, 2016-10-22:
Back then ;) So I did try `_menuU8G.drawn = NULL;` at the end of the `setup()` function but it seems that the Menu values are still not updated when entering in Menu :/

neu-rah, 2016-10-22:
and teh values are not being initialized/reset after the eeprom load? can we print the values and check they are different than what is on the menu?

can I see more code?

christophepersoz, 2016-10-23:
Hi Rui, yes the reset happens after the EEPROM load. I'm currently trying to trace the values.

I can sent you the complete code, but privately - Can I do that on the email includes in your ArduinoMenu code ?

christophepersoz, 2016-10-23:
I made some serial trace to understand what's going on. When I'm tracing `op` and `sel` in `menuFields.h` and entering in SELECT or CHOOSE for the first time, `sel = 0` and `op = -2`, always - whatever the value stored in program; 
Once I changed the value (set to 2 fro example), leave the menu to store it in EEPROM and go back to the menu (without doing a board reset), going back on the same item, I'm getting the right value displayed and `sel = 2` and `op = -2`. So `sel` value is correctly set at this moment.
I traced the value stored in EEPROM, and the value and the behavior of the program is correct and reflecting the value stored in EEPROM, so it's really a matter of how the values are took into account in the menu once they are updated by the EEPROM load. Can it be a pointer issue ?

neu-rah, 2016-10-23:
serial trace at the end of the EEPROM load?

christophepersoz, 2016-10-23:
Yes, serial trace of the menu behavior inside the loop() after the EEPROM load (setup) and _menuU8G.drawn = NULL;

I also have a trace on setup() to be sure that the value loaded from EEPROM are correct.

neu-rah, 2016-10-23:
so the values on trace are correct but menu shows them wrong, even after moving the cursor on menu.

right?

neu-rah, 2016-10-23:
this issue title was about color, so we should be discussing this on a new issue for the sake of future users reading this to help themselves... also feel free contact me on fb or send me code, i'm kind of blind here.

christophepersoz, 2016-10-23:
Exact for your first answer, and I was thinking the same about the thread.

neu-rah, 2016-10-23:
testing with eeprom

```c++
#include <Arduino.h>

/********************
#include <menu.h>//menu macros and objects
#include <menuPrint.h>
#include <menuFields.h>
#include <EEPROM.h>
#include <Streaming.h>

menuPrint menu_out(Serial);//describe output device

/* _____ MENUS DEFINITION _____ */
enum nSTEPS {_200,_400,_800};

struct myData {
  bool autoLoad=false;
  int _mMAXSPD=1;
  int _mGEARBOX=1;
  nSTEPS _mSTEPS=_200;
  int _hCURRENTMODE=0;
} data;

bool updMCONTANTS() {return false;}
bool updMKVALS() {return false;}

bool saveData() {
  EEPROM.put(0,data);
  Serial<<endl<<"data saved to EEPROM"<<endl;
  return false;
}

bool loadData() {
  EEPROM.get(0,data);
  menu_out.drawn=NULL;
  Serial<<endl<<"data loaded from EEPROM"<<endl;
  return false;
}

TOGGLE( data.autoLoad, useAutoLoad, "Auto load from EEPROM:",
  VALUE( "No", false),
  VALUE( "Yes", true)
);

TOGGLE( data._mSTEPS, nbMSteps, "STEPPER RESOLUTION:",
       VALUE( "200 steps", nSTEPS::_200, updMCONTANTS ),
       VALUE( "400 steps", nSTEPS::_400, updMCONTANTS ),
       VALUE( "800 steps", nSTEPS::_800, updMCONTANTS )
       );

TOGGLE( data._hCURRENTMODE, holdCurrents, "HOLD CURRENTS STRENGTH:",
      VALUE( "LIGHT", 0, updMKVALS ),
      VALUE( "STANDARD", 1, updMKVALS ),
      VALUE( "STRONG", 2, updMKVALS )
      );

// MAIN Menu
MENU( mainMenu, "MAINMENU"
  ,SUBMENU(useAutoLoad)
  ,SUBMENU( nbMSteps )
  ,SUBMENU( holdCurrents )
  ,FIELD( data._mMAXSPD, "MAX SPEED:", " stp/s", 1, 8000, 100, 5, updMCONTANTS )
  ,FIELD( data._mGEARBOX, "GEARBOX RATIO 1 :", "", 1, 255, 5, 1, updMCONTANTS )
  ,OP("Save to EEPROM",saveData)
  ,OP("Load from EEPROM",loadData)
);

void setup()   {
  Serial.begin(115200);
  menu::wrapMenus=true;
  bool autoLoad=EEPROM.get(0,data.autoLoad);
  if (autoLoad) loadData();
}

void loop()
{
  mainMenu.poll(menu_out,Serial);
}
```

christophepersoz, 2016-10-23:
Rui, I'm starting a new issue, to avoid losing people in a thread which is not talking about coloring now ;)

christophepersoz, 2016-11-08:
Working perfectly ;) Thanks Rui !

closed 2016-11-08 by christophepersoz

neu-rah, 2016-11-08:
v3 is compiling on teensy, uno, due and esp