neu-rah/ArduinoMenu #57: Wrong displayed values in menu after EEPROM load inside setup() - Teensy

christophepersoz, 2016-10-23:
This issue has been start in an another thread but it can be confuse for other people to read to issues in one thread.

After the coloring version you made Rui and which is great I discover a new bug on my Teensy board. I started to implement in my program a EEPROM configuration load which happens during the setup(). When I'm entering in the menu, I can modify some of those values and on Exit menu, the values are stored in EEPROM to be load on the next restart.

The issue concerns the values displayed in the menu. The Menu take into account the values of the vars as they are declared in the init of the ino project (boolean x = true; uint8_t y = 2; etc.) but the menu do not update them if those vars has been changed during the setup() when a EEPROM load values occurs.

neu-rah, 2016-10-24:
ok, got it
select field was missing sync
fixed https://github.com/neu-rah/ArduinoMenu/commit/df00a79f123c95ead1c9ec059b93c53bf510e9b2

still chasing lack of update when editing select field, dont know if it affects U8Glib, but at least it affects serial.... #

neu-rah, 2016-10-24:
however this sync is causing problems with serial (not refreshing)
u8glib might not be affected because it always redraws
still checking

christophepersoz, 2016-10-28:
Hi Rui,

I tried the new the sync patch you made on my teensy board and now, all the values except the FIELDS ones are no more displayed. I have to check what you changed to understand what's going wrong ;)

neu-rah, 2016-10-29:
ok, checking it also with the master branch...

neu-rah, 2016-10-29:
I'm using serial as output (no colors), toggle and select showing ok
my cut down version to fit nano is not using any field :( 
if its a u8 color problem it wont be on the master branch.

christophepersoz, 2016-10-29:
I'm testing the mods on valuecolor version, not on master branch. I should have a test with master branch and see how it works.

christophepersoz, 2016-11-01:
Hi Rui,

Sorry for the delay, to many work ;)
I had a look on the issue and replaced the value color branch by the master branch to see how it works. It is definitely a problem related to the value color branch. With the master branch version, everything is displayed correctly and the values match the ones recorded on EEPROM, even on the first load.
I'm going to try to create a new valuecolor version from master and see if I can achieve that alone and if it works... ;) Keep you in touch soon.

neu-rah, 2016-11-01:
version 3 already uses color... but of course API still might change (dev version), by using a simplistic API (just calling poll) the worst that can happen is data structure require more parameters on parameter change (it can be a problem with large menu structure)... and of course the bug pains of development.

and thanks for the patience of testing this things :D

christophepersoz, 2016-11-01:
;) I'm actually facing to an unknown bug on my dev, don't know what's going wrong actually... oups, I surely made something wrong somewhere...

Do you think I can try to move to v3 at this time and have a working menu with ?

neu-rah, 2016-11-01:
maybe try to compile some examples, 
Blink and serialio are pretty simple
but if U8Glib example compiles well its a good start
at this moment i have no idea about teensy, arduino due and nano seem to work ok (ARM and AVR)

christophepersoz, 2016-11-01:
Ok, I'm going to have a look on it and see how it compile. Keep you in touch. I found my bug so I can move forward on valuecolor ;)

christophepersoz, 2016-11-01:
Hmm, both of the dev version serialio and blink does not compile for Teensy board due to an error here (which is not mentioned as an error...) :

```
In file included from /Users/christophepersoz/Developer/_Arduino/libraries/ArduinoMenu-dev/examples/Serial/serialio/serialio.ino:1:0:
/Users/christophepersoz/Developer/_Arduino/libraries/ArduinoMenu-dev/src/menu.h: In constructor 'Menu::menuOut::menuOut(Menu::idx_t*, Menu::panelsList&, bool, bool)':
```

I'm going to try to have a valuecolor version working ;)

christophepersoz, 2016-11-01:
Hi Rui,

After some search on PJRC forum on integer I saw that I could same some of bites memory space on Teensy by redefining some of the var you are using on your lib. Maybe that can be the case on Arduino too (I don’t know, I’m really focus on Teensy boards now). Is that the case ?

Here the post link : #6 <https://forum.pjrc.com/threads/25799-integer-type-definitions-on-Teensy-3-1?p=47731&viewfull=1#post47731>

It seems that int is considered as int16_t on AVR where as it is considered as a int32_t on ARM, which means 4 bytes on ARM instead of 2. I still have plenty of RAM available, but that can be useful for all the new ARM boards (Arduino and Teensy) that are on the market now.
That fact is I don’t know if I can make this changes on a new branch and push them on your fork.
What do you think ? - If I’m right… :)

Christophe
Directeur de la photographie
Cinematographer
 
phone +33 625 71 00 61 <tel:+33-625-71-00-61>
blog.christophepersoz.com <http://blog.christophepersoz.com/>
 <skype:christophepersoz?userinfo>  <http://www.twitter.com/@cpersoz>  <http://vimeo.com/christophepersoz/showreel-dop-christophepersoz>  <https://www.linkedin.com/in/christophepersoz>

neu-rah, 2016-11-01:
i've been +95% focus on v3, where i use definitions of idx_t as int8_t (signed char, 1 byte), because AVR's are 8bit uC and internal register ar 8 bits (1 byte), avoiding unecessary convertions and 2 byte operations on AVR.

It was done indirect because we can change it to be something else on other platforms.

neu-rah, 2016-11-01:
the teensy error is not specific... :\

christophepersoz, 2016-11-01:
The current valuecolor branch cause a crash of the Teensy. I think I'm gonna try to go back the the master and apply the changes you made for the value color step by step and see how it works. Keep you in touch.

neu-rah, 2016-11-01:
github probably can compare both version and help hi-liting the diff's

christophepersoz, 2016-11-02:
Got it works! :)
I will push that tomorrow, I need to sleep now

christophepersoz, 2016-11-02:
Hi Rui, I pulled my changes from the master branch, which also includes all your last modification (safer). I don't know really well how github manage the pull request and if it pull that for the master or for an another branch. Let me know if I did it wrong ;) Thanks a lot !

christophepersoz, 2016-11-02:
Just a question, is that possible to simply determine in menufield.h if the current menu items that will be print is an OP, EXIT or FIELD ?

neu-rah, 2016-11-02:
Hi!

classes menuField, menuVariant and menuSelect have their own version of (virtual) menuPrint.
menuVariant is the base for toggle, select and choose so the derived classes use the base version except  menuSelect that has a specific version.

```c++
class menuSelect: public menuVariant<T> {
  \\...
  virtual void printTo(menuOut& p) {//<--- this printTo version is only used for nenuSelect
    menuVariant<T>::sync();
    print_P(p,menu::text);
    p.print(menu::activeNode==this?':':' ');
    ((prompt*)pgmPtrNear(menu::data[menu::sel]))->printTo(p);//calling 
  }
```

if you need specific code for any of the derived you can implement a specific virtual function on the derived class.

virtual functions of the base class are used when no class specific code is provided and derived versions can call base class version (base::funcName(...))

maybe this is not your question... do you want to know the type from the inside a printMenu or printPrompt function?

christophepersoz, 2016-11-02:
Hi Rui !

In fact, I saw that if a change the printTo() related to OP, it also change the printTo of the canExit displayed label (in a submenu). I would like to add a special char in front of OP field in sub-menu but I'm not sure that is possible, because it will be added on my "Exit Menu" which is also an OP field.

Am I clear ? I not so sure ;)

neu-rah, 2016-11-02:
yup, because Exit option are not a separate class from prompt options, so they share same code.

christophepersoz, 2016-11-02:
Hmm, so I'm gonna put this idea aside, it's nice as it is with valuecolor ;)
Thanks for your support Rui!

closed 2016-11-02 by christophepersoz

neu-rah, 2016-11-02:
I've added your changes to master branch... hope i'm not messing things around because i'm not a github expert and i was not sure on some steps, I will deal with messes as we find them.
if master is ok then we can delete valuecolor branch

christophepersoz, 2016-11-02:
Hmm, I can't tell you honestly, does the master branch compile with the modifications ?

neu-rah, 2016-11-02:
at least u8 example does dor teensy and nano and also some others that I've tryed

neu-rah, 2016-11-02:
i thought that the last would include all changes, then i had to apply all the others, resulting in applying the changes in reverse order... not quite sure it was ok... will see