neu-rah/ArduinoMenu #58: Define for 2-keys keyboard for loop FIELD value

dontsovcmc, 2016-10-27:
I you have only 2 keys in your device you can't change values in FIELDs cause you can't go down after maximum value. So, we can change clamp() function with ONLY_UP_KEY definition.

```
void clamp() {
      if (value<low) value=low;
#ifdef ONLY_UP_KEY
      else if (value>high) value=low;
#else
      else if (value>high) value=high;
#endif
}
```

neu-rah, 2016-10-27:
true indeed

can you open a pull request with this changes and also with the debounce safe code?
This way you keep the authoring, otherwise it would be as if it was all my code.

Note for myself: use wrapStyle on version 3 to control this.

dontsovcmc, 2016-10-27:
Wait a minute please, I try to understand how to do it correctly. 
https://guides.github.com/activities/contributing-to-open-source/

dontsovcmc, 2016-10-27:
done )

neu-rah, 2016-10-27:
https://github.com/neu-rah/ArduinoMenu/pull/59

closed 2016-10-27 by neu-rah

dontsovcmc, 2016-10-27:
It's better to do loop in menu items also... I try to find place in your code and make pull request.

neu-rah, 2016-10-27:
worry not , version 3 already has it, its on dev branch

neu-rah, 2016-10-27:
syntax is a bit different, but you might guess from the examples (still have to write the docs :\ ) and still missing things like UTFT/UTouch support.
on v3 wrap is done at item level instead of global