neu-rah/ArduinoMenu #61: Crash if wrap in submenu

dontsovcmc, 2016-10-27:
1. bool menu::wrapMenus=true;
2. create menus:

```
promptFeedback quit() {
  return true;
}

TOGGLE(settings.alarm_enable[0],timer1_on_menu,"Work: ",
  VALUE("YES",1),
  VALUE("NO",0)
);

TOGGLE(settings.alarm_type[0],timer1_type_menu,"Type: ",
  VALUE("Poliv",POLIV_TYPE),
  VALUE("Relay 1",RELAY1_TYPE),
  VALUE("Relay 2",RELAY2_TYPE)
);

MENU(timer1_menu,"Timer 1"
  , SUBMENU(timer1_on_menu)
  , SUBMENU(timer1_type_menu)
  , FIELD(settings.alarm_duration[0],"Dlit","sec", 1, 60, 5, 1)
  , FIELD(settings.alarm_hour[0],"Start","hour", 0, 23, 1, 1)
  , FIELD(settings.alarm_min[0],"Start","min", 0, 59, 10, 1)
  , OP("Exit", quit)
);

TOGGLE(settings.alarm_enable[1],timer2_on_menu,"Work: ",
  VALUE("YES",1),
  VALUE("NO",0)
);

MENU(mainMenu,"Main"
  , SUBMENU(timer1_menu)
  , OP("Reset", reset_settings)
  , OP("Exit", close_settings_menu)
);
```

maybe I can't use quit() function both in Main meny and SubMeny?

dontsovcmc, 2016-10-27:
I can't find using of exitOption in examples.. I think my crash cause my incorrect code.

neu-rah, 2016-10-27:
checking...

neu-rah, 2016-10-27:
checked your code ok (cutting some missing parts) on a serial IO plus some extras that came from the initial serial example that I've used.

what kind of display are your using?

here is my test code:

```c++
#include <Arduino.h>
#include <menu.h>//menu macros and objects
#include <menuFields.h>
#include <menuPrint.h>
//#include <streamFlow.h>

#define LEDPIN 13

//aux vars
int ledCtrl=0;
int percent;//just testing changing this var
double fps=0;
int counter=0;

///////////////////////////////////////////////////////////////////////////
//functions to wire as menu actions
bool ledOn() {
  Serial.println("set led on!");
  digitalWrite(LEDPIN,ledCtrl=1);
  return false;
}

bool ledOff() {
  Serial.println("set led off!");
  digitalWrite(LEDPIN,ledCtrl=0);
  return false;
}

/////////////////////////////////////////////////////////////////////////
// MENU DEFINITION
// here we define the menu structure and wire actions functions to it
// empty options are just for scroll testing

bool reset_settings()
{
    //init_settings();
    //write_settings();
    return false;
}

bool close_settings_menu()
{
    //write_settings();
    //screen_info("Settings saved", 500);
    //load_settings();

    //start_screen();
    return true;
}

promptFeedback quit() {
  return true;
}

int alarmEn=0;

TOGGLE(alarmEn,timer1_on_menu,"Work: ",
  VALUE("YES",1),
  VALUE("NO",0)
);

enum alarmTypes {POLIV_TYPE,RELAY1_TYPE,RELAY2_TYPE};
alarmTypes alarmType=POLIV_TYPE;

TOGGLE(alarmType,timer1_type_menu,"Type: ",
  VALUE("Poliv",POLIV_TYPE),
  VALUE("Relay 1",RELAY1_TYPE),
  VALUE("Relay 2",RELAY2_TYPE)
);

int dur=0;
int hour=0;
int minuts=0;

MENU(timer1_menu,"Timer 1"
  , SUBMENU(timer1_on_menu)
  , SUBMENU(timer1_type_menu)
  , FIELD(dur,"Dlit","sec", 1, 60, 5, 1)
  , FIELD(hour,"Start","hour", 0, 23, 1, 1)
  , FIELD(minuts,"Start","min", 0, 59, 10, 1)
  , OP("Exit", quit)
);

bool alarmEnabled=false;

TOGGLE(alarmEnabled,timer2_on_menu,"Work: ",
  VALUE("YES",true),
  VALUE("NO",false)
);

MENU(mainMenu,"Main"
  , SUBMENU(timer1_menu)
  , OP("Reset", reset_settings)
  , OP("Exit", close_settings_menu)
);

menuPrint menu_out(Serial);//describe output device

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("menu system test");
  pinMode(LEDPIN,OUTPUT);
  //mainMenu.data[0]->enabled=false;
  menu::wrapMenus=true;
}

void loop() {
  mainMenu.poll(menu_out,Serial);
  delay(300);
}
```

dontsovcmc, 2016-10-28:
You example works good, exept I see 2 "exit" in SubMenu. Looping works.
But, if I do this with LCD 16x2 Funduino LCM1602 i2c, program crashed.
See video: https://youtu.be/6cLgbG4PyoE

Code:
```
#include <Arduino.h>
#include <menu.h>//menu macros and objects
#include <menuFields.h>
#include <menuPrint.h>
#include <menuLCDs.h>
#include <keyStream.h>
#include <chainStream.h>

#include <LiquidCrystal_I2C.h>
extern LiquidCrystal_I2C lcd;

#define LEDPIN 13

#define BUTTON_1 4   
#define BUTTON_2 5   

//aux vars
int ledCtrl=0;
int percent;//just testing changing this var
double fps=0;
int counter=0;

///////////////////////////////////////////////////////////////////////////
//functions to wire as menu actions
bool ledOn() {
  Serial.println("set led on!");
  digitalWrite(LEDPIN,ledCtrl=1);
  return false;
}

bool ledOff() {
  Serial.println("set led off!");
  digitalWrite(LEDPIN,ledCtrl=0);
  return false;
}

/////////////////////////////////////////////////////////////////////////
// MENU DEFINITION
// here we define the menu structure and wire actions functions to it
// empty options are just for scroll testing

bool reset_settings()
{
    return false;
}

bool close_settings_menu()
{
    return true;
}

promptFeedback quit() {
  return true;
}

int alarmEn=0;

TOGGLE(alarmEn,timer1_on_menu,"Work: ",
  VALUE("YES",1),
  VALUE("NO",0)
);

int dur=0;

MENU(timer1_menu,"Timer 1"
  , SUBMENU(timer1_on_menu)
  , FIELD(dur,"Dlit","sec", 1, 60, 5, 1)
  , OP("Exit", quit)
);

MENU(mainMenu,"Main"
  , SUBMENU(timer1_menu)
  , OP("Reset", reset_settings)
  , OP("Exit", close_settings_menu)
);

menuLCD menu_lcd(lcd,16,2);

keyMap encBtn_map[]={{BUTTON_1,menu::upCode}, {BUTTON_2,menu::enterCode}}; //negative pin numbers means we have a pull-up, this is on when low
keyLook<2> encButton(encBtn_map);
Stream* in2[]={&encButton,&Serial};
chainStream<2> allIn(in2);

void setup() {
	
  Serial.begin(9600);
  while(!Serial);
  
  lcd.begin(16,2); 
  lcd.clear();	
  
  lcd.setCursor(0, 0); 
  lcd.print("menu system test");
  Serial.println("menu system test");
  delay(1000);
  pinMode(LEDPIN,OUTPUT);
  pinMode(BUTTON_1,INPUT);
  pinMode(BUTTON_2,INPUT); 
  //mainMenu.data[0]->enabled=false;
  menu::wrapMenus=true;
}

void loop() {
  //mainMenu.poll(menu_out,allIn);
  mainMenu.poll(menu_lcd,allIn);
  delay(300);
}
```

neu-rah, 2016-10-28:
yes, noticed the double exit, its because you have an Exit option on "Timer 1" menu

Exit options are automatic on 2.4 (all submenus), you can insert a manual one only on the main menu.

maybe that's the cause of reset, please let me know.

thanks

dontsovcmc, 2016-10-28:
I remove my 'exit'. 
Submenu:
Work: NO
Dlit 0sec
Exit

When I press Up after 'Dlit' to select 'Exit' program crashes.

neu-rah, 2016-10-28:
ok, an issue with the wrap for sure, i will wire an i2c here and check it out

thanks for testing it. :)

neu-rah, 2016-10-28:
checked your code (using serial input) and an I2C Lcd, with an adapter like this one https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/schematics#!i2c-connection

works nice even with the 2 exit options

I'm using this library https://bitbucket.org/fmalpartida/new-liquidcrystal/downloads (1.3.4)

changed pins to match my breadboard and pinMode to INPUT_PULLUP because i had no external pullups on my setup.

heres the code:

```c++
#include <Wire.h>
#include <LiquidCrystal_I2C.h>//F. Malpartida LCD's driver
#include <menu.h>//menu macros and objects
#include <menuLCDs.h>
#include <menuFields.h>
#include <keyStream.h>
#include <chainStream.h>

#define LEDPIN 13

#define BUTTON_1 3
#define BUTTON_2 4

//LiquidCrystal_I2C lcd(0x27);// Set the LCD I2C address
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address and pinout

//aux vars
int ledCtrl=0;
int percent;//just testing changing this var
double fps=0;
int counter=0;

///////////////////////////////////////////////////////////////////////////
//functions to wire as menu actions
bool ledOn() {
  Serial.println("set led on!");
  digitalWrite(LEDPIN,ledCtrl=1);
  return false;
}

bool ledOff() {
  Serial.println("set led off!");
  digitalWrite(LEDPIN,ledCtrl=0);
  return false;
}

/////////////////////////////////////////////////////////////////////////
// MENU DEFINITION
// here we define the menu structure and wire actions functions to it
// empty options are just for scroll testing

bool reset_settings()
{
    return false;
}

bool close_settings_menu()
{
    return true;
}

promptFeedback quit() {
  return true;
}

int alarmEn=0;

TOGGLE(alarmEn,timer1_on_menu,"Work: ",
  VALUE("YES",1),
  VALUE("NO",0)
);

int dur=0;

MENU(timer1_menu,"Timer 1"
  , SUBMENU(timer1_on_menu)
  , FIELD(dur,"Dlit","sec", 1, 60, 5, 1)
  , OP("Exit", quit)
);

MENU(mainMenu,"Main"
  , SUBMENU(timer1_menu)
  , OP("Reset", reset_settings)
  , OP("Exit", close_settings_menu)
);

menuLCD menu_lcd(lcd,16,2);

keyMap encBtn_map[]={{BUTTON_1,menu::upCode}, {BUTTON_2,menu::enterCode}}; //negative pin numbers means we have a pull-up, this is on when low
keyLook<2> encButton(encBtn_map);
Stream* in2[]={&encButton,&Serial};
chainStream<2> allIn(in2);

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  Wire.begin();
  lcd.begin(16,2);
  lcd.home();
  lcd.print("Menu 2.x test");
  menu::wrapMenus=true;
  delay(1000);
  pinMode(LEDPIN,OUTPUT);
  pinMode(BUTTON_1,INPUT_PULLUP);
  pinMode(BUTTON_2,INPUT_PULLUP); 
}

void loop()
{
  //serial can use numbers or +/- and Enter=* Esc=/ to navigate (nav keys can be redefined)
  mainMenu.poll(menu_lcd,allIn);
}

```

please let me know the results and please check if behavior is the same when using the serial as input

neu-rah, 2016-10-28:
P.S. also check the free memory after compiling your sketch, you can have stack problems and resets if free memory gets too low

dontsovcmc, 2016-10-28:
'check the free memory after compiling your sketch, you can have stack problems and resets if free memory gets too low' How can I do this?

neu-rah, 2016-10-28:
when just compiling (first button) the IDE terminates with a report of flash and ram used.

dontsovcmc, 2016-10-28:
```
avrdude: Device signature = 0x1e950f
avrdude: reading input file ".pioenvs\uno\firmware.hex"
avrdude: writing flash (11510 bytes):

Writing | ################################################## | 100% 1.85s

avrdude: 11510 bytes of flash written
avrdude: verifying flash memory against .pioenvs\uno\firmware.hex:
avrdude: load data flash data from input file .pioenvs\uno\firmware.hex:
avrdude: input file .pioenvs\uno\firmware.hex contains 11510 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.46s

avrdude: verifying ...
avrdude: 11510 bytes of flash verified

avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.
```

neu-rah, 2016-10-28:
that was an upload (the first button is verify)

dontsovcmc, 2016-10-28:
```
Archiving .pioenvs\uno\lib\libNewliquidCrystal.a
Archiving .pioenvs\uno\lib\libArduinoMenu.a
Linking .pioenvs\uno\firmware.elf
Checking program size .pioenvs\uno\firmware.elf
text       data     bss     dec     hex filename
11252       258     594   12104    2f48 .pioenvs\uno\firmware.elf
Building .pioenvs\uno\firmware.hex
```

neu-rah, 2016-10-28:
what is your ide version?

dontsovcmc, 2016-10-28:
platformio cli 3.1.0

neu-rah, 2016-10-28:
ahh, nice choice :D
i was talking about arduino IDE

dontsovcmc, 2016-10-28:
Oh, sorry ((. Should I use Arduino IDE?

neu-rah, 2016-10-28:
just for the memory report (it gives you percentage),
but if you are using linux you can also use:

**"avr-size" --mcu=atmega328p -C -d .pioenvs\uno\firmware.elf**

**sudo apt install binutils-avr** if avr-size id not installed (ubuntu/debian)

dontsovcmc, 2016-10-28:
I use Windows (. So output from Arduino IDE (russian):
"Скетч использует 10 362 байт (32%) памяти устройства. Всего доступно 32 256 байт.
Глобальные переменные используют 776 байт (37%) динамической памяти, оставляя 1 272 байт для локальных переменных. Максимум: 2 048 байт."

Crash :(

neu-rah, 2016-10-28:
not a memory problem 37%, unless you abuse the stack, because vars and stack share the same memory and they can overlap (with catastrophic results).
also can have memory problems if you use too much **new** or **malloc**, anyway, doing a small version of your project should indicate if its a memory problem or not.

we can do some tricks if you use many duplicate menus (for each timer)  by reusing them. if it is the case of memory shortage, but at the moment it does not look so.

dontsovcmc, 2016-10-28:
I can't understand what the matter :(. 

What trick you can offer? :) I have 5 timers and can't imagine 5 same SubMenus...

neu-rah, 2016-10-28:
one possible way to reuse a menu structure, there are others...

https://github.com/neu-rah/ArduinoMenu/blob/master/examples/reuse/reuse.ino

can be combined with menu suspension from other examples

neu-rah, 2016-11-02:
did help? was this solved?

dontsovcmc, 2016-11-02:
reuse.ino is fantastic. works good.
But Submenu crash is present.... I can't loop submenu (main menu is ok) in LCDmenu - crash

neu-rah, 2016-11-02:
:D i will look at that

christophepersoz, 2016-11-02:
Your reuse.ino is really smart Rui, good job. I'm gonna use it on v3 of my dev for sure, I have to param up to 8 motors which have all the same parameters to set. Thanks a lot ! ;)

neu-rah, 2016-11-03:
there are still other ways of reusing even on version 2.x, let me know if you need variantions of it

Fenixin, 2016-12-14:
Hello!

I'm having exactly the same problem with ArduinoMenu. I can reproduce the crash with this small piece of code:

```

// Includes of ArduinoMenu
#include <menu.h>
#include <chainStream.h>
#include <ClickEncoderStream.h>
#include <keyStream.h>
#include <pcint.h>
#include <menuFields.h>
#include <menuLCD.h>
#include <quadEncoder.h>

#include <LiquidCrystal.h>

// LCD screen
// for now digital pins, in the future i2c
LiquidCrystal lcd1(8, 9, 4, 5, 6, 7); // Config for lcd shield
// Pass it to the menuLCD object
menuLCD lcd(lcd1, 16, 2);

// Rotary Encoder
#define encBtn A3
quadEncoder quadEncoder(A1,A2);
quadEncoderStream enc(quadEncoder,1);// the last number es sensitivity

// Keyboard with one button (encoder button)
keyMap encBtn_map[]={{encBtn,menu::enterCode}};
keyLook<1> encButton(encBtn_map);

// Join all the inputs in a single one
Stream* in2[]={&enc, &encButton};
chainStream<2> allIn(in2);

/////////////////
//
//  Menu definition
//
bool nothing(){
  return false;
}
MENU(subMenu,"Sub-Menu",
  OP("Op1",nothing),
  OP("Op2",nothing)
);

MENU(mainMenu,"Sistema",
  OP("A",nothing),
  SUBMENU(subMenu)
);

void setup() {
  // LCD
  lcd1.begin(16,2);
  // Encoder
  quadEncoder.begin();
  pinMode(encBtn, INPUT);
}

void loop() {
  mainMenu.poll(lcd,allIn);
}
```

As soon as the `Exit` option appears the arduino board crashes. Is there anything I can do to solve the problem? Can I provide any data to help solve this?

neu-rah, 2016-12-15:
tested this with your code (just replaced lcd with pure Serial IO) and code seems to work OK both on Arduino IDE and PlatformIO on my arduino nano. It should not be related to LCD code.. but, you you're kind enough to try the code with pure serial and feedback the results i would appreciate.

I've even downloaded the git version just to make sure that we are running the same code.

```c++
#include <Arduino.h>
#include <menu.h>
#include <menuFields.h>
#include <menuPrint.h>

bool nothing(){return false;}

MENU(subMenu,"Sub-Menu",
  OP("Op1",nothing),
  OP("Op2",nothing)
);

MENU(mainMenu,"Sistema",
  OP("A",nothing),
  SUBMENU(subMenu)
);

menuPrint menu_out(Serial);//describe output device

void setup() {
  Serial.begin(115200);
  while(!Serial);
  menu::wrapMenus=true;
}

void loop() {
  mainMenu.poll(menu_out,Serial);
}
```

Fenixin, 2016-12-15:
Hello

Thanks for your answer. You are right, my code has no `menu::wrapMenus=true`. After reading the issue and seeing that happened exactly the same thing (when you hover above the default `exit` option arduino crashes) I got carried away and wrote a comment. But I have should wrote a new issue. I can do that if you wish.

I have tested the code and it doesn't work as expected. You can navigate through the menu but as soon as I enter the sub-menu the exit option has the same number as the `Op 2`, so it's impossible to choose it:

```

 1>A
 2 Sub-Menu

 1 A
 2>Sub-Menu

 1>Op1
 2 Op2
 2 Exit

 1 Op1
 2>Op2
 2 Exit

 1>Op1
 2 Op2
 2 Exit

 1 Op1
 2>Op2
 2 Exit
```
Also, I'm using the last release 2.4. Should I use the last code in master?

neu-rah, 2016-12-15:
yes, please use the latest code in master branch

Fenixin, 2016-12-15:
Using the last code in master branch fixes everything, thanks a lot! The behavior using menu in the LCD and in the terminal is just perfect.

I thought the last release was going to be more stable than the master branch, sorry for the inconvenience.

neu-rah, 2016-12-15:
no problem, thanks for testing it :D

closed 2017-01-06 by neu-rah