neu-rah/ArduinoMenu #67: Can't compile Dev

datacute, 2016-11-06:
I've been looking for a menu system to use on a hobby project of mine using a couple of Wemos D1 mini ESP8266 based controllers, one of which is a joystick I've created.

This project looked most suited to what I wanted to do, simply needing me to simulate a Serial input from my joystick, but I've had trouble compiling it.

I'm very interested to see you've recently added an esp8266 example to the dev branch, but I still can't get it to compile.

The error I'm getting is:
```
xtensa-lx106-elf-gcc: error: libraries\ArduinoMenu\ArduinoMenu.a: No such file or directory
```

Any hints?

Windows 10 pro, Arduino IDE v1.6.12, esp8266/Arduino v2.3.0

neu-rah, 2016-11-06:
yes esp8266 is very fresh, i've got it working on esp-01 serving html menu over wifi, but my first server choice WiFiServer/WiFiClient was not the best, I'm now looking at https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer but with some chalanges, its not stream friendly..

about the error, its complaining about missing ArduinoMenu.a, the cause of it missing must be earlier and might contain the key, some compiler error.

paste the log pls

also dev branch is changing, help welcome

datacute, 2016-11-06:
```

In file included from C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\src/menuBase.h:32:0,

                 from C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:22,

                 from C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino:6:

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\src/macros.h:31:50: warning: narrowing conversion of '800' from 'int' to 'Menu::idx_t {aka signed char}' inside { } [-Wnarrowing]

   const panel _panels_##id[] MEMMODE={__VA_ARGS__};\

                                                  ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino:97:1: note: in expansion of macro 'PANELS'

 PANELS(webPanels,{0,0,800,600});

 ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\src/macros.h:31:50: warning: overflow in implicit constant conversion [-Woverflow]

   const panel _panels_##id[] MEMMODE={__VA_ARGS__};\

                                                  ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino:97:1: note: in expansion of macro 'PANELS'

 PANELS(webPanels,{0,0,800,600});

 ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\src/macros.h:31:50: warning: narrowing conversion of '600' from 'int' to 'Menu::idx_t {aka signed char}' inside { } [-Wnarrowing]

   const panel _panels_##id[] MEMMODE={__VA_ARGS__};\

                                                  ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino:97:1: note: in expansion of macro 'PANELS'

 PANELS(webPanels,{0,0,800,600});

 ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\src/macros.h:31:50: warning: overflow in implicit constant conversion [-Woverflow]

   const panel _panels_##id[] MEMMODE={__VA_ARGS__};\

                                                  ^

C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino:97:1: note: in expansion of macro 'PANELS'

 PANELS(webPanels,{0,0,800,600});

 ^

xtensa-lx106-elf-gcc: error: libraries\ArduinoMenu\ArduinoMenu.a: No such file or directory

exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

datacute, 2016-11-06:
I had moved and renamed the esp8266.cpp file

neu-rah, 2016-11-06:
yup, the original organization was for platformio, guess that arduino menu would like it as an .ino file and at the root ofr the directory with the same name. Also remove all other files and folders.

neu-rah, 2016-11-06:
no luck, also Arduino IDE 1.6.10 + Esp 2.3.0 linux, exact same error

xtensa-lx106-elf-gcc: error: /tmp/buildab3f1aef9408892f0c6cb4fad432248a.tmp/libraries/ArduinoMenu/ArduinoMenu.a: No such file or directory...

datacute, 2016-11-06:
Here's the verbose output:
```
Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 160 MHz, 921600, 4M (1M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Redacted\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Redacted\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Redacted\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=160,UploadSpeed=921600,FlashSize=4M1M -ide-version=10612 -build-path C:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Redacted\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Redacted\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Redacted\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=160,UploadSpeed=921600,FlashSize=4M1M -ide-version=10612 -build-path C:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu\examples\esp8266\esp8266.ino
Using board 'd1_mini' from platform in folder: C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Build options changed, rebuilding all
Detecting libraries used...
"C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "C:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032\sketch\esp8266.ino.cpp" -o "nul"
"C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032\sketch\esp8266.ino.cpp" -o "nul"
"C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032/core" -c [... cut ...]
xtensa-lx106-elf-gcc: error: C:\Users\REDACT~1\AppData\Local\Temp\arduino_build_916032\libraries\ArduinoMenu\ArduinoMenu.a: No such file or directory

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Redacted\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library Streaming in folder: C:\Users\Redacted\Documents\Arduino\libraries\Streaming (legacy)
Using library ArduinoMenu at version 3.0.1 in folder: C:\Users\Redacted\Documents\Arduino\libraries\ArduinoMenu 
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

neu-rah, 2016-11-06:
got same error on linux (IDE 1.6.10 + esp 2.3.0), even after changing files

neu-rah, 2016-11-06:
ok, compiling here with IDE 1.6.10 + esp 3.2.0, changes pushed to dev branch

neu-rah, 2016-11-06:
ok, compiling with ide 1.6.10 + esp 3.2, changes pushed to dev branch

datacute, 2016-11-06:
Compiles for me now too.

closed 2016-11-06 by datacute