neu-rah/ArduinoMenu #68: esp8266 error: 'typeof' was not declared in this scope

datacute, 2016-11-07:
I'm trying dev branch, but I had this problem in master too.
When compiling for esp8266, and if I use FIELD, SELECT, TOGGLE or CHOOSE, I can't compile because I get the error that typeof was not declared in this scope, for the line:

```
   const MEMMODE Menu::menuValueShadowRaw<typeof(target)> choice##cnt##ShadowRaw={\
```

Replacing all the instances of "<typeof(target)>" in macros.h with "<int>" allows compilation to succeed.

I'm guessing that's a limitation of the toolchain used for esp8266?

neu-rah, 2016-11-07:
not yet, esp8266 is fresh new on dev branch, i have not looked into fields, only working on normal options and eventually menus... but it will be a boring limitation if so, i depend on that to support all kind of numeric fields and variants (unsigned, long, double, float) and also for enumerated (that support things like char*).
I will take a look at it, thanks for the interest.

neu-rah, 2016-11-07:
ArduinoMenu/src/macros.h:210:55: error: 'typeof' was not declared in this scope
same here, this is very odd, typeof is a standard part of c++ ...

neu-rah, 2016-11-07:
but there must be a solution, because I've started on esp8266 by compiling and run a serial menu that worked, but now it presents the same error, so the there must be something that i'm including now or some change that made the diff.

neu-rah, 2016-11-07:
using
```c++
#define typeof(x) __typeof__(x)
```
will solve the problem, I'm adding it as a conditional directive, for now just for my esp01_1m, need a better define to catch all esp's

datacute, 2016-11-08:
Excellent, thanks.

platform.txt includes:
build.extra_flags=-DESP8266

So this should work (for arduino):

#ifdef ESP8266
  #define typeof(x) __typeof__(x)
#endif

I'm now on to see how hard it is to create a u8g2Out.h

neu-rah, 2016-11-08:
yes, thanks :D

closed 2016-11-08 by neu-rah