neu-rah/ArduinoMenu #70: valores do menu da cor do highlight menuU8G.h

peekpt, 2016-12-18:
boas, estou a tentar aprender a trabalhar com o seu menu, penso que encontrei um bug:

no ficheiro menuU8G.h, linha 52, para funcionar direito alterei estas duas linhas para:
valueColor(valueColor),
valueColorHi(valueColorHi),

ou seja, tinha o valor com a mesma cor (preta) da barra preta de selecção(preta).

Também tive de alterar no exemplo a instancia do gfx: (faltam esses dois parâmetros)
master/examples/U8GLib_menu/U8GLib_menu.ino:linha21:

mudei de: menuU8G gfx(u8g,1,0,1,1,7,9);
para: menuU8G gfx(u8g,1,0,1,1,1,0,7,9);

Agora o menu já aparece correcto!

neu-rah, 2016-12-20:
Obrigado :D

o único display U8G que tenho aqui é monocromático e só o tenho usado na versão 3. A implementação da u8g (v2.x) foi feita por alguém que usa um ecran policromático... daí que me interrogo se na livraria u8g as cores possam variar consoante o dispositivo? ou eventualmente foi um erro (typo).
corriges tu com um push request ou corrijo eu?

peekpt, 2016-12-20:
De nada , já "pushei" 👍 
Que eu saiba biblioteca u8g é só para mono, eu uso nos meus projectos é muito completa.

Encontrei o teu menu quando andava a tentar funcionar com a m2tklib (o que ainda não consegui :P).
Tenho usado o arduino-menusystem que é só um controlador de menu em que podes fazer a tua própria interface, em alguns casos dá mais jeito, mas tens de fazer as views e controlar a paginação etc..
Já espreitei a v3. mas foi só mesmo espreitar, parece que já é thread safe :D 

Obrigado pelo teu projecto.

neu-rah, 2017-02-03:
Olá :D
a versão 3 já está a rolar

peekpt, 2017-02-03:
Fixe vou testar este fim de semana. Obrigado!

closed 2017-03-01 by neu-rah