neu-rah/ArduinoMenu #71: Menu shift on a long SELECT choices

christophepersoz, 2016-12-20:
Hi Rui,

I hope you are good. I far further in my dev and I found a new bug in v2.4 color. 
I have a SELECT entry in my menu which include 8 values. My base SUBMENU include 5 items + Exit which is the maximum that the display can handle on a single page.
A soon as I edit the SELECT field and browse the values, when I'm reaching the 7th value, all the page scrolls up instead of staying as it is.
Is that a problem of the U8G implementation or something else ?
If you have an idea ;)

Thanks a lot.

neu-rah, 2016-12-20:
Hey!
no, its a problem of 2.x sharing variables, design flaw on 2.x because initially seemed a good idea to save memory

christophepersoz, 2016-12-20:
;)
So, there is no workaround for that unfortunately. The only way is to move to v3, am I right ? 
Do you think it can be done as you now the current hierarchy of the menu and the content ?

neu-rah, 2017-03-01:
its just a matter of adding event handler functions (or doNothing), eventMask and style info for the struct.

also a central navigation object must be created, now i have some docs about it.

closed 2017-03-01 by neu-rah

christophepersoz, 2017-03-03:
Hi Rui,

I haven't tried the v3.0 of your Menu library yet, but I'm planning to do so soon (on a SSD1306 with U8G2), I will keep you in touch but it seems a really promising update ;) Thanks a lot !
Christophe