neu-rah/ArduinoMenu #74: Use of Flash (program) memory by ArduinoMemu and by application??

TerryKing, 2017-01-14:
Hello and THANKS for a great library.
I know you put much data into Flash (program) memory to save dynamic memory.
How can we find out how much program memory you used?
Is what you used reflected in the IDE report of program memory size?
How can an application ALSO put data in Program Memory?
Regards, Terry King
...In The Woods in Vermont, USA
terry@yourduino.com

neu-rah, 2017-01-16:
Hi :D
version 2.x puts some data into progmem area (strings and menu options lists).
it can reflect on the ide if bypassing the #if here https://github.com/neu-rah/ArduinoMenu/blob/master/src/macros.h#L3

forcing the use of ram.
... In Azores, PT

neu-rah, 2017-09-04:
Hi :)

version 3 takes a more aggressive ram conservation pumping more data to flash

flash memory use was discussed and improved here https://github.com/neu-rah/ArduinoMenu/issues/108

determining menu item class memory requirements.

1) Take note of flash memory with the option
2) Add a class of option -> memory usage increase is: option data+class code+1 menu slot
3) Add another option of the same class and datatype -> increase is: options data+1 menu slot

1 menu slot should be a pointer

option data: memory3-memory2-sizeof(void*)
code: memory2-memory1-(option data)

because the library is modular the first time you use a structure it adds the structure code to flash.
subsequent structure usage will only add data.

neu-rah, 2017-10-20:
V4 allows menu structure to be on RAM by default on non-AVR's and by conditional compile on AVR's.
example here: https://github.com/neu-rah/ArduinoMenu/blob/master/examples/dynamic/dynamic/dynamic.ino

Also an issue related with flash memory usage and techniques to conserve them.
https://github.com/neu-rah/ArduinoMenu/issues/108

yes, this menu is quite heavy for small AVR's like the Atmega328, specially if used with a graphics display.

still some recent efforts to conserve flash on AVR's resulted in 1.4k+ of flash savings
ex: https://github.com/neu-rah/ArduinoMenu/commit/8cff7548d05232a83fdb2c225a5ba1a7a158fd61

you are welcome to join us at: https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

or reopen this issue if you prefer this channel.

thank you. :D

closed 2017-10-20 by neu-rah