neu-rah/ArduinoMenu #75: New Output device ??

ok1cdj, 2017-01-30:
Hi, thank you for great library. Can you add new out device please:
 https://github.com/greiman/SSD1306Ascii

The Adafruit GFX or U8Glib are too big for 32k MCUs.

The functionality is similar to LiquidCrystal lib.

Regards, Ondra

neu-rah, 2017-01-30:
agree, both adagfx and u8 and big leaving me almost no space.
Hope that your lib supports at least one of my devices.. have to check that
thanks

neu-rah, 2017-02-02:
ok, I have no SSD306 display :\
I've just released v3.x, adding support for new display libraries is even simpler on 3.x
check this file for an example on how to add support for an extra library:
https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/u8g2Out.h

ok1cdj, 2017-02-02:
ok, thank you.. will try to implement ASCII only library as output..

neu-rah, 2017-02-02:
feel free to contact me if you need any help, I just cant test the result due to the lack of the appropriate device :D

and thanks for your interest

neu-rah, 2017-02-04:
I've pushed to dev branch a SSD1306Ascii menu driver with an example using Serial input.
It compiles here but I have no idea if it runs.

ok1cdj, 2017-02-05:
tried, but only Hello worl (without d!) blinking on the screen..

neu-rah, 2017-02-05:
confirmed, this is a bug on the u8g2 example,
please remove this line:
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/U8G2/U8G2/U8G2.ino#L205
```c++
nav.sleepTask=alert;
```
because alert is not the correct function and sleepTask will put the menu ionto sleep mode with the wrong function.

two lines above that there is already a correct function assigned to idleTask.

```c++
nav.idleTask=idle;//point a function to be used when menu is suspended
```

neu-rah, 2017-10-20:
solved?

neu-rah, 2017-10-25:
I have feedback from users that this driver is running ok, so I'm closing this.
feel free to reopen or open new issue if some problems persist.

You are welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

closed 2017-10-25 by neu-rah