neu-rah/ArduinoMenu #76: U8G2 compatibility

AlexFundorin, 2017-01-31:
Hi. Could you, please, adapt u8lib example to the latest version of that library - u8g2? 
https://github.com/olikraus/U8g2_Arduino

neu-rah, 2017-02-02:
v3.x supports u8g2

AlexFundorin, 2017-02-02:
That was quick. Thanks. 👍

neu-rah, 2017-02-02:
It was already done  :D
just awaiting some fixes on esp support
this v3 is a new API and it is the first release.. so if you get into trouble just let me know

AlexFundorin, 2017-02-02:
I'm using Teensy++ and 128x32 oled display from Adafruit. I'm planning to control the menu, using thumbstick and touch pads. So, I feel like it would take a lots of changes in example code to adapt the library to my need. But I'm willing to try it, cause the logic behind the menu seems the most suitable to me.
Will test the new version today. 
I'm also interested, is it possible to use icons with ArduinoMenu like in this video?
https://www.youtube.com/watch?v=wNLlGsv7eTU
Thanks again!

neu-rah, 2017-02-02:
very interesting that icons thing, v3.0 allows custom option drawing (included on some examples), that might be the solution for using icons. the examples just custom print text but they might do something else.
v2.5.0 has an example using an analog keyboard, it uses a very generic input driver
v3.0.1 has also the possibility to control the menu by calling

```c++
nav.doNav(«command»);
```
 a list of commands is available on the readme.md

AlexFundorin, 2017-02-02:
I'm looking forward to use that icons thing, cause u8g2 allows user to make custom icons, but at first it would make more sense to use just plain text for my project.
Where can I get that newest 3.0.1 version? Only 3.0 is available at GitHub.

I'm also interested if it's possible to create several menus? I want to add at least three: Presets, Quick setting, Full menu. Each of the menus will be selected separately from each other.

P.S. This is the online graphics builder that I'm using for drawing icons with u8g2.drawXBM function. I hope you find it interesting, Neu.
http://4umi.com/web/javascript/xbm.php

AlexFundorin, 2017-02-02:
I've only changed u8g2 constructor, so it wold match my OLED display in u8g2 example and this is what I've got after trying to compile the code.
https://dl.dropboxusercontent.com/u/8162257/ArduinoMenu.log

P.S. Both streaming and asster4a were installed from GitHub. Library version 3.0.

neu-rah, 2017-02-02:
what target device, IDE and OS?

AlexFundorin, 2017-02-03:
Teensy++ 2.0, 1.8.1, w10.

neu-rah, 2017-02-03:
confirmed, same kind of message even with 1.6 IDE, on simple serialio example using teensy++ 2.0
it compiles ok on teensy 3.2 / 3.1
it's not your code problem
I'm working on it, thanks for noticing this.

neu-rah, 2017-02-03:
I've opened a new issue for this because the title of this one is U8G2 related.
https://github.com/neu-rah/ArduinoMenu/issues/78

AlexFundorin, 2017-02-03:
Ok. Thanks. Can I delete that log link, then?

neu-rah, 2017-02-03:
yes

AlexFundorin, 2017-02-03:
I think that this issue #76 can be closed now, as v3 would allow me to use your library with both Teensy++ and u8g2 library, after #78 is fixed. Thanks.

closed 2017-02-03 by AlexFundorin