neu-rah/ArduinoMenu #78: Teensy++ 2.0 compilation errors

neu-rah, 2017-02-03:
as noticed on this issue https://github.com/neu-rah/ArduinoMenu/issues/76 there are some problems/errors when compiling for teensy++ 2.0

neu-rah, 2017-02-03:
to: https://github.com/AlexFundorin

I can see it is related to the lack of **typeof**, adding
```c++
#define typeof(x) __typeof__(x)
```
can solve the problem, now i just need a define that identifyes the teensy++ 2.0 to glue this in

neu-rah, 2017-02-03:
Ok, done it for all teensy core family (CORE_TEENSY) it works ok also on 3.1/3.2 so i wont look further for other preprocessor directive.
I've compiled an example ok (previously on error).
Please let me know if any error persists

neu-rah, 2017-02-03:
correction pushed to v3.0.3
https://github.com/neu-rah/ArduinoMenu/releases/tag/v3.0.3

closed 2017-02-03 by neu-rah

AlexFundorin, 2017-02-06:
Defining typeof in inside the sketch doesn't help with warnings. They are still present in v3.0.5 and the example sketch.