neu-rah/ArduinoMenu #79: Teensy++ U8G2 v3.0.1 Adafruit SPI 128x32 OLED

AlexFundorin, 2017-02-03:
This is what I've got after uploading example u8g2 sketch to the controller. Constructor wass replaced with this string: U8G2_SSD1306_128X32_UNIVISION_F_4W_SW_SPI u8g2(U8G2_R0, 21, 22, 18, 20, 19);
https://www.youtube.com/watch?v=5j3QaG1q2bE

neu-rah, 2017-02-03:
that message is shown when you are "out of menu" (exit from main menu) but it seem to be scrolling down... that is a bit weird.
what happens if you press enter/select or click the encoder or whatever input method you have to select?

AlexFundorin, 2017-02-03:
I did not assign any controls yet. Just uploaded the example to test if it's working, after you've fixed Teensy's issue. Could this effect happen because I'm using 32 lines screen, instead of 64? Are there any settings or configuration options, where I can set up the screen height?

neu-rah, 2017-02-03:
screen size... good question, I'm not the author of the u8g2 driver but I'm sure I can take a look on it, its a matter of setting up an assembly with a screen that is supported by u8g2 (my u8glib old screen must do, nokia5110).

you can try it using serial input.

also make sure that you do not have input pins floating because that could cause the exit from the main menu into that screen.

AlexFundorin, 2017-02-03:
There might be floating input pins, cause I'm using touch pads and one of them might be connected to the LED pin. The controller is inside the case and all wires are soldered, so I can't disconnect anything.

I've tried changing thу example, so that it would work as full buffer instead of page buffer, but I those multiple errors returned again.
Code changes (setup and loop) plug log output when tried to compile the example. Apart from setup and loop, the only change I've made is constructor line, that I've show earlier.
http://i.imgur.com/8Ietrbf.png

neu-rah, 2017-02-03:
Ok, on this line

https://github.com/neu-rah/ArduinoMenu/blob/master/examples/U8G2/U8G2/U8G2.ino#L177

we setup the output device

```c++
U8G2_OUT(u8g2,colors,fontX,fontY,offsetX,offsetY,{0,0,64/fontX,48/fontY})
```
that {0,0,64/fontX,48/fontY} is an output panel
it describes a panel size in terms of characters (not pixels)

the author used teh screen size in pixels divided by the font size in pixels making it very easy to describe another panel size in terms of pixels, just fill in your screen size instead of the 64x48 (assuming the font size is the same from the example)

neu-rah, 2017-02-03:
image shows warnings.. .can you paste the errors text here?

AlexFundorin, 2017-02-03:
`In file included from C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/menu.h:23:0,

                 from D:\mailcloud\Controller\Sketches\!2016\U8G2_ArduinoMenu\U8G2_ArduinoMenu.ino\U8G2_ArduinoMenu.ino.ino:32:

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/menuBase.h: In member function 'Menu::StringStream::operator const String&()':

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/menuBase.h:170:52: warning: returning reference to temporary [-Wreturn-local-addr]

         operator const String&() {return String(src);}

                                                    ^

In file included from C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/menuBase.h:30:0,

                 from C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/menu.h:23,

                 from D:\mailcloud\Controller\Sketches\!2016\U8G2_ArduinoMenu\U8G2_ArduinoMenu.ino\U8G2_ArduinoMenu.ino.ino:32:

D:\mailcloud\Controller\Sketches\!2016\U8G2_ArduinoMenu\U8G2_ArduinoMenu.ino\U8G2_ArduinoMenu.ino.ino: At global scope:

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:229:41: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

     *(menuValueShadow<typeof(target)>*)&choice##cnt##ShadowRaw;\

                                         ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:217:23: note: in expansion of macro '_MK_VALUE'

 #define MK_VALUE(...) _MK_VALUE(__VA_ARGS__)

                       ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:215:32: note: in expansion of macro 'MK_VALUE'

 #define DECL_VALUE(target,...) MK_VALUE(target, _##__VA_ARGS__)

                                ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/baseMacros.h:58:3: note: in expansion of macro 'DECL_VALUE'

   what(params,x)\

   ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/baseMacros.h:7:35: note: in expansion of macro 'XFOR_EACH_2'

 #define CONCATENATE2(arg1, arg2)  arg1##arg2

                                   ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/baseMacros.h:112:42: note: in expansion of macro 'XFOR_EACH_'

 #define XFOR_EACH(what, params , x, ...) XFOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, params, x, __VA_ARGS__)

                                          ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:146:3: note: in expansion of macro 'XFOR_EACH'

   XFOR_EACH(DECL_VALUE,target,__VA_ARGS__)\

   ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:143:21: note: in expansion of macro 'altVARIANT'

 #define TOGGLE(...) altVARIANT(toggle,((systemStyles)(Menu::_menuData|Menu::_isVariant)),__VA_ARGS__)

                     ^

D:\mailcloud\Controller\Sketches\!2016\U8G2_ArduinoMenu\U8G2_ArduinoMenu.ino\U8G2_ArduinoMenu.ino.ino:74:1: note: in expansion of macro 'TOGGLE'

[... 1308 lines cut ...]
 #define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)

                                ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:124:3: note: in expansion of macro 'FOR_EACH'

   FOR_EACH(DECL,__VA_ARGS__)\

   ^

C:\Users\fundorin\Documents\Arduino\libraries\ArduinoMenu-master\src/macros.h:122:42: note: in expansion of macro 'altMENU'

 #define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,__VA_ARGS__)

                                          ^

D:\mailcloud\Controller\Sketches\!2016\U8G2_ArduinoMenu\U8G2_ArduinoMenu.ino\U8G2_ArduinoMenu.ino.ino:124:1: note: in expansion of macro 'MENU'

 MENU(mainMenu,"Main menu",zZz,noEvent,noStyle

 ^

Скетч использует 27314 байт (21%) памяти устройства. Всего доступно 130048 байт.
Глобальные переменные используют 2349 байт (28%) динамической памяти, оставляя 5843 байт для локальных переменных. Максимум: 8192 байт.
`

AlexFundorin, 2017-02-03:
And here's the sketch itself - https://dl.dropboxusercontent.com/u/8162257/U8G2_ArduinoMenu_issue.ino.ino

neu-rah, 2017-02-03:
hehehe

ok, I've copied the text into my text editor and still can not find the word "error" probably because its in...ahhh.. russian? (just guessing)

AlexFundorin, 2017-02-03:
There're no errors, neither in english or russian, but those log entries should not present if the sketch is compiled properly.

neu-rah, 2017-02-03:
ok google translated that to me and it seems to compile, and yes with a LOT of warnings most of them because i'm twisting C++ to support PGM, but it should run with the warnings.

> Sketch uses 27314 bytes (21%) memory device. Total available 130048 bytes.
> Global  variables used 2349 bytes (28%) of the heap, leaving 5843 bytes for local variables. Maximum: 
> 8192 bytes.

about floating inputs... that might not be the case because the example uses serial input
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/U8G2/U8G2/U8G2.ino#L150

it means you can open your serial monitor at 115200 and use keys 
+ = move down 
- = move up
/ = esc
* = enter

on arduino IDE you must press enter after the key and also you can send several keys at once pressing enter at the end

well they do because C++ does not really support PGM (and i use C++ classes for the menu) and there are also some warnings about references being returned, its ok because we do not move the content in memory, c++ is only being zealous.

AlexFundorin, 2017-02-03:
Even though it compiles, the screen is blank with the above sketch (my attempt to use full buffer).
I might try page buffer again, but I won't be using it in my sketch and the screen is flickering/scrolling with page buffer.
So, even though you library seems to be organized neatly (I like the idea of OP/FIELD/SUBMENU/EXIT alot), I feel like I'm going to use some other library for my project. Thanks for your support.

neu-rah, 2017-02-03:
ok

neu-rah, 2017-02-03:
sorry, I was not paying enough attention and was talking about other kind of stuff, you are right about those warnings... just removed them, i had some silencing flags locally so i was thinking about something else.

AlexFundorin, 2017-02-03:
Still the same amount of warnings with default example, where only the constructor is changed
http://pastebin.ru/uqrYt67S#

neu-rah, 2017-02-03:
Ok, thanks
I understand that you do not want to bother at this level with this things, anyway i depend on some feedback because I do not have all platforms, however I can compile to any.

AlexFundorin, 2017-02-03:
I can help you with additional tests, if necessary. No biggie. 
Yet, I'm going to use Jomelo's library for my project, since it's also compatible with U8g2 and it's working with my setup.

neu-rah, 2017-02-03:
thanks a lot for your patience.. but i'm afraid that it would be asking too much for you to test this things, specially if you are not using it.
no hard feeling.
and warnings appeared to me because i've change the target and that made my IDE (atom+platformio) recompile the library due to target change... so they are not teensy specific, they are specific to GCC and are caused by me allocating structs on PROGMEM (flash) because C++ does not initialize PROGMEM correctly and then i cast the structs to C++ classes, its a dangerous thing because I had to take account of vtable pointers (and i did).
so that “dereferencing type-punned pointer will break strict-aliasing rules” warning if part of the construction and the twist that I've made to use C++ and PROGMEM.
GCC has a flag to disable the warning, and i use it on platformio as a way of saying "its ok, I know what I'm doing) but arduino ide does not allow compiler flags.
It is the price that i am paying for using PROGMEM and C++ as the horrible macro error (when they happen) are also the price of having simpler initialization.
PROGMEM is a thing that is only interesting and even essential on AVR micro-controllers because of memory constrains... teensy does not have the same constrains and therefor you probably will be confortable with a dynamic menu on RAM instead of flash.
The warning i got rid was something about returning a local address...
again, thanks for your patience.

AlexFundorin, 2017-02-03:
No probs. You can ask me to test something on my setup, if needed.

neu-rah, 2017-02-05:
flicking screen with message scrolling down is a bug on the example fixed on 3.0.5 and caused by this line:
nav.sleepTask=alert;
just remove the line, because alert is not the correct function and sleepTask will put the menu into sleep mode with the wrong function.

two lines above that there is already a correct function assigned to idleTask.

nav.idleTask=idle;//point a function to be used when menu is suspended

AlexFundorin, 2017-02-05:
Where do I find that line? Example file doesn't contain those lines. It only exist in u8glib.

I also don't get where can I find the contents of the nav.doOutput function. It's used to output menu to the display. Your example uses page buffer output, while I'm using full buffer in my project, so I need to change the example, to fit my needs. I can't do that until I'd find out what's inside nav.doOutput function. Do I need to use ClearBuffer outside of this function or is it already used in it? Do I need to use SendBuffer, also?

AlexFundorin, 2017-02-05:
This is how the example looks on my device's screen - http://i.imgur.com/bLi3YkH.png
The menu itself is much bigger, according to serial monitor - http://i.imgur.com/IjsYZK3.png

There are still lots of warnings in the output, but I'll ignore them, for now, though they can be irritating, when it's needed to look for the real errors in the log output.

neu-rah, 2017-02-05:
the line is already removed from the u8g2 example on v3.0.5

for a full buffered you can try just to call 
```c++
nav.poll();
```
inside the loop functions as done in some other full buffered examples (lcd and adafruit)

poll function will do input and output

AlexFundorin, 2017-02-05:
Thanks for your suggestions. I'll try your library again tomorrow, including custom control mappings, several separate menus. I also need to find out how to display menu at full screen, as oposed to this - http://i.imgur.com/bLi3YkH.png

neu-rah, 2017-02-05:
the author is using an offset

#define offsetX 32
#define offsetY 16

just put them to 0

AlexFundorin, 2017-02-05:
Already figured out the offset thing. Thanks. To whom are you reffering to, when talking about "author"?

I need to set physical controls to be sure, but it seems like there's a bug happening.
*All controls performed via serial monitor and PC keyboard

"1" pressed - http://i.imgur.com/tuBRby6.png
"2" pressed - http://i.imgur.com/5oeYR7w.png

neu-rah, 2017-02-05:
the second option is disabled (was disabled on setup function)

navigation keys on serial
[+] move down
[-] move up
[*] enter
[/] escape

if there is a bug on key numeric accelerators the nav keys above should work

i'm not the author of the u8g2 driver, it was done by https://github.com/datacute who contributed, as far as i can tell he has done an excellent work

AlexFundorin, 2017-02-05:
The same thing happens when navigating with +- buttons:
http://i.imgur.com/knu5KTs.png

neu-rah, 2017-02-05:
the defined display must be larger than your physical display, no?

AlexFundorin, 2017-02-05:
What do you mean? The constructor is exactly the same as I use in my main sketch.
http://i.imgur.com/Oraiox0.png
http://i.imgur.com/R18jXLv.png

AlexFundorin, 2017-02-05:
You might be right, meaning that the cursor is below the screen. I've changed 48/fontY to 32/fontY and now I can see the cursor, but the issue with line2 still remains.

neu-rah, 2017-02-05:
line 2 is disable as an example, are you referring to that? if so its disable on the setup function

AlexFundorin, 2017-02-05:
I mean the second line of the displayed menu. The cursor symbol turns from ">" into "-" and the cursor itself becomes half of the frame, instead of the filled box:
On display - https://i.imgur.com/5oeYR7w.png
In the port - https://i.imgur.com/knu5KTs.png

neu-rah, 2017-02-05:
yup changing from > to - means a disabled option the lines should be a border, that might be a miscalculation of font size, gona check that

the line is disabled on the example line
```c++
mainMenu[1].enabled=disabledStatus;
```

AlexFundorin, 2017-02-05:
Thanks. Didn't you thought about creating a proper Wiki for your library, where all features would be described in a compact and neat manner, like this one? 
https://github.com/olikraus/u8g2/wiki/u8g2reference

neu-rah, 2017-02-05:
yes i did.. and i have to, but writing this 5000+ lines of C++ took me a lot of time and there was people waiting for the changes, so the time for writing docs and make videos is now.
version 3 is a reflex of some things people were requesting/missing from version 2

AlexFundorin, 2017-02-05:
I mean, real Wiki, where users can contribute in writing it.

neu-rah, 2017-02-05:
interesting.. didn't knew people could edit, yes nice idea... i will check it

AlexFundorin, 2017-02-05:
https://help.github.com/articles/changing-access-permissions-for-wikis/

AlexFundorin, 2017-02-06:
BTW, flicking of the screen is still present in "Alert test".

neu-rah, 2017-10-20:
version 4 corrected some examples problems including flicking.
please reopen or create new if problems persist

closed 2017-10-20 by neu-rah