neu-rah/ArduinoMenu #81: Help / Documentation

sphh, 2017-02-05:
Hi,

I want to use your menu, but have trouble to get it working.

My set-up is:
- LCD screen 2x40 (using the LiquidCrystalFast library)
- Custom-made input

My code so far
```
#include <LiquidCrystalFast.h>
#include <menu.h>
#include <menuIO/liquidCrystalOut.h>
#include "myInput.h"

// LCD
LiquidCrystalFast lcd(
  PIN_LCD_RS, PIN_LCD_RW, PIN_LCD_ENABLE,           // Control pins
  PIN_LCD_D4, PIN_LCD_D5, PIN_LCD_D6, PIN_LCD_D7);  // Data pins

// Input returned by 'scan_input()': NONE, CONFIRM, ESCAPE, LEFT, RIGHT, UP, DOWN.
byte my_input;

// Menu definition (minimal example, taken from 'examples/LiquidCrystal/LiquidCrystal.ino')
#define MAX_DEPTH 2
int timeOn=10;
int timeOff=90;
MENU(mainMenu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle,
  FIELD(timeOn, "On", "ms", 0,500, 100, 10, Menu::doNothing, Menu::noEvent, Menu::noStyle),
  FIELD(timeOff, "Off", "ms", 0,500, 100, 10, Menu::doNothing, Menu::noEvent, Menu::noStyle),
  EXIT("<Back")
);
MENU_INPUTS(in, ???);                                 // <== ???
MENU_OUTPUTS(out, MAX_DEPTH
  ,LIQUIDCRYSTAL_OUT(lcd, {0, 0, 16, 20})  // <== ???
  ,NONE
);
NAVROOT(nav, mainMenu, MAX_DEPTH, in, out);

void setup() {
  lcd.begin(40, 2);
}

void loop() {
  my_input = scan_input();

  switch (my_input) {
    case CONFIRM:
          nav.doNav(Menu::enterCmd);
          break;
    case BTN_ESCAPE:
          nav.doNav(Menu::escCmd);
          break;
    case BTN_LEFT:
          nav.doNav(Menu::leftCmd);
          break;
    case BTN_RIGHT:
          nav.doNav(Menu::rightCmd);
          break;
    case BTN_UP:
          nav.doNav(Menu::upCmd);
          break;
    case BTN_DOWN:
          nav.doNav(Menu::downCmd);
          break;
    }
    nav.poll();
}
```
My questions are:

1. What do I put into the macro `MENU_INPUTS(in, ...)`, if I just want to control the menu programmatically by myself?
2. What are the meanings of the arguments `{0, 0, 16, 20}` of the `LIQUIDCRYSTAL_OUT` macro. I guess the `16` are the number of columns, but what is `20`? (A typo and it should read `2`??)

Thank you for your help!

neu-rah, 2017-02-05:
Hi!

1. yes, that 20 is a typo, should be 2

2. you can initialize your menu IO without macros
```c++
//how deep is the menu system? we need to know this because the menu is static 
//and we want to allocate all needed control variables before program startup
// no dynamic memory allocation
#define MAX_DEPTH 2

//describing a menu output device without macros
//define at least one panel for menu output
const panel panels[] MEMMODE={{0,0,16,2}};
navNode* nodes[sizeof(panels)/sizeof(panel)];//navNodes to store navigation status
panelsList pList(panels,nodes,1);//a list of panels and nodes
idx_t tops[MAX_DEPTH];//store cursor positions for each level
liquidCrystalOut outLCD(lcd,tops,pList);
//SSD1306AsciiOut outOLED(&oled,tops,pList);//oled output device menu driver
menuOut* outputs[]={&outLCD};//list of output devices
outputsList out(outputs,1);//outputs list

//NAVROOT(nav,mainMenu,MAX_DEPTH,NONE,out);
//above macro can be replaced by:
navNode navCursors[MAX_DEPTH];//objects to control each level of navigation
navRoot nav(mainMenu,navCursors,MAX_DEPTH-1,*(Stream*)NULL,out);//the root navigation object in this case with a NULL input
```

sphh, 2017-02-06:
Hi!

Thank you for your help. I would never have figured out these command. However, when I want to update the menu with `nav.poll()` the Arduino reboots (in my program below, if `scan_input()` returns a non-0 value).

I now have the following program:

```
#include <LiquidCrystalFast.h>
#include <menu.h>
#include <menuIO/liquidCrystalOut.h>
#include "myInput.h"

// LCD
LiquidCrystalFast lcd(
  PIN_LCD_RS, PIN_LCD_RW, PIN_LCD_ENABLE,           // Control pins
  PIN_LCD_D4, PIN_LCD_D5, PIN_LCD_D6, PIN_LCD_D7);  // Data pins

// Input returned by 'scan_input()': NONE, CONFIRM, ESCAPE, LEFT, RIGHT, UP, DOWN.
byte my_input;

// Needed, because we use the LiquidCrystalFast library instead of LiquidCrystal! It's a verbatim copy
// of the liquidCrystalFast class with all references to 'LiquidCrystal' replaced by 'LiquidCrystalFast'
namespace Menu {

  class liquidCrystalFastOut:public menuOut {
    public:
      LiquidCrystalFast& device;
      inline liquidCrystalFastOut(LiquidCrystalFast& o,idx_t *t,panelsList &p,menuOut::styles s=minimalRedraw)
        :menuOut(t,p,s),device(o) {}
      size_t write(uint8_t ch) override {return device.write(ch);}
      void clearLine(idx_t ln,idx_t panelNr=0,colorDefs color=bgColor,bool selected=false,status stat=enabledStatus,bool edit=false) override {
        setCursor(0,ln,panelNr);
        for(int n=0;n<maxX();n++) print(' ');
        setCursor(0,ln,panelNr);
      }
      void clear() override {
        device.clear();
        panels.reset();
      }
      void clear(idx_t panelNr) override {
        const panel p=panels[panelNr];
        //device<<endl;
        fill(p.x,p.y,p.x+p.w-1,p.y+p.h-1);
        setCursor(0,0,panelNr);
        panels.nodes[panelNr]=NULL;
      }
      void setCursor(idx_t x,idx_t y,idx_t panelNr=0) override {
        const panel p=panels[panelNr];
        device.setCursor(p.x+x,p.y+y);
      }
  };

}//namespace Menu

// Menu definition (minimal example, taken from 'examples/LiquidCrystal/LiquidCrystal.ino')
#define MAX_DEPTH 2
int timeOn=10;
int timeOff=90;
MENU(mainMenu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle,
  FIELD(timeOn, "On", "ms", 0,500, 100, 10, Menu::doNothing, Menu::noEvent, Menu::noStyle),
  FIELD(timeOff, "Off", "ms", 0,500, 100, 10, Menu::doNothing, Menu::noEvent, Menu::noStyle),
  EXIT("<Back")
);

const Menu::panel panels[] MEMMODE={{0,0,40,2}};
Menu::navNode* nodes[sizeof(panels)/sizeof(Menu::panel)];//navNodes to store navigation status
Menu::panelsList pList(panels,nodes,1);//a list of panels and nodes
Menu::idx_t tops[MAX_DEPTH];//store cursor positions for each level
Menu::liquidCrystalFastOut outLCD(lcd,tops,pList);
Menu::menuOut* outputs[]={&outLCD};//list of output devices
Menu::outputsList out(outputs,1);//outputs list

Menu::navNode nav_cursors[MAX_DEPTH];//objects to control each level of navigation
Menu::navRoot nav(main_menu, nav_cursors, MAX_DEPTH-1, *(Stream*)NULL, out);//the root navigation object in this case with a NULL input

void setup() {

  lcd.begin(LCD_COLS, LCD_ROWS);

  // Show boot message
  lcd.noCursor();
  lcd.print("booting...");
  lcd.print("done");
}

void loop() {
  my_input = scan_input();

  if (my_input) {
    switch (my_input) {
      case CONFIRM:
            nav.doNav(Menu::enterCmd);
            break;
      case BTN_ESCAPE:
            nav.doNav(Menu::escCmd);
            break;
      case BTN_LEFT:
            nav.doNav(Menu::leftCmd);
            break;
      case BTN_RIGHT:
            nav.doNav(Menu::rightCmd);
            break;
      case BTN_UP:
            nav.doNav(Menu::upCmd);
            break;
      case BTN_DOWN:
            nav.doNav(Menu::downCmd);
            break;
      }
      nav.poll();
  }
}
```
The compilation message is:
```
Sketch uses 23086 bytes (80%) of program storage space. Maximum is 28672 bytes.
Global variables use 1167 bytes (45%) of dynamic memory, leaving 1393 bytes for local variables. Maximum is 2560 bytes.
```

Do you have any guess where the crash can happen?

The `LiquidCrystalFast` library should be a drop-in replacement for the `LiquidCrystal` library.

Thanks again for your help!

neu-rah, 2017-02-06:
yes, crash is expected on poll...
poll is for full automated mode and it does IO
to have programatic mode I've set the input to NULL, so you can not has for poll neither doInput
however you can use doOutput.
U8Glib and U8G2 examples show a loop function without the poll, however you can not call the doInput method as in the example.
and sorry documentation is a pain at this moment.. guess that i will use this issues to build one.
thanks for your interest.

neu-rah, 2017-02-06:
**note to myself:** ignore null IO to avoid crashing even if IO functions are called

sphh, 2017-02-08:
Thank you for your help. Unfortunately the Arduino still crashes. I removed the `LiquidCrystalFast` library and minimized my sketch to:
```
#include <LiquidCrystal.h>

#include <menu.h>
#include <menuIO/liquidCrystalOut.h>

// == LCD ========================================================

LiquidCrystal lcd(13, 12, 11, 10, 9, 8);

// == Menu =======================================================

#define MAX_DEPTH 2
int timeOn=10;
int timeOff=90;
MENU(main_menu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle,
  FIELD(timeOn,"On","ms",0,500,100,10, Menu::doNothing, Menu::noEvent, Menu::noStyle),
  FIELD(timeOff,"Off","ms",0,500,100,10,Menu::doNothing, Menu::noEvent, Menu::noStyle),
  EXIT("<Back")
);

const Menu::panel panels[] MEMMODE={{0,0,40,2}};
Menu::navNode* nodes[sizeof(panels)/sizeof(Menu::panel)];//navNodes to store navigation status
Menu::panelsList pList(panels,nodes,1);//a list of panels and nodes
Menu::idx_t tops[MAX_DEPTH];//store cursor positions for each level
Menu::liquidCrystalOut outLCD(lcd,tops,pList);
Menu::menuOut* outputs[]={&outLCD};//list of output devices
Menu::outputsList out(outputs,1);//outputs list

Menu::navNode nav_cursors[MAX_DEPTH];//objects to control each level of navigation
Menu::navRoot nav(main_menu, nav_cursors, MAX_DEPTH-1, *(Stream*)NULL, out);//the root navigation object in this case with a NULL input

void setup() {
  // put your setup code here, to run once:

  lcd.begin(40, 2);

  pinMode(13, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:

  nav.doOutput();
  digitalWrite(13, HIGH);
}
```
I would have thought that the LED is illuminated, but that does not happen. So I believe that the Arduino crashed and reboots whenever `nav.doOutput()` is called, hence never lighting the LED.

I have to admit that I am totally lost why this can happen. Do you have any idea?

Thanks for your help!

neu-rah, 2017-02-27:
using that NULL output implies that you can not call doOutput, i'v been writing documentation (wiki) that covers the case of non-standard (or empty) input or output. https://github.com/neu-rah/ArduinoMenu/wiki/Input

to obtain a an empty input list use

```c++
chainStream<0> noInputs(NULL);
```

no input will be read (happily  ignored). Hope it helps

neu-rah, 2017-02-27:
also on your sketch you are using pin 13 for lcd and also for the led.. that can confuse the lcd driver, please correct first.

neu-rah, 2017-02-27:
ok, back

fixed that construction by initializing tops memory like this:

```c++
Menu::idx_t tops[MAX_DEPTH]={0,0};//for MAX_DEPTH=2
```

because you do not use submenus here then max. depth can be 1

i've traced down the problem to out of bound access derived from tops memory not being initialized properlly.

let me know if any problem persists... my tests are limited by not having input schema wired in

neu-rah, 2017-02-27:
of course they persist,  also the list of outputs need to be on flash memory (AVR's) by addind the **const** and **MEMMODE**.

```c++
Menu::menuOut* const outputs[] MEMMODE=...
```

neu-rah, 2017-03-01:
is this solved?

sphh, 2017-03-01:
Hi!

I just got it working! That's my code at the moment (just in case somebody is stumbling upon this thread):
```C++
#include <LiquidCrystal.h>

#include <menu.h>
#include <menuIO/liquidCrystalOut.h>

// == LCD ========================================================

LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

// == Menu =======================================================

#define MAX_DEPTH 2
int timeOn=10;
int timeOff=90;
MENU(main_menu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle,
  FIELD(timeOn,"On","ms",0,500,100,10, Menu::doNothing, Menu::noEvent, Menu::noStyle),
  FIELD(timeOff,"Off","ms",0,500,100,10,Menu::doNothing, Menu::noEvent, Menu::noStyle),
  EXIT("<Back")
);

const Menu::panel panels[] MEMMODE={{0,0,40,2}};
Menu::navNode* nodes[sizeof(panels)/sizeof(Menu::panel)];//navNodes to store navigation status
Menu::panelsList pList(panels,nodes,1);//a list of panels and nodes
Menu::idx_t tops[MAX_DEPTH] = {0,0};//store cursor positions for each level
Menu::liquidCrystalOut outLCD(lcd,tops,pList);
Menu::menuOut* const outputs[] MEMMODE={&outLCD};//list of output devices
Menu::outputsList out(outputs,1);//outputs list

Menu::chainStream<0> in(NULL);

Menu::navNode nav_cursors[MAX_DEPTH];//objects to control each level of navigation
Menu::navRoot nav(main_menu, nav_cursors, MAX_DEPTH-1, in, out);//the root navigation object in this case with a NULL input

void setup() {
  // put your setup code here, to run once:

  lcd.begin(40, 2);

  pinMode(13, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:

  my_input = scan_input();

  if (my_input) {
    switch (my_input) {
      case CONFIRM:
            nav.doNav(Menu::enterCmd);
            break;
      case BTN_ESCAPE:
            nav.doNav(Menu::escCmd);
            break;
      case BTN_LEFT:
            nav.doNav(Menu::leftCmd);
            break;
      case BTN_RIGHT:
            nav.doNav(Menu::rightCmd);
            break;
      case BTN_UP:
            nav.doNav(Menu::upCmd);
            break;
      case BTN_DOWN:
            nav.doNav(Menu::downCmd);
            break;
      }
      nav.doOutput();
  }

  digitalWrite(13, HIGH);
}
```

Thank you for your patience and help!

PS: Using the pin 13 for the LCD and LED was a copy and paste error.

closed 2017-03-01 by sphh