neu-rah/ArduinoMenu #82: External Power Usage

AlexFundorin, 2017-02-06:
I've tried connecting my controller to the powerbank and the screen is just blank. Nothing happened. Even the LED light that is set up in the example sketch wasn't lit.
The very same sketch works fine when the controller is connected to the PC.
I wonder if you could provide an example that doesn't use serial connection and would work without a PC.
I've tried removing all the data that is related to Serial from the sketch, but stuck at many lines that I don't understand. 
Like, what should I replace "Serial" with in this line of code, for example?
`
NAVROOT(nav,mainMenu,MAX_DEPTH,Serial,out);`

I'm controlling the menu with a thumbstick, so, no need of a serial connection at all.

neu-rah, 2017-02-06:
similar case
https://github.com/neu-rah/ArduinoMenu/issues/81

AlexFundorin, 2017-02-07:
I've fixed this issue. In my case, I didn't use poll, only output, from the very start.

I also have some other questions. Do I need to start a new issue or can I ask them here?

Anyway, I would like to totally disable idle/suspend mode, but can't find a way to do this.

Second thing is that I want at least two competely independent menus, so I would call and use ach one of them separately. Is there a way to call nav.output with the name of the particular menu?

The third question is about fine step tuning in FIELD. Is is possible to skip this part of defining a value?

Next question: is is possible to do so that the menu will automatically jump back to the root if no actions were made inside the menu for a particular amount of time?
In my case, I go to menu to change some option, like root>sub>sub>sub>option. Then, I quit the menu and do other things with the controller. After some time I want to change some other option from the menu, that is, for example, in the root, but the menu opens in root>sub>sub>sub>option and it's needed to get back to the root folder to change the needed option.

Hope that you understood what I'm asking about. Thanks in advance.

AlexFundorin, 2017-02-07:
I can partially answer some of those questions myself:
To remove fine step tuning, just use "0" as a step counter.
To use different menus, use NAVROOT with a different menu name as a second argument. I'm still trying to figure out how to use this in my sketch. When I'm using different NAVROOT strings inside my loop, I can call completely separate menus, but the controls aren't working. I'm using a my own function to control the menu. I feel like the first argument of NAVROOT might be responsible for this. Sadly, there's no docs/info about navroot here. I wonder, what is the forth vaiable responsible for ("Serial")?

neu-rah, 2017-02-07:
if not related to "esternal power usage" then its better to open a new issue, that helps people with similar questions.

MENU - define a menu

MENU( id, title, action, events mask, styles, Option, Option, [Option,...])

events mask are values from https://github.com/neu-rah/ArduinoMenu#events

it is a binary mask, values can be combined with | like:

enterEvent | exitEvent

meaning your action function (if defined) will be called for those events

AlexFundorin, 2017-02-07:
I would open new issue, related to the questions, from now on. Thanks.

My previous question wasn't about MENU initialisation, though, but about NAVROOT. As far as I understand, it defines, which menu should be used in nav.poll. 
I've tried using NAVROOT inside the loop, so it would be possible to define several different menus.
It worked and I had two independant menus available, but it wasn't possible to control them.

And I still don't understand of how do completely disable menu suspending. I would prefer that when esc button was pressed, while in the root menu, nothing would happen. As for now, the menu goes into suspend and I don't need that. 
Thanks.

neu-rah, 2017-02-07:
yes, just noticed
MENU_INPUTS(in,&Serial);

creates an **in** object (input description) that is passed to NAVROOT macro

aerodolphin, 2017-07-15:
I have the same problem when I use an external power source. I get a blank screen. What is the solution?

neu-rah, 2017-07-15:
external source should have no effect... i'm clueless... any1?
external source can supply the needed current (lcd+arduino)?
does the mcu boot? --> suggestion try to blink led on setup function before anything else.

neu-rah, 2017-10-20:
closing due to lack of info. please reopen or join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link to proceed with this case

closed 2017-10-20 by neu-rah