neu-rah/ArduinoMenu #84: prompt output error

BennehBoy, 2017-03-02:
Compiling the codectrl example against Uno with IDE 1.8.1 if either Opt1 or Opt2 are selected results in the out put of garbage and a MAX_DEPTH warning:

Here is the output for Opt1

`2 event on ÿ”„”¬”¬”¬”¬”¬”¬”¬”¬”¬”¬”¬”¬”¬”¬”¬”Õ”¬”p”J”¬”¬”¬”¬”¬”¬Error: maxDepth reached!

increase maxDepth on your scketch., proceed menu
`

This appears to be the 'item' output in the following line:

`Serial<<e<<" event on "<<item<<", proceed menu"<<endl;`

neu-rah, 2017-03-02:
I've checked that sketch again with ide 1.6.10 on both Uno and Nano with console baud 115200 and was not able to reproduce your results.. I will check with IDE 1.8

BennehBoy, 2017-03-02:
I've just recloned and can also not reproduce the problem, apologies, I must've done something wrong somehow.

Incidentally, I'm not sure if you have any STM32 devices available, but the output of 'item' also crashes those.  Most all else appears to work. (using stm32duino core)

neu-rah, 2017-03-02:
no problem, all feedback is always welcome, an dno I have no STM32

I only have some raw AVR's, some Uno's, Nano's, Pro, Due and ESP01's.

interesting that output crashes on STM, can you try with an empty action function. That print of item was just a demo thing, the purpose of the function is to do some action.

I will check the streaming operator and its variants among devices.

Thanks for the interest.

P.S. does STM have PROGMEM using flash? (googling)

BennehBoy, 2017-03-02:
No, STM32 doesn't use PROGMEM, it's Arm Cortex so already places const data in flash.

BennehBoy, 2017-03-02:
Empty action functions work just fine, it's only when the calling class instance is referenced that it crashes.

This crashes (serial output halts at "on")
```
result showEvent(eventMask e,navNode& nav,prompt& item) {
  Serial<<e<<" on "<<item<<endl;
  return proceed;
```

This does not
```
result showEvent(eventMask e,navNode& nav,prompt& item) {
  Serial<<e<<" on "<<endl;
  return proceed;
```

neu-rah, 2017-03-02:
i was looking at 
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/cores/maple/avr/pgmspace.h

as usual they just emulate PROGMEM macros for compatibility reasons, I'm using very similar macros here: https://github.com/neu-rah/ArduinoMenu/blob/master/src/macros.h#L4

for the arm and esp PROGMEM is just ignoring and we use ram because there is plenty of it, so maybe STM's need to be catch on the same line...

a nice test would be to just comment out the conditionals and force a compilation on STM to see if it works, if it works then we can patch that conditional to catch STM's too

can you comment those lines on macros.h file and try a compilation on STM?

```c++
//#if defined(__arm__) | defined(ESP8266)
  #define MENU_USERAM
//#endif
```

BennehBoy, 2017-03-02:
No change, pretty sure the __arm__ conditional already forced MENU_USERAM

BennehBoy, 2017-03-02:
The F4 core is probably not the best place to look, F1 is the most complete, but yes the pgm stuff is emulated, so usually the first thing to be done when porting is to remove all progmem related calls and replace any _p functions with the standard equivalent.

BennehBoy, 2017-03-02:
What other scenarios are there where it's useful to know the class instance which fired the event?  I can probably live without it.... :)

neu-rah, 2017-03-02:
yes, you probably can live without it and probably you already know the exact option that fired the event if you attribute unique functions to each option.

I will try to run the code on non-avrs and see if has the same error and why.

about the _P replacement, the STM macros already doing that... maybe some of mine are not doing it correct.

let me know if you managed to proceed without the item or not.

BennehBoy, 2017-03-02:
Will do, thanks.

BennehBoy, 2017-03-03:
I am managing to use the menu system OK on STM32 without item.

I have some other things I could do with asking however - related to driving mutliple SSD1306 displays at the same time (cleardisplay empties the frame buffer, and I believe the menu system does lazy writing, so only draws if anything has changed - this means that operations on the other displays result in a blank menu screen after the initial menu draw)....

Open new issue?

neu-rah, 2017-03-03:
yes, new issue please for the sake of people trying to find info in the future

closed 2017-03-04 by BennehBoy