neu-rah/ArduinoMenu #85: update/enterEvent after exitEvent for FIELD

sphh, 2017-03-02:
I have a callback for a FIELD menu item:
```C++
FIELD(timeOn, "On", "ms", 0, 500, 100, 10, go_timeon, anyEvent, noStyle),
```
The `go_timeon(eventMask event)` callback function sees the following events:

1. `enterEvent` with `enterCmd` (entering the edit mode)
2. `updateEvent` with every `upCmd` and `downCmd`
3. nothing with `enterCmd` (switching from big to small steps)
4. `updateEvent` with every `upCmd` and `downCmd`
5. `exitEvent`, `updateEvent` with `enterCmd` (leaving the edit mode)

The last `updateEvent` seems a bit odd to me. Is there any reason, why another `updateEvent` is emitted after the `exitEvent`? IMHO this event is not needed at all, because there was no update of the value.

The same sequence happens if the `useUpdateEvent` is set to false, but with `enterEvent` instead of `updateEvent`, sending an `enterEvent` after `exitEvent`.

neu-rah, 2017-03-03:
you can enter a numeric value too, in that case there is no up/down command

neu-rah, 2017-03-03:
still the logic can be refined to dispense the update unless changed and one of the enters

sphh, 2017-03-03:
The update is not the problem, it's the order that the `updateEvent` comes after the `exitEvent`. Just changing the order to `updateEvent` → `exitEvent` would be helpful.

sphh, 2017-03-29:
I made a little patch which seems to get rid of this. Please see #87 for the actual patch (which also handles unsigned variables).

neu-rah, 2017-10-20:
I have the idea that this solved, if not please reopen :D

closed 2017-10-20 by neu-rah