neu-rah/ArduinoMenu #86: Escaping from `FIELD` with `escCmd`?

sphh, 2017-03-02:
I tried to make a `FIELD` where the editing can be cancelled by `escCmd`, but it does not react to this command.

The procedure is simple. Here we want to modify the variable `int value`:

1. Set `useUpdateEvent` to `true` in `config`.
2. Have `FIELD` for `temp_value` and a callback called at `anyEvent`.
3. When entering edit mode (`enterCmd`) the `enterEvent` is emitted and the callback saves the current value into the temporary variable.
4. The temporary variable is changed.
5. When leaving the edit mode (with `enterCmd`), the `exitEvent` is emitted and the temporary value is saved back to the variable.
6. When leaving the edit mode with `escCmd`, the `cancelEvent` is emitted and the original value is restored.

The last point 6. is not possible at the moment. An example for the callback is given below.

Is that possible and does it make sense?

```C++
byte value = 128;
analogWrite(6, value);  // Dim a LED on pin 6

byte temp_value;
void go_value(eventMask event) {
  switch (event) {
    case enterEvent:
      // Store actual value in temporary variable
      temp_value = value;
      break;
    case updateEvent:
      // Show the user some feedback, e.g. dimming an LED
      analogWrite(6, temp_value);
      break;
    case exitEvent:
      // Confirmed exit, save the temporary value back
      value = temp_value;
      break;
    case cancelEvent:    // <- This does not exist
      // Value has not changed! Restore original brightness
      analogWrite(6, value);
      break;
  }
```

neu-rah, 2017-03-02:
yes it does not, because we have live control, we change the variable and the internal stored value on each change from iteration. its done by design to allow iteration feed back. The internal stored value is used by the reflexion system (it means, if you change the value on your code we know that the shown value had to be redrawn <== reflexivity).

Doing that case would require us to allocate an extra internal value (ram) to store the original value.

To honor memory conservation we've choose the shorter version (no cancel) and because **it can still be done** by deriving from the original field class.

There is s short example on using a customized prompt here https://github.com/neu-rah/ArduinoMenu/blob/master/examples/Serial/serialio/serialio/serialio.ino#L63

the idea is to allow customization of menu objects like number formatting, text editors, or cancel-able fields. without having to overload the base objects upfront with extra information

cancel-able can be generalized... but i'm not there yet. once we have a working code we can share it as an extra editing ability... sort of pluggins, to be used like the actual altPrompt.

```c++
class altPrompt:public prompt {
public:
  altPrompt(const promptShadow& p):prompt(p) {}
  idx_t printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len) override {
    return out.printRaw("special prompt!",len);
  }
};

...

altOP(altPrompt,"",showEvent,anyEvent)
```
altOP macro allows link in of user defined menu prompt objects.

neu-rah, 2017-03-03:
here is a field that reacts to esc and that preserves the original +value upon escaping

```c++
#include <Arduino.h>

#include <menu.h>
#include <menuIO/serialOut.h>

using namespace Menu;

int test=55;

//custom field edit
template<typename T>
class cancelField:public menuField<T> {
protected:
  T original;//to use when canceling
public:
  cancelField(const menuFieldShadow<T>& shadow):menuField<T>(shadow) {
    original=shadow.target();
  }
  void doNav(navNode& nav,navCmd cmd) {
    switch(cmd.cmd) {
      case escCmd:
        target()=original;
        //send enter bacause it might be the only catched event that updates
        //the hardware and because we are restoring the value.
        nav.event(options->useUpdateEvent?updateEvent:enterEvent);
        nav.root->exit();
      case enterCmd:
        if (menuField<T>::tunning||options->nav2D||!tune()) {//then exit edition
          menuField<T>::tunning=false;
          menuField<T>::dirty=true;
          nav.root->exit();
        } else menuField<T>::tunning=true;
        menuField<T>::dirty=true;
        break;
      case upCmd:
        target()+=(menuField<T>::tunning?tune():step())*(options->invertFieldKeys?-1:1);
        menuField<T>::dirty=true;
        clamp();
        nav.event(options->useUpdateEvent?updateEvent:enterEvent);
        break;
      case downCmd:
        target()-=(menuField<T>::tunning?tune():step())*(options->invertFieldKeys?-1:1);;
        menuField<T>::dirty=true;
        clamp();
        nav.event(options->useUpdateEvent?updateEvent:enterEvent);
        break;
      default:break;
    }
  }
};

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,altFIELD(cancelField,test,"Custom edit","%",0,100,10,1,doNothing,noEvent,wrapStyle)
  ,FIELD(test,"Original edit","%",0,100,10,1,doNothing,noEvent,wrapStyle)
  ,EXIT("<Back")
);

#define MAX_DEPTH 1

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

NAVROOT(nav,mainMenu,MAX_DEPTH,Serial,out);

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial<<"menu 3.0 testing custom fields"<<endl;Serial.flush();
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
```

if this runs well i might make the first plugin with it, along with an example of how to add custom components

i've also taken the opportunity to review the event sending logic, the _dirty_ relly was not ok but i still send enter|update event on cancel, because we are potentially changing the value by restoring it and hardware could already been updated by a previous change (live update).
and some users just might submit to the enter event.

neu-rah, 2017-03-03:
lloking at it, dirty should have been set on value restore, and this enter thing is really a compatibility thing with 2.x and maybe i should drop it in favor of a more clear logic as you suggest...

neu-rah, 2017-03-03:
same with some refinement 
 ```c++
//custom field edit
template<typename T>
class cancelField:public menuField<T> {
protected:
  T original;//to use when canceling
public:
  cancelField(const menuFieldShadow<T>& shadow):menuField<T>(shadow) {
    original=target();
  }
  void doNav(navNode& nav,navCmd cmd) {
    switch(cmd.cmd) {
      case escCmd:
        target()=original;
        //send enter bacause it might be the only catched event that updates
        //the hardware and because we are restoring the value.
        menuField<T>::tunning=false;
        menuField<T>::dirty=true;
        nav.event(options->useUpdateEvent?updateEvent:enterEvent);
        nav.root->exit();
      case enterCmd:
        if (menuField<T>::tunning||options->nav2D||!tune()) {//then exit edition
          menuField<T>::tunning=false;
          menuField<T>::dirty=true;
          original=target();
          nav.root->exit();
        } else menuField<T>::tunning=true;
        menuField<T>::dirty=true;
        break;
      case upCmd:
        target()+=(menuField<T>::tunning?tune():step())*(options->invertFieldKeys?-1:1);
        menuField<T>::dirty=true;
        clamp();
        nav.event(options->useUpdateEvent?updateEvent:enterEvent);
        break;
      case downCmd:
        target()-=(menuField<T>::tunning?tune():step())*(options->invertFieldKeys?-1:1);;
        menuField<T>::dirty=true;
        clamp();
        nav.event(options->useUpdateEvent?updateEvent:enterEvent);
        break;
      default:break;
    }
  }
};
```

sphh, 2017-03-03:
Thanks!

This does not compile:
```
In file included from /.../sketchbook/sketch_feb27a/sketch_feb27a.ino:121:0:
sketch/lcdmenu.h: In constructor 'Menu::cancelField<T>::cancelField(const Menu::menuFieldShadow<T>&)':
sketch/lcdmenu.h:41:22: warning: there are no arguments to 'target' that depend on a template parameter, so a declaration of 'target' must be available [-fpermissive]
      original=target();
```
When I sprinkled some `menuField<T>::` over the code, it compiled (for the updated code see below). I don't know if the following problems are due to this.

Unfortunately it has two problems:
1. When entering the edit-mode the current value is not saved in `original`. It is shown and can be edited, but when cancelling/escaping it cannot be restored and `0` is written back into the variable.
2. When cancelling the editing, a re-entering gives the following behaviour: Cancelling during `:` mode (big steps), re-entering puts one into `>` mode (fine steps). Cancelling during `>` mode brings one back to `>` mode. I would expect in both cases to re-enter in the `:` mode.

Is there a way to solve these problems?

```C++
//custom field edit
template<typename T>
class cancelField:public menuField<T> {
protected:
 T original;//to use when canceling
public:
 cancelField(const menuFieldShadow<T>& shadow):menuField<T>(shadow) {
   original=menuField<T>::target();
 }
 void doNav(navNode& nav,navCmd cmd) {
   switch(cmd.cmd) {
     case escCmd:
       menuField<T>::target()=original;
       //send enter bacause it might be the only catched event that updates
       //the hardware and because we are restoring the value.
       menuField<T>::tunning=false;
       menuField<T>::dirty=true;
       nav.event(options->useUpdateEvent?updateEvent:enterEvent);
       nav.root->exit();
     case enterCmd:
       if (menuField<T>::tunning||options->nav2D||!menuField<T>::tune()) {//then exit edition
         menuField<T>::tunning=false;
         menuField<T>::dirty=true;
         nav.root->exit();
       } else menuField<T>::tunning=true;
       menuField<T>::dirty=true;
       break;
     case upCmd:
       menuField<T>::target()+=(menuField<T>::tunning?menuField<T>::tune():menuField<T>::step())*(options->invertFieldKeys?-1:1);
       menuField<T>::dirty=true;
       menuField<T>::clamp();
       nav.event(options->useUpdateEvent?updateEvent:enterEvent);
       break;
     case downCmd:
       menuField<T>::target()-=(menuField<T>::tunning?menuField<T>::tune():menuField<T>::step())*(options->invertFieldKeys?-1:1);;
       menuField<T>::dirty=true;
       menuField<T>::clamp();
       nav.event(options->useUpdateEvent?updateEvent:enterEvent);
       break;
     default:break;
   }
 }
};
```

neu-rah, 2017-03-03:
yes it has, i've been working on it since yesterday and already fixed that... it seems to work ok here with a shorter version using the non revised event send logic. but i've revised logic also on the original field, so that we can just call the original and have same behavior. I hope this will be included on next push

```c++
template<typename T>
class cancelField:public menuField<T> {
protected:
  T original;//to use when canceling
  bool editing;
public:
  cancelField(const menuFieldShadow<T>& shadow):menuField<T>(shadow),editing(false) {}
  void doNav(navNode& nav,navCmd cmd) override {
    if (!editing) {
      original=target();
      editing=true;
    }
    switch(cmd.cmd) {
      case escCmd:
        editing=false;
        target()=original;
        menuField<T>::tunning=true;
        break;
      case enterCmd:
        if (menuField<T>::tunning||options->nav2D||!tune())
          editing=false;
        break;
      default: break;
    }
    menuField<T>::doNav(nav,cmd);
  }
};
```

sphh, 2017-03-03:
Yes that works. And it is clever idea not to leave the editing mode but simply reset the value to the original value!!

Again I had to add some `menuField<T>::`s in front of `target()` and `tune()` to get it to compile.

Thank you for your help!

One last thing: How can I access the `original` value in my callback?

neu-rah, 2017-03-03:
my pleasure :)

There are several ways of accessing the object
1 - your callback can receive the prompt item.. being a prompt it has not that information, but if you only deliver that kind of callback to the same kind of object then its safe to cast to the original kind `((cancelField*)&item)->...` here you can either make the T original public and access to it or add a function to your class that return the original and call it here

I had some reports that item is not behaving well on STM32, and i'm not sure if it is limited just to STM32, so if you have troubles please let me know.

2-if your callback is attributed to a single menu item then you can index on the menu that contains it
ex: `mainMenu[prompt_index]` to obtain the element 

3-or call the navRoot::selected() if the event is called on a selected element (might not).

etc...

all of them require the cast, and because all require the cast i would go for the first unless the received pointer is not ok (possible STM32 Bug)

about not leaving edit on escape... it might be a possibility, an option would be nice, not wanting to add extra memory cargo to the class we could make either a static option (for all cancelField items) or template meta-programming generating two possible versions... interesting.

thanks for your good suggestions, they are welcome.

neu-rah, 2017-03-25:
#a note on cancel-able fields
should not be used to control some types of hardware, like a motor, because the cancel could produce a 
abrupt change in the value and therefor on the hardware that might not be adequate or desirable.

sphh, 2017-03-29:
Just to wrap it up:

I use the code below for cancel-able fields. Pressing `cancel` will restore the original value and ends the editing mode (whereas the code from above restores the original value but stays in editing mode requiring you to press `enter` once or twice to get out of it):

```C++
namespace Menu {
  template<typename T>
  class cancelField:public menuField<T> {
  protected:
    T original;  // to be used when cancelling
    bool editing;
  public:
    cancelField(const menuFieldShadow<T>& shadow): menuField<T>(shadow), editing(false) {}
    void doNav(navNode& nav, navCmd cmd) override {
      if (!editing) {
        original = menuField<T>::target();
        editing = true;
      }
      switch(cmd.cmd) {
        case escCmd:
          editing = false;
          menuField<T>::target() = original;
          menuField<T>::tunning = false;
          menuField<T>::dirty = true;
          nav.event(options->useUpdateEvent ? updateEvent : enterEvent);
          nav.root->exit();
          return;
        case enterCmd:
          if (menuField<T>::tunning || options->nav2D || !menuField<T>::tune())
            editing = false;
          break;
      }
      menuField<T>::doNav(nav, cmd);
    }
  };
}
```//namespace Menu

Thanks @neu-rah !

closed 2017-03-29 by sphh