neu-rah/ArduinoMenu #88: Multiple SSD1306 screens being updated causes menu to not display

BennehBoy, 2017-03-03:
I am building an in vehicle gauge system.

It uses 8 SSD1306 OLED displays to output readings from a combination of multiple analogue sensors & OBD inputs.

I like this menu system because it is non blocking.

This means that the menu can be used whilst the remaining displays are outputting sensor data.

I am using Adafruits SSD1306 library to drive my displays, it uses an in RAM framebuffer.  This framebuffer is shared by each display - this is not an issue since each display is drawn sequentially.

My code sets a flag when the users presses a button to enter menu mode, display 1 is then updated only by the menu system.

However, I believe (perhaps incorrectly) that the menu system uses lazy updates.  This means that when my remaining screens clear the framebuffer, the menu is no longer displayed.

My workaround is to not draw the other displays when in menu mode, but it would be good if a 'force full redraw' option could be included.  This is of course if my assumption about the lazy updates is correct.

neu-rah, 2017-03-03:
the menu output devices have a redraw flag, for use on devices that need to full redraw on every output, because they start as a clear buffer. That flag should be on at U8GLib and U8G2 because they use fresh buffers, however the example shows a code that only draws when menu change detected, to minimize draws... i think that you can elaborate on that and also call for drawing when you know the buffer has been cleared.

there is also a way of signaling that the menu need to redraw... just not on top of my head now, but let me know if the above is not working

neu-rah, 2017-03-03:
BTW: this menu started for gauge live control ;)

BennehBoy, 2017-03-03:
Thanks, I'll do some reading about how to set the redraw flag.

Have you a link to the original project which inspired the menus?  It would be interesting to see.

BennehBoy, 2017-03-03:
Hmm, I tried nav.doOutput();

Unfortunately it resulted in the same issue - although it's late on a Friday and I've had a few beers :)

I'll take another look in the morning.

neu-rah, 2017-03-04:
the redraw flag is already set for U8g's

are you using the firstpage/nextpage thing of u8g?

BennehBoy, 2017-03-04:
I'm not using u8g? Sorry I'm probably misunderstanding.

Are you suggesting that I should use u8g to drive just the menu display?

neu-rah, 2017-03-04:
sorry my bad, whats is your gfx library?

BennehBoy, 2017-03-04:
Yes, Adafruit_SSD1306 & Adafruit_GFX :)

neu-rah, 2017-03-04:
anyway, setting it latter is something like this:

```c++
out[0].style|=menuOut::redraw;
```

where:

**out** is your menu output manager

**out[0]** the first menuOut device

but now i'm not so sure if its your case, i was thinking about u8g

also by setting your menu::dirty member to true will force next redraw on all output devices, i hope :D

```c++
nav.active().dirty=true;//for a menu
nav.navFocus->dirty=true;//should invalidate also full screen fields assert(nav.navFocus!=NULL)
```

set that when you know the buffer has been cleared

BennehBoy, 2017-03-04:
Works, excellent!

BennehBoy, 2017-03-04:
Although it also of course forces the serial output to redraw - but I can live with that I'm using that for debugging purposes only.

neu-rah, 2017-03-04:
nice :D

one of the experiments with an ancestor of this menu, controlling a motor with PID, sustain rotain under variable load, the idea was to later go into a car.

https://www.youtube.com/watch?v=J1c3t0Zr0Sw

here an even older experiment, rudimentar car motor control

https://www.youtube.com/watch?v=R8YIdzfnRFg

BennehBoy, 2017-03-04:
Looks interesting, launch control?

Here's a tiny video of my gauge system on the bench, menus don't have any functions behind them yet

https://www.youtube.com/watch?v=7fDYw-HfoS0

Next to do is non blocking button code, but I think I've found a library to cover that.

neu-rah, 2017-03-04:
menu supports non-blocking buttons

neu-rah, 2017-03-04:
nice gauge system you have there :D
I'm more a software guy...

about non flicking all devices:

A
if menu not dirty yet (and buffer cleared):
    1. signal menu dirty
    2. ask specific device to draw
    3. clean up dirty flags to avoid other devices to draw
otherwise just proceed as normal

B
or use device with redraw flag call only when menu changed or buffer cleared

neu-rah, 2017-03-04:
well we can chat on other channels

BennehBoy, 2017-03-04:
Oops, closed the issue without refreshing the page so I missed your comments!

I'll try the device specific dirty/clear later on.

I'm more hardware :D  Especially hardware with large enignes in :D

I'm going to use an external button system because I want the menu to be inactive in certain states, long press of select enables the menu.

What other options are there for discussions without fouling up these issues?

neu-rah, 2017-03-04:
still you can use the buttons on and off the menu, they are just buttons

BennehBoy, 2017-03-04:
Well, I couldn't figure out how to force just one display to draw.

I'm not really that worried about it.  But it may be useful for others if they come looking.

What I can't figure out is how to reset the menu back to top level and first item selected - I need to go read the wiki.

neu-rah, 2017-03-04:
and i depend on this questions to write the wiki :D no worries

reset current menu to top `nav.doNav(navCmd(idxCmd,0))`

BennehBoy, 2017-03-04:
Thankyou.

closed 2017-03-05 by BennehBoy