neu-rah/ArduinoMenu #89: Streaming.h (and hence ArduinoMenu) conflict with SdFat.h

BennehBoy, 2017-03-05:
See here -> https://forum.arduino.cc/index.php?topic=374550.0

Using SdFat would greatly simplify my project (very simple to do datalogging), but this error results if I include the library:

```
error: previous declaration of 'ostream& endl(ostream&)'

 inline ostream& endl(ostream& os) {

                 ^

exit status 1
```

Do you have any suggestions of a workaround?

neu-rah, 2017-03-05:
that can be fixed

BennehBoy, 2017-03-05:
Will that need changes to the menu system, or is it something I can do myself - I'm not really a software guy! :-)

neu-rah, 2017-03-05:
it has to do with streaming, or the lack of them and me implementing streams on the menu.

I think i use them for debug... but have to check first if its not used outside debug, i can use normal printing if i find any on non debug code.

for a quick try you can just comment the lines and see if the compiler complains about it (probably not) even if it is used, it will use the existing. i guess :D

BennehBoy, 2017-03-05:
These are the fatal ones:

`In file included from C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:12:0:

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In member function 'Menu::menuOut& Menu::navRoot::printPath(Menu::menuOut&) const':

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:487:39: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'char')

           for(idx_t n=0;n<level;n++) o<<'/'<<path[n].sel;

		   
C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In instantiation of 'Menu::idx_t Menu::menuField<T>::printTo(Menu::navRoot&, bool, Menu::menuOut&, Menu::idx_t, Menu::idx_t) [with T = int; Menu::idx_t = signed char]':

C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:1173:1:   required from here

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:546:12: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'char')

         out<<((root.navFocus==this&&sel)?(tunning?'>':':'):' ');

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In instantiation of 'void Menu::menuField<T>::printValue(Menu::menuOut&) const [with T = int]':

C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:1173:1:   required from here

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:143:54: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'const int')

         void printValue(menuOut& o) const override {o<<reflex;}

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In instantiation of 'void Menu::menuField<T>::printHigh(Menu::menuOut&) const [with T = int]':

C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:1173:1:   required from here

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:144:53: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'int')

         void printHigh(menuOut& o) const override {o<<high();}

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In instantiation of 'void Menu::menuField<T>::printLow(Menu::menuOut&) const [with T = int]':

C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:1173:1:   required from here

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:145:52: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'int')

         void printLow(menuOut& o) const override {o<<low();}

 
C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In instantiation of 'Menu::idx_t Menu::menuVariant<T>::printTo(Menu::navRoot&, bool, Menu::menuOut&, Menu::idx_t, Menu::idx_t) [with T = bool; Menu::idx_t = signed char]':

C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:1173:1:   required from here

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:641:10: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'char')

       out<<(this==&root.active()?':':' ');`

There are also multiple notes pertaining to no known conversions

BennehBoy, 2017-03-05:
That's without sdfat included, with it included the errors are similar:

```
C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In member function 'Menu::menuOut& Menu::navRoot::printPath(Menu::menuOut&) const':

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:487:39: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'char')

           for(idx_t n=0;n<level;n++) o<<'/'<<path[n].sel;
```

neu-rah, 2017-03-06:
this sdfat?
https://github.com/greiman/SdFat

neu-rah, 2017-03-06:
on menu.h file replace the Streaming include with:
```c++
  #if !defined(ArduinoStream_h)
    #include <Streaming.h>
  #endif
```

and some lines after th edefinition of endl should be:

```c++
    #if !defined(ARDUINO_STREAMING) || !defined(ArduinoStream_h)
      #define endl "\r\n"
    #endif
```

it compile here ok with sdfat... and will include on next push if it works ok

BennehBoy, 2017-03-06:
Thanks, I'll try it after work :thumbsup:

BennehBoy, 2017-03-06:
I'm afraid I still get errors.

If I include sdfat.h prior to the menus:

```
In file included from C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:12:0:

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h: In member function 'Menu::menuOut& Menu::navRoot::printPath(Menu::menuOut&) const':

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:489:39: error: no match for 'operator<<' (operand types are 'Menu::menuOut' and 'char')

           for(idx_t n=0;n<level;n++) o<<'/'<<path[n].sel;
```

If I include the menus first:

```
In file included from C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:12:0:

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:35:20: error: expected unqualified-id before string constant

       #define endl "\r\n"

                    ^

```

There are more but I've just copied the first occurrences.  I've checked that I have the latest master versions of the repo's for both.

Here's the top of menu.h with the changes: (google seems to think the 2nd error is down to a missing semi-colon somewhere?)

```
#ifndef RSITE_ARDUINO_MENU_SYSTEM
  #define RSITE_ARDUINO_MENU_SYSTEM
  #include <Arduino.h>
  #if !defined(ArduinoStream_h)
    #include <Streaming.h>
  #endif
  #include "menuBase.h"
  #include "shadows.h"

  namespace Menu {

    static const char* numericChars="0123456789.";

    #define _MAX(a,b) (((a)>(b))?(a):(b))
    //#if defined(ESP8266)
    //#if !defined(endl)
    #if !defined(ARDUINO_STREAMING) || !defined(ArduinoStream_h)
      #define endl "\r\n"
    #endif

```

neu-rah, 2017-03-06:
the top here
```c++
#ifndef RSITE_ARDUINO_MENU_SYSTEM
  #define RSITE_ARDUINO_MENU_SYSTEM
  #include <Arduino.h>
  #if !defined(ArduinoStream_h)
    #include <Streaming.h>
  #endif
  #include "menuBase.h"
  #include "shadows.h"

  namespace Menu {

    static const char* numericChars="0123456789.";

    #define _MAX(a,b) (((a)>(b))?(a):(b))
    //#if defined(ESP8266)
    //#if !defined(endl)
    #if !defined(ARDUINO_STREAMING) || !defined(ArduinoStream_h)
      #define endl "\r\n"
    #endif
```

neu-rah, 2017-03-06:
they seem identical and i'm compiling here.. for uno/nano with SdFat

neu-rah, 2017-03-06:
be sure to include sdfat before the menu.h

BennehBoy, 2017-03-06:
Yes I thought it might be an stm32 so I've tested it within codectrl against an uno - with sdfat included before menu.h and I get the same error.

I recloned sdfat, streaming, and checked arduinmenu - only changes are those you suggested :(

What else could be different/causing this?

neu-rah, 2017-03-06:
well i have some other changes here because this is my dev version, i will try to make a functional version and push it... will let you know

BennehBoy, 2017-03-06:
I suspect it's more likely to be something my side

neu-rah, 2017-03-06:
this last error

In file included from C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:12:0:

C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:35:20: error: expected unqualified-id before string constant

       #define endl "\r\n"

means the #if directive had no success on skiping the endl definition, because SdFat also defines it and probably as a string the code end up something like 

#define "\r\n" "\r\n"

that is not legal

i've checked the sdfat library and they define ArduinoStream_h here:

https://github.com/greiman/SdFat/blob/bc32f5afdd35f8344cc6953c99287b3dc4255667/SdFat/src/FatLib/ArduinoStream.h#L21

but yours seem to not... also i'm using atom+platformio, will check on arduino ide 1.6 or 1.8

BennehBoy, 2017-03-06:
Line 20/21 of same file on my hdd:

```
#ifndef ArduinoStream_h
#define ArduinoStream_h
```

I recloned the library earlier today to be sure...

neu-rah, 2017-03-06:
i'm just including <SdFat.h> but not using it, maybe if i initialize the lib... 

arduino ide 1.6 also compiling ok with just the SdFat include, i've pushed my current code to branch dev on ArduinoMenu

BennehBoy, 2017-03-06:
This is the only reference to endl that I can find in the sdfat src code:

SdFat\src\FatLib\iostream.h
	Line 40: inline ostream& endl(ostream& os) {

BennehBoy, 2017-03-06:
No luck with dev branch either:

```
In file included from C:\Users\bennehboy\Documents\Arduino\libraries\ArduinoMenu\src/menu.h:21:0,

                 from C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:12:

C:\Users\bennehboy\Documents\Arduino\libraries\Streaming/Streaming.h:114:21: error: 'endl' redeclared as different kind of symbol

 enum _EndLineCode { endl };

                     ^

In file included from C:\Users\bennehboy\Documents\Arduino\libraries\SdFat\src/FatLib/bufstream.h:27:0,

                 from C:\Users\bennehboy\Documents\Arduino\libraries\SdFat\src/FatLib/ArduinoStream.h:29,

                 from C:\Users\bennehboy\Documents\Arduino\libraries\SdFat\src/FatLib/FatLib.h:23,

                 from C:\Users\bennehboy\Documents\Arduino\libraries\SdFat\src/SdFat.h:28,

                 from C:\Users\bennehboy\Documents\Arduino\LRDuino\LRduinoV0.01\LRDuino\LRDuino.ino:5:

C:\Users\bennehboy\Documents\Arduino\libraries\SdFat\src/FatLib/iostream.h:40:17: error: previous declaration of 'ostream& endl(ostream&)'

 inline ostream& endl(ostream& os) {

                 ^

exit status 1
```

neu-rah, 2017-03-06:
with the dev branch of the menu?

BennehBoy, 2017-03-06:
Tried against both nano/uno & stm32

nano/uno throws more errors, stm32 just the above

BennehBoy, 2017-03-06:
I'm using this streaming.h -> https://github.com/scottdky/Streaming

Thought it worth checking :D

neu-rah, 2017-03-06:
yup, do you include it on your sketch?

because the:
 #if !defined(ArduinoStream_h)
    #include <Streaming.h>
  #endif

was the instruction to not use it when we are using SdFat, but if there are other includes it will be included

BennehBoy, 2017-03-06:
No:

#include <Adafruit_SSD1306.h>
#include <SdFat.h>
#include "LRDuinoGFX.h"
#include <Fonts/FreeSansBoldOblique12pt7b.h>
#include <Fonts/FreeSansBoldOblique24pt7b.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>
#include <Adafruit_HMC5883_U.h>
#include <menu.h>
#include <menuIO/serialOut.h>
#include <menuIO/chainStream.h>
#include <menuIO/adafruitGfxOut.h>
#include <Buttons.h>

Also, I get the same error when compiling codectrl.ino from the examples if I include sdfat.h before menu.h

neu-rah, 2017-03-06:
welp dev was not a good push, travis is failing all tests..please forget the dev branch, its a mess... it seems it compiles only here.

but using codeCtrl or some other example is an excellent start, this was i can put assumption about your code aside and work on the examples, i will stick with the master branch for testing...

neu-rah, 2017-03-06:
all right! i was able to reproduce it here, by using master branch and codeCtrl with SdFat :D

at least now i have a target... and i'm on top of it :D

BennehBoy, 2017-03-06:
cool. thanks for your efforts.

neu-rah, 2017-03-07:
Ok, got dev branch build passing on travis, master became a mess after me merging stuff...

Removed the usage of stream (<<) from non debug code, it happens that I was using quite a lot on examples.

neu-rah, 2017-03-07:
master branch also compiling on travis and with stream dependency removed from non-debug code and from examples

BennehBoy, 2017-03-07:
Works!

I'll do some more testing then close this off.

Thanks you!

BennehBoy, 2017-03-07:
sdfat library also working fine.  👍

closed 2017-03-07 by BennehBoy

neu-rah, 2017-03-07:
using master branch?

BennehBoy, 2017-03-07:
Yes

BennehBoy, 2017-03-09:
Hey sorry to tack this on here but I couldn't think how else to ask...

Is there any way that I can dynamically generate a scrollable list of items (doesn't need to be any action other maybe a BACK) from an array/struct etc?  I think this would be a fantastic feature if somehow part of the code could be called to draw this.

Anyhow, just a thought before I try and hackjob something!

neu-rah, 2017-03-09:
yes it would, but at moment not really. i've done some dissertations about it but not any tests yet.

about other contacts, my youtube channel has links to my facebook and G+

lostangles, 2017-03-29:
Hey quick question since BennehBoy brought this up earlier this month and I was trying to do a similar thing.

I'm trying to read through a list of SDCard files.  Is there an example or way to do this using Arduino Menu?  I figured there was a way to iterate over an array or a list, but the above leads me to believe there isn't.  

Is there any mechanism within this library that would help with that?

neu-rah, 2017-03-29:
the short answer: for the moment no

this is a static (PROGMEM menu) so all of its structure must be known at compile time... for mcu's that do not have PROGMEM or that simulate it with RAM as is the case of STM, Arduino due, arms, esp's, etc... the menu is then on ram.
I had some success doing dynamic creation of menu's on such systems, but its not yet clear about the integration with the API. Probably will had to do extensive classification of existing data with preprocessor conditional... or something else based of class derivation (not explored this path yet)