neu-rah/ArduinoMenu #92: Blank screen after splashscreen when starting with idle menu

Lu0z, 2017-04-01:
Hello Rui
The project is almost complete, i just need to fix a couple of things.

I'm having a little issue with the menu idling:
i made a little splash screen at the start, after that i'm trying to show a control a frontpage while having the menu idling. My problem is that afte the splash screen ends, instead of showing that page, the screen goes blank. i think that something in my code is missing at this point. When i quit the menu i get redirected to the page correctly, it seems like the idle event fails to be triggered.

```
 void setup() {
    pinMode(encBtn,INPUT_PULLUP);
    Serial.begin(9600);
    while(!Serial);
    encoder.begin();
    lcd.begin(20,4);
    lcd.setCursor(0, 0);
    lcd.print("Lambda-Caster VHF");
    lcd.setCursor(0, 1);
    lcd.print("Model: D105CH");
    lcd.setCursor(0, 2);
    lcd.print("FIRMWARE:");
    lcd.setCursor(0, 3);
    lcd.print("TransDriver v1.0");
    delay(3000);
    nav.idleOn();
    nav.idleTask=idle;
  }
```

Another little thing i'd like to ask you:
i saw on your video [https://www.youtube.com/watch?v=wHv5sU-HXVI&](url)

that you were changing float values in a field function. how did you do that?

neu-rah, 2017-04-01:
remove these two lines 
```c++
    nav.idleOn();
    nav.idleTask=idle;
```
an idle menu is like non-existing, guess its not what you want, the delay is enough

about float or other numeric values, just use them, the menu will adapt to your variable type because it is a template

neu-rah, 2017-04-01:
can you paste your idle function?

Lu0z, 2017-04-01:
this is the idle function.
```
 result idle(menuOut& o,idleEvent e) {
    if (e==idling){
 
    lcd.setCursor(0,0);
    lcd.print("Tx:");
    lcd.print(TxFrq);
    lcd.print("Mhz");

    lcd.setCursor(0,1);
    lcd.print("Rx:");
    lcd.print(RxFrq);
    lcd.print("Mhz");
    }
    return proceed;
  }
```

for what it concerns the float value part, i'm doing this

```
float TxFrq=134.0000;
float RxFrq=134.0000;
float ChSpVal=0.1250;

MENU(TUNING,"Tuning",doNothing,noEvent,noStyle
  ,FIELD(TxFrq,"Tx Freq","MHz",134.0000,174.0000,ChSpVal,0,Tune,exitEvent,wrapStyle)
  ,FIELD(RxFrq,"Rx Freq","MHz",134.0000,174.0000,ChSpVal,0,Tune,exitEvent,wrapStyle)
  ,SUBMENU(ChSpSel)
  ,EXIT("<Back")
);

```

also i put an if cycle under void loop for changing  ChSpVal in 0.1250 or 0.2500 using another variable goin from 0 to 1.

the float values do not change on display.

neu-rah, 2017-04-01:
confirmed non-incrementing floats, yesterday fix for unsigned values validation has broken the  validation of floats and other values (just considering int's), its fixed and will be included on next push.
On your code i can see that you are trying to refer to a step variable... this wont work well because this is a static menu (all structure is known at compile time). Its a demand for using PROGMEM.
So the menu will be built on PROGMEM with the initial value 0.1250 and will not use other, no mather what you set the variable after.

solutions:
1 - use separate menu item for each step (easy)
2 - use a translated value by implementing a custom draw field <-- number of total step will still be const (not good)
3 - change the value on the handler functions <--  number of total steps will still be const (not good)
4 - use a custom field and override the doNav function, this is the best way (more diffcult)

neu-rah, 2017-04-01:
fixed float stepping

https://github.com/neu-rah/ArduinoMenu/commit/994034e603a9e0c67cef5f7fe4ec5d83a2581371

neu-rah, 2017-04-01:
here's an example of a custom field that can have a variable step

```c++
//custom field edit
//implementing a customized menu component
//this components can be latter shared as plugins
template<typename T>
class customStepField:public menuField<T> {
public:
  customStepField(const menuFieldShadow<T>& shadow):menuField<T>(shadow) {}
  void doNav(navNode& nav,navCmd cmd) override {
    switch(cmd.cmd) {
      case upCmd:
        customStep(1);
        break;
      case downCmd:
        customStep(-1);
        break;
      default:
        menuField<T>::doNav(nav,cmd);
        break;
    }
  }
  void customStep(T increment) {
    T thisstep = increment*ChSpVal;//this  using a custom step!
    prompt::dirty=true;
    if (thisstep < 0 && (target()-low()) < -thisstep) {
      if (style()&wrapStyle) {
        target() = high();
      } else {
        target() = low();
      }
    } else if (thisstep > 0 && (high()-target()) < thisstep) {
      if (style()&wrapStyle) {
        target() = low();
      } else {
        target() = high();
      }
    } else {
      target() += thisstep;
    }
  }
};

MENU(TUNING,"Tuning",doNothing,noEvent,noStyle
  ,altFIELD(customStepField,TxFrq,"Tx Freq","MHz",134.0,174.0,0.125,0,Tune,exitEvent,wrapStyle)
  ,FIELD(RxFrq,"Rx Freq","MHz",134.0000,174.0000,ChSpVal,0,Tune,exitEvent,wrapStyle)
  ,SUBMENU(ChSpSel)
  ,EXIT("<Back")
);
```

Lu0z, 2017-04-01:
so what you are saying is that i cannot change the value of the increment step dynamically with a variable?

the option number one would be a messy result because it would be like having 2 menus and also there would be a ton of more code to put in because that value will be part of a data stream trough uart.

i'd like to understand what you have in mind in option four.

also going back to the first problem, found anything on that idle function?

i'm sorry for bothering you this much, you are really helping me a lot and i'm really grateful for this.
This is a really difficult project for me because my coding skill are not that good, i still need to learn a lot, 
thank you very much for your patience.

neu-rah, 2017-04-01:
:D no problem, and thanks for your interest. this tests and applications are important, otherwise what would be the purpose of a menu that no1 uses ;)

thank for your patience.

my last comment was about option number 4, using a custom field

i've just code a custom field that use a variable for step (that's your case) the example include the usage in menu with the altFIELD macro.

let me know if you need to tune it

Lu0z, 2017-04-01:
let's try it!

i was thinking to fix the variable at 12.5khz because that would have done anyway. Variable channel space is just a feature of the module that may come handy in some particular cases.

anyway if it's possible to fix it, i'm eager to try it :)

neu-rah, 2017-04-01:
that code uses 12.5 and 25 as defined by the menu

closed 2017-04-08 by neu-rah