neu-rah/ArduinoMenu #94: Possible new bug on i2c display

Lu0z, 2017-04-01:
Hi Rui

i think i found a little bug with the I2C 20x4, same setup as the last time.

i'll try to explain:
 
starting from my splashcreen code:
````
 Serial.begin(9600);
    while(!Serial);
    encoder.begin();
    lcd.begin(20,4);
    lcd.setCursor(0, 0);
    lcd.print("Lambda-Caster VHF");
    lcd.setCursor(0, 1);
    lcd.print("Model: D105CH");
    lcd.setCursor(0, 2);
    lcd.print("FIRMWARE:");
    lcd.setCursor(0, 3);
    lcd.print("TransDriver v1.0");
    delay(3000);
````

in the fourth row the "1.0" gets cut off, i can only see "TransDriver v"

if i try to edit the text by filling it with characters in order to light up the whole row, the text is cut off anyway.

but if i try to fill a fourth row text generated by the menu, like the "<Back" function, the splash screen is shown properly and obviously that one "<Back" function shows all the extra characters. 

honestly for a moment i belived that my lcd was broken, so i build another sketch for filling all the columns and rows at once, all in a half minute because i was totaly terrorized hahahaha.

neu-rah, 2017-04-01:
I have a similar setup here, running your code (partial) and i can see the v1.0 ok
it might be something you are printing somewhere on some part of the code (that i do not have here), or a broken LCD :O

Lu0z, 2017-04-01:
ok i will give you the full code. actually this is weird because i tested the lcd with all teh digits lighten up. and it works, must be the code then!

```
#ifndef ARDUINO_SAM_DUE
  #include <DRA818.h>
  #include <Wire.h>
  #include <LiquidCrystal_I2C.h>//F. Malpartida LCD's driver
  #include <menu.h>//menu macros and objects
  #include <menuIO/lcdOut.h>//malpartidas lcd menu output
  #include <menuIO/encoderIn.h>//quadrature encoder driver and fake stream
  #include <menuIO/keyIn.h>//keyboard driver and fake stream (for the encoder button)
  #include <menuIO/chainStream.h>// concatenate multiple input streams (this allows adding a button to the encoder)

  using namespace Menu;

  //LiquidCrystal_I2C lcd(0x27);//, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
  LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address and pinout

 // Encoder /////////////////////////////////////
  #define encA 2
  #define encB 3
  //this encoder has a button here
  #define encBtn 4

  encoderIn<encA,encB> encoder;//simple quad encoder driver
  #define ENC_SENSIVITY 4
  encoderInStream<encA,encB> encStream(encoder,ENC_SENSIVITY);// simple quad encoder fake Stream

  //a keyboard with only one key as the encoder button
  keyMap encBtn_map[]={{-encBtn,options->getCmdChar(enterCmd)}};//negative pin numbers use internal pull-up, this is on when low
  keyIn<1> encButton(encBtn_map);//1 is the number of keys

  //input from the encoder + encoder button + serial
  Stream* inputsList[]={&encStream,&encButton,&Serial};
  chainStream<3> in(inputsList);//3 is the number of inputs

float TxFrq=134.0000;
float RxFrq=134.0000;
float ChSpVal=1250;
int ChSp=0;
int PDEF=0;
int HPF=0;
int LPF=0;

result Tune(eventMask e,navNode& nav,prompt& item) {
    Serial.print("Tuning Updated! ");
    Serial.print("TX is ");
    Serial.println (TxFrq);
    Serial.print("RX is ");
    Serial.println(RxFrq);
    Serial.print("Ch. Space is");
    Serial.println(ChSp); 
   
    return proceed;
  }

  result Dispclear(eventMask e,navNode& nav,prompt& item) {
       lcd.clear();
      return proceed;
  }
  

SELECT(ChSp,ChSpSel,"Channel Space",Tune,noEvent,noStyle
  ,VALUE("12.5 KHz",0,Tune,enterEvent)
  ,VALUE("25 KHZ",1,Tune,enterEvent)
);

MENU(TUNING,"Tuning",doNothing,noEvent,noStyle
  ,FIELD(TxFrq,"Tx Freq","MHz",134.0000,174.0000,ChSpVal,0,Tune,exitEvent,wrapStyle)
  ,FIELD(RxFrq,"Rx Freq","MHz",134.0000,174.0000,ChSpVal,0,Tune,exitEvent,wrapStyle)
  ,SUBMENU(ChSpSel)
  ,EXIT("<Back")
);

result Filter(eventMask e,navNode& nav,prompt& item) {
    Serial.print("filters Updated!");
    
    return proceed;
  }

SELECT(PDEF,PDEFSel,"Pre/De Emph.",doNothing,exitEvent,noStyle
  ,VALUE("OFF",0,Filter,enterEvent)
  ,VALUE("ON",1,Filter,enterEvent)
);

SELECT(HPF,HPFSel,"Highpass",doNothing,exitEvent,noStyle
  ,VALUE("OFF",0,Filter,enterEvent)
  ,VALUE("ON",1,Filter,enterEvent)
);

SELECT(LPF,LPFSel,"Lowpass",doNothing,exitEvent,noStyle
  ,VALUE("OFF",0,Filter,enterEvent)
  ,VALUE("ON",1,Filter,enterEvent)
);

MENU(FILTERING,"Filtering",doNothing,noEvent,noStyle
    ,SUBMENU(PDEFSel)
    ,SUBMENU(HPFSel)
    ,SUBMENU(LPFSel)
    ,EXIT("<BACK")
  );

  MENU(mainMenu,"Main menu",Dispclear,exitEvent,wrapStyle
    ,SUBMENU(TUNING)
    ,SUBMENU(FILTERING)
    ,EXIT("<Back")
  );

 

  #define MAX_DEPTH 4

  MENU_OUTPUTS(out,MAX_DEPTH
    ,LCD_OUT(lcd,{0,0,20,4})
    ,NONE
  );
  
  NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);//the navigation root object

  result idle(menuOut& o,idleEvent e) {
    if (e==idling){
 
    lcd.setCursor(0,0);
    lcd.print("Tx:");
    lcd.print(TxFrq);
    lcd.print("Mhz");

    lcd.setCursor(0,1);
    lcd.print("Rx:");
    lcd.print(RxFrq);
    lcd.print("Mhz");
    }
    return proceed;
  }

  void setup() {
    pinMode(encBtn,INPUT_PULLUP);
    Serial.begin(9600);
    while(!Serial);
    encoder.begin();
    lcd.begin(20,4);
    lcd.setCursor(0, 0);
    lcd.print("Lambda-Caster VHF");
    lcd.setCursor(0, 1);
    lcd.print("Model: D105CH");
    lcd.setCursor(0, 2);
    lcd.print("FIRMWARE:");
    lcd.setCursor(0, 3);
    lcd.print("TransDriver v1.0");
    delay(3000);
    nav.idleOn();
    nav.showTitle=false;
    nav.idleTask=idle;
  }

  void loop() {
    
  if (ChSp==1){
    ChSpVal=2500;
  }
    else{
    ChSpVal=1250;  
  }

    nav.poll();
    
  }

#endif
```

neu-rah, 2017-04-01:
that idle thing is a mess, i have to look better at it and probably do a better documentation

but i managed to put it to work...

also included the customStepField

about the missing v1.0 on initial splash it shows ok here (i've just commented out the encoder and the DRA818 part to match my breadboard)

check this out:

(I've re-enabled some parts... hope it compiles ok and not forgot anything)

```c++
#ifndef ARDUINO_SAM_DUE
  #include <DRA818.h>
  #include <Wire.h>
  #include <LiquidCrystal_I2C.h>//F. Malpartida LCD's driver
  #include <menu.h>//menu macros and objects
  #include <menuIO/lcdOut.h>//malpartidas lcd menu output
  #include <menuIO/encoderIn.h>//quadrature encoder driver and fake stream
  #include <menuIO/keyIn.h>//keyboard driver and fake stream (for the encoder button)
  #include <menuIO/chainStream.h>// concatenate multiple input streams (this allows adding a button to the encoder)
  //#include <menuIO/serialOut.h>

  using namespace Menu;

  //LiquidCrystal_I2C lcd(0x27);//, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
  LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address and pinout

 // Encoder /////////////////////////////////////
  #define encA 2
  #define encB 3
  //this encoder has a button here
  #define encBtn 4

  encoderIn<encA,encB> encoder;//simple quad encoder driver
  #define ENC_SENSIVITY 4
  encoderInStream<encA,encB> encStream(encoder,ENC_SENSIVITY);// simple quad encoder fake Stream

  //a keyboard with only one key as the encoder button
  keyMap encBtn_map[]={{-encBtn,options->getCmdChar(enterCmd)}};//negative pin numbers use internal pull-up, this is on when low
  keyIn<1> encButton(encBtn_map);//1 is the number of keys

  //input from the encoder + encoder button + serial
  Stream* inputsList[]={&encStream,&encButton,&Serial};
  chainStream<3> in(inputsList);//3 is the number of inputs

float TxFrq=134.0000;
float RxFrq=134.0000;
float ChSpVal=0.1250;
int ChSp=0;
int PDEF=0;
int HPF=0;
int LPF=0;

template<typename T>
class customStepField:public menuField<T> {
public:
  customStepField(const menuFieldShadow<T>& shadow):menuField<T>(shadow) {}
  void doNav(navNode& nav,navCmd cmd) override {
    switch(cmd.cmd) {
      case upCmd:
        customStep(1);
        break;
      case downCmd:
        customStep(-1);
        break;
      default:
        menuField<T>::doNav(nav,cmd);
        break;
    }
  }
  void customStep(T increment) {
    T thisstep = increment*ChSpVal;//this  using a custom step!
    prompt::dirty=true;
    if (thisstep < 0 && (target()-low()) < -thisstep) {
      if (style()&wrapStyle) {
        target() = high();
      } else {
        target() = low();
      }
    } else if (thisstep > 0 && (high()-target()) < thisstep) {
      if (style()&wrapStyle) {
        target() = low();
      } else {
        target() = high();
      }
    } else {
      target() += thisstep;
    }
  }
};

result Tune(eventMask e,navNode& nav,prompt& item) {
    Serial.print("Tuning Updated! ");
    Serial.print("TX is ");
    Serial.println (TxFrq);
    Serial.print("RX is ");
    Serial.println(RxFrq);
    Serial.print("Ch. Space is");
    Serial.println(ChSp);

    return proceed;
  }

  result Dispclear(eventMask e,navNode& nav,prompt& item) {
       lcd.clear();
      return proceed;
  }

//this can do the direct change on ChSpVal instead of the indirection
SELECT(ChSp,ChSpSel,"Channel Space",Tune,noEvent,noStyle
  ,VALUE("12.5 KHz",0,Tune,enterEvent)
  ,VALUE("25 KHZ",1,Tune,enterEvent)
);

MENU(TUNING,"Tuning",doNothing,noEvent,noStyle
  ,altFIELD(customStepField,TxFrq,"Tx Freq","MHz",134.0000,174.0000,0,0,Tune,exitEvent,wrapStyle)
  ,altFIELD(customStepField,RxFrq,"Rx Freq","MHz",134.0000,174.0000,0,0,Tune,exitEvent,wrapStyle)
  ,SUBMENU(ChSpSel)
  ,EXIT("<Back")
);

result Filter(eventMask e,navNode& nav,prompt& item) {
    Serial.print("filters Updated!");

    return proceed;
  }

SELECT(PDEF,PDEFSel,"Pre/De Emph.",doNothing,exitEvent,noStyle
  ,VALUE("OFF",0,Filter,enterEvent)
  ,VALUE("ON",1,Filter,enterEvent)
);

SELECT(HPF,HPFSel,"Highpass",doNothing,exitEvent,noStyle
  ,VALUE("OFF",0,Filter,enterEvent)
  ,VALUE("ON",1,Filter,enterEvent)
);

SELECT(LPF,LPFSel,"Lowpass",doNothing,exitEvent,noStyle
  ,VALUE("OFF",0,Filter,enterEvent)
  ,VALUE("ON",1,Filter,enterEvent)
);

MENU(FILTERING,"Filtering",doNothing,noEvent,noStyle
    ,SUBMENU(PDEFSel)
    ,SUBMENU(HPFSel)
    ,SUBMENU(LPFSel)
    ,EXIT("<BACK")
  );

  MENU(mainMenu,"Main menu",Dispclear,exitEvent,wrapStyle
    ,SUBMENU(TUNING)
    ,SUBMENU(FILTERING)
    ,EXIT("<Back")
  );

  #define MAX_DEPTH 4

  MENU_OUTPUTS(out,MAX_DEPTH
    ,LCD_OUT(lcd,{0,0,20,4})
    ,NONE
  );

  NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);//the navigation root object

  result idle(menuOut& o,idleEvent e) {
    Serial.println("IDLEING");
    if (e==idling){

    lcd.setCursor(0,0);
    lcd.print("Tx:");
    lcd.print(TxFrq);
    lcd.print("Mhz");

    lcd.setCursor(0,1);
    lcd.print("Rx:");
    lcd.print(RxFrq);
    lcd.print("Mhz");
    }
    return proceed;
  }

  void setup() {
    //pinMode(encBtn,INPUT_PULLUP);
    Serial.begin(9600);
    while(!Serial);
    //encoder.begin();
    lcd.begin(20,4);
    nav.showTitle=false;
    lcd.setCursor(0, 0);
    lcd.print("Lambda-Caster VHF");
    lcd.setCursor(0, 1);
    lcd.print("Model: D105CH");
    lcd.setCursor(0, 2);
    lcd.print("FIRMWARE:");
    lcd.setCursor(0, 3);
    lcd.print("TransDriver v1.0");
    delay(3000);
    nav.idleTask=idle;
    nav.idleOn(idle);
  }

  void loop() {

  if (ChSp==1){
    ChSpVal=0.2500;
  }
    else{
    ChSpVal=0.1250;
  }

    nav.poll();

  }

#endif
```

Lu0z, 2017-04-01:
RUI YOU ARE A GENIUS!

The code works great!!! :D 

even the fourth row bug is fixed.

This is just amazing, i only need to put another couple of things and it's all done.

only one thing. is there a way to show a third decimal in the frequency thing? I think it rounds up the variable value and i also fear that the module could not accept it. because it only sets on values that are multiple of 12.5 or 25.0

when i am in 12.5 mode i get rounded up values like 146.38 when it should be 146.375
 Is that possible to have this grade of precision?

yes it's a hell of a little picky ba***rd XD

anyway thanks a lot for your help, i cannot wait to show you the final product.

neu-rah, 2017-04-01:
:D glad it works!

if you feel like sharing some photos or movies of your assembly i would like to use some extra hardware on my movies besides my own.

and yes its possible to customize the look of the field, many examples include a custom printing option

it goes like this:

```c++
  idx_t printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len) override {
    //menuFieldShadow<T>& s=*(menuFieldShadow<T>*)shadow;
    menuField<T>::reflex=target();
    idx_t l=prompt::printTo(root,sel,out,idx,len);
    bool ed=this==root.navFocus;
    //bool sel=nav.sel==i;
    if (l<len) {
      out.print((root.navFocus==this&&sel)?(menuField<T>::tunning?'>':':'):' ');
      l++;
      if (l<len) {
        l+=out.print(menuField<T>::reflex,3);//NOTE: this can exceed the limits!
        if (l<len) {
          l+=print_P(out,units(),len);
        }
      }
    }
    return l;
  }
```

so **just add the above method to your existing customStepField**
it will now have a custom print, here simplified to work with an LCD (removed crazy esp8266 stuff)

Lu0z, 2017-04-02:
i am making a little demo video for this setup i will share with you for sure :)

Could you give me one little more information? what's the function that picks up increment or decrement for the encoder rotation? i'm coding the volume part in the idle routine, and i need a way to increment and decrement a variable for the volume regulation. Also could this work with a regular if cycle, while displaying the value change in real time on the display or is it still linked to some static behaviour of the menu?

neu-rah, 2017-04-02:
hi :D

menu inputs are just streams (like the Serial) were you can read characters.
by default "+-/*"

that chain stream thing just joins inputs into a single input

as with the Serial you can check if a character is available and can read from the stream when it is, then increment or decrement your value when receiving characters '+' and '-'

the file is encoderIn.h

but you will probably use the **in** object  already in your sketch like this:

```c++
while(in.available()) {
    switch(in.read()) {
       case '+': /*increment value here*/; break;
       case '-':/*decrement value here*/;break
       default:break;
    }
}
```

please note that i have NOT compiled this...

neu-rah, 2017-08-25:
what is the state of this?

Lu0z, 2017-08-25:
Hello. I've decided to use a potentiometer for volume tuning, working directly on the amplifier stage. i thought that having it tuned from the microcontroller trough the encoder would be too messy to be operated, so i decided to split the funtions. unfortunately the project is a little stuck due having very little time myself to work on it. i prototyped the whole radio on a wired protoboard, but it does not work properly since wiring is not very friendly to high frequencies; i can hear rs232 commutation trough the speaker. so right now i am designing a proper pcb but as i just said i have really little time to do that and also pulling the board design off is not that easy since i never made one.

neu-rah, 2017-08-25:
I sure understand that having little time part :D, no rush, i was just wondering if i can help on this so it could be closed.

neu-rah, 2017-10-20:
please reopen if you feel this is not concluded.
you are also welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

closed 2017-10-20 by neu-rah